zigbeefordomoticz/Domoticz-Zigbee #128: DeviceList rempli de ligne sans contenu

thiklop, 2018-08-24:
J'ai été vérifier le contenu de DeviceList.txt car j'avais pas mal de KeyError.

Grosse surprise il était rempli de Device sans contenu, voire de Device mal complété voire même il y avait un vieux Device que j'avais supprimé depuis.
En résumé il faisait 92 lignes contre 12 lignes une fois purgé (80 lignes ne servant à rien).
[DeviceList.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2318262/DeviceList.txt)

Je joins le fichier origine avant purge.

Smanar, 2018-08-24:
Il y a 12 lignes de bonnes ?
Je pensais que quand il n y avait pas de valeur pour ZDeviceID, ce n'etait pas valide ?
Moi quand j'avais un truc de ce genre, j'ai du faire un reset pour avoir des trucs plus valides.

`eee8 : {'Type': {}, 'Heartbeat': '335', 'ProfileID': 'c05e', 'Status': 'inDB', 'Model': 'Ampoule.LED1624G9.Tradfri', 'RIA': '0', 'ZDeviceID': '0200', 'IEEE': '90fd9ffffe831d17', 'Battery': {}, 'MacCapa': '8e', 'RSSI': 135, 'Ep': {'01': {'0006': {}, '1000': {}, '0003': {}, '0b05': {}, '0000': {}, '0004': {}, '0300': {}, '0005': {}, '0008': {}}}}
`

thiklop, 2018-08-26:
Aucun idée, ça marche très bien sans. Question idiote : c'est quoi les ZDeviceID et à quoi ça sert ?

Après j'avais pas mal de lignes de ce genre - totalement vides :
`01 : {'RIA': '10', 'IEEE': {}, 'MacCapa': {}, 'Model': {}, 'ProfileID': {}, 'Type': {}, 'Status': '0045', 'Ep': {'': {'': {}}}, 'Battery': {}, 'ZDeviceID': {}, 'Heartbeat': '0'}`

pipiche38, 2018-08-27:
Il pourrait également servir à une reprise à chaud/froid , aujourd'hui il y a encore quelque disconnect entre celui-ci et ce qui est dans la base Domoticz.
Mais la question est raisonable

Smanar, 2018-09-06:
Je me pose encore des question sur le fichier DeviceList.txt.
Pourquoi on ne s'en servirait pas juste a la connexion du device (pour aller plus vite) ? plutot que de rajouter son contenu dans la liste de device ? Car la, a part m'afficher des devices qui n'existent plus, il me sert pas a grand chose ...

pipiche38, 2018-09-06:
Pour moi il peut etre un dictionnaire de l'ensemble de ton parc avec notamment des informations qui ne sont pas capturée comme : 
- Logical Type ( Router, End device )
- manufacturer code ( qui pourrait nous aider à identifier /différencier un device d'un autre )

zaraki673, 2018-09-06:
a la base il etait la pour verifier les infos mis en base en reçu par la
zigate (en cumulant les infos des différentes trames)
et a la création manuel de Devices (surtout utile pour réaliser des test ou
réinitialiser la base domoticz sans avoir a appairer de nouveaux les
devices)

il n est d'ailleurs utilisé qu au démarrage et comparé avec le dictionnaire
"ListOfDevices" au travers de la fonction *CheckDeviceList*
et n est écrit qu au moment de quitter/arreter le plugin

normalement il ne devrait pas etre possible d'avoir des lignes vides :( (je
check le codes pourvoir comment tu as pu les avoir)

pipiche38, 2018-10-03:
I feel that this problem has been address by @Smanar by adding this piece of code in onHeartbeat process. So even if we didn't fix the root cause, we are limiting the impact.

In parallel, we have address in many area the way entries are created.
The file is now optional (on 'dev' branch), but we still have to remind that even if we don't have the file, we are allocating objects in memory !

if len(self.ListOfDevices[key]) == 0:
       Domoticz.Debug("Bad devices detected (empty one), remove it, adr :" + str(key))
       del self.ListOfDevices[key]
       continue

I suggest to close this and if it is coming back to re-opne or open a new one

closed 2018-10-03 by pipiche38