zigbeefordomoticz/Domoticz-Zigbee #132: Large delay between Xiaomi PIR detection and status reflect in Domoticz

Mat2Vence, 2018-08-27:
I dont know when the problem has started but I am facing an unexpected behavior with xiaomi Motion detector. I have the same kind of behavior with both old PIR and new Aqara PIR.
If I am moving in front of the motion detector, I am seeing the detection on the Domoticz device minutes after it occurs. I am missing also a lot of event... even if the PIR are placed very closed to the zigate.

I am seeing this message is my log. Don't know if there is a link:
`(Zigate) Transport is not connected, write directive to 'ZiGate' ignored`

Mat2Vence, 2018-08-27:
seems to work using version beta 4.9954. I will make some trial before closing.

pipiche38, 2018-08-28:
@Mat2Vence what type of Zigate key do you have USB or Wifi ?
Are you on the Master or Developement branch of the plugin ?

thiklop, 2018-08-28:
Are you using master or developement branch ? I use ti have the same issue
but it was solved using the developement branch

Mat2Vence, 2018-08-28:
I guess I am on the master branch of the plugin... I will have to double check this evening.
However, it seems to be solved on the branch of the plugin I am using in case I am switching to the beta version of Domoticz.
I will make some trials this evening in order to see how it behaves depending on Domoticz/Plugin version.

Mat2Vence, 2018-08-30:
Here are my trials...
So using the latest beta 4.9954 of domoticz and the master branch of plugin, I am still facing some use delay. I though the problem was solve but it's not.

I then turn the plugin to the developement branch. I am still using the same beta version of Domoticz. It looks much better then. 

Do you have an idea were it was comming from in the master branch? If so, are we expected a fix in a the master branch?
Thanks for your help.

pipiche38, 2018-08-31:
@Mat2Vence thanks for making a trial of developement branch.

We have some major changes in the way we are capturing the Zigate frame/data and which could have a positive impact on your issue.

Last we are currently evaluating the way to push what have been done on developement to master.

So stay tune.

BTW, It would be really great if you could contribute or support us by providing a list of ZigBee devices you are using with the plugin. Here after is what we are currently building:
https://github.com/sasu-drooz/Domoticz-Zigate/blob/developement/Compatible-Hardware.md

pipiche38, 2018-09-01:
@Mat2Vence we would push the current 'developement' branch to the master pretty soon a week or 2. In the mean time, as you have use it could you let me know if you have seen any regression with other devices ? Could you kindly let me know what devices are connected to the Zigate Plugin ?

pipiche38, 2018-09-08:
Version 2.4.0 has just been release today on Master branch.

closed 2018-09-08 by pipiche38