zigbeefordomoticz/Domoticz-Zigbee #135: ZigBee SQN. SQN is provided as part of the zigbee protocol. This is a number managed by each IoT device.

pipiche38, 2018-08-29:
We should make a use of it in order to control that the communication stack is operating correctly , we are not loosing message and when command are send to a device we get the acknowledgement and the response if expected

pipiche38, 2018-09-03:
This is also link to the Decode8000 where we get status of each of the command but nothing is done.
I'm wondering if it wouldn't make sense to have a kind of block sendZigateCommand() where you would be waiting for the ack of your command. This might bring a hard link between onMessage() and where the sendZigateCommand() is called, but we could also give a short timeout like 1s .

Then it would interesting to track the response to command (when expecting). probably through an enhancement of the ListOfDevices, where we could add fields like : 

- LastSQN         // Last known SQN
- LastCMD        // Last Command send ( cmd+datas) so eventually we could resend it !
- ExpectedRSP   // Expected responde code 0x8042 ....

pipiche38, 2018-09-23:
Xcube Aqara s'incrémente de 1 en 1

pipiche38, 2018-09-23:
It could be possible that for Powered Devices MacCapa 8e the SQN works in a different way

ricky074, 2018-09-23:
Xiaomi Capteur de température/humidité/pression s'incrémente de 1 en 1
Xiaomi Détecteur d'ouverture de porte s'incrémente de 1 en 1

pipiche38, 2018-10-03:
for now implemented as far as Battery powered devices are concerned on 'dev'

closed 2018-10-03 by pipiche38