zigbeefordomoticz/Domoticz-Zigbee #140: Update of Device[ID].Options['Zigate'] . Currently this is never updated as far as I understood. So it might be good to update .

pipiche38, 2018-08-29:


pipiche38, 2018-09-03:
Options['Zigate'] is stored to the Domoticz database when creating the device. 
However a long the life of the system some of the fields will be updated RSSI, BatteryLevel and it might be a good idea to update the Domoticz Database with the latest information on a regular basis (or at shutdown time)

pipiche38, 2018-09-12:
@Smanar tu peux préciser où dans le code, car je n'ai pas trouvé ?

Smanar, 2018-09-12:
Dans la partie desactivee, dans le coin de ca https://github.com/sasu-drooz/Domoticz-Zigate/blob/code-split/z_heartbeat.py#L237
Affiche en debug la valeur de DOptions['Zigate']
Ca va t'afficher la valeur d'un autre device que celui sur lequel tu est Devices[x].DeviceID.
Moi ca m'affichait les vlaur d'une lampe alors que j'etais en train d'apairer une sonde de temperature.

pipiche38, 2018-09-12:
Mais c'est un commentaire , non ?

pipiche38, 2018-09-12:
Je suis désolé. Peut-etre fatigué ce soir, mais je ne vois pas le problème/
```

for x in Devices:
                                if Devices[x].DeviceID == str(key) :
                                        IsCreated = True
                                        Domoticz.Debug("onHeartbeat - Devices already exist. Unit=" + str(x) + " versus " + str(self.ListOfDevices[key]) )
```

Quand Devices[x].DeviceID == str(key) est vrai, on affiche x qui est l'Index dans la table des Devices Domoticz et self.ListOfDevices[key]. Key venant du parcours de ListOfDevices.

Donc je ne vois pas ou l'un peut pointer sur un autre ...

Smanar, 2018-09-12:
Essaye de reparer la partie desactivee, tu vas comprendres.
En supposant que tu essayes de pairer un detecteur de temperature, tu vas avoir Devices[x].DeviceID = key = adresse du detecteur mais si tu fais
```
DOptions = Devices[x].Options
print DOptions['Zigate']
```

Ca va t'afficher les parametre du device ZZZZ au lieu de ceux du capteur.
J'ai pas dit que le bug etait la, mais y a plus qu'un soucis de mise a jour.

pipiche38, 2018-09-12:
Ce qui ne me suprend pas puisque ces commentaires ne sont dans le if !

Smanar, 2018-09-12:
Ok donc c'est normal que Devices[1].Options['Zigate']['IEE'] corresponde a l'IEE de Devices[2].DeviceID au lieu de Devices[1].DeviceID ?

pipiche38, 2018-10-03:
Implemented in the recent 'dev' branch / 445bb321d77078214722ea9f77435e6b01b2fce2

closed 2018-10-03 by pipiche38