zigbeefordomoticz/Domoticz-Zigbee #146: Checksum error at startup

Smanar, 2018-08-31:
**Describe the bug**
Error message at startup
```
 2018-08-31 20:11:59.137 Status: (Zigate) Started.
2018-08-31 20:11:59.359 Status: (Zigate) Entering work loop.
2018-08-31 20:11:59.359 Status: (Zigate) Initialized version 2.3.5 developement, author 'zaraki673'
2018-08-31 20:11:59.754 Error: (Zigate) ZigateRead - Checksum error: 69 / 97 MsgType = 8000 MsgLength : 0005 RawData '0180000005970000001200ff018003001e9b01010400000001000300040005000600080019010110000300020102040003' 
```

**To Reproduce**
Steps to reproduce the behavior:
1 - set "erase persistent data" to True
2- restart plugin

Developement version

pipiche38, 2018-08-31:
Raw Data is the data just extracted from ZigateDecode.
The Message Type 8000 shoudl expect a message of 8! That one seems to be much more !

As we are in a restart mode of the plugin, could we imagine that we are receiving partialy message from the past ?

@Smanar, could you attach a better log related to that issue. For instance I would like to understand if we this frame (generating the checksum issue ) is happening before that we Start the Network ... So I would appreciate to have the log file from the plugin start until the error (at least).

pipiche38, 2018-09-01:
@Smanar  no issue at all on my side.

Could you clarify what is your Domoticz version ? Are you on the 3.xx , 4.xxx, beta ?
Do you have other python plugin going on at the same time ?

Smanar, 2018-09-02:
Domoticz V4.9700. No more python script ATM.
And I have used the same command with ZWGUI on windows, no problem. "Erase PD" and "Reset".
But just after, I have the bug on Domoticz every time I re enable "Erase persistent data"

pipiche38, 2018-09-02:
And is that always the same frames ? 

```
2018-09-01 19:17:12.563 Error: (Zigate) ZigateRead - Checksum error: 69 / 97 MsgType = 8000 MsgLength : 0005 RawData '0180000005970000001200ff018003001e9b01010400000001000300040005000600080019010110000300020102040003'
2018-09-01 19:17:22.167 Error: (Zigate) ZigateRead - Checksum error: 76 / 8d MsgType = 8003 MsgLength : 001a RawData '018003001a8d01011180000005970000001200ff018003001e9b01010400000001000300040005000600080019010110000300020102040003'
2018-09-01 19:17:32.082 Error: (Zigate) ZigateRead - Checksum error: 69 / 97 MsgType = 8000 MsgLength : 0005 RawData '0180000005970000001200ff018003001e9b01010400000001000300040005000600080019010110000300020102040003'
```

Smanar, 2018-09-02:
These 3 with total restart but only

> 2018-08-31 20:11:59.754 Error: (Zigate) ZigateRead - Checksum error: 69 / 97 MsgType = 8000 MsgLength : 0005 RawData '0180000005970000001200ff018003001e9b01010400000001000300040005000600080019010110000300020102040003'

If I just stop/restart plugin.

Smanar, 2018-09-06:
It's ok for me ^^.
I haven't see the command have worked (too much line in log) but no more error message.

closed 2018-09-06 by Smanar