zigbeefordomoticz/Domoticz-Zigbee #147: Compatibility issue with Connected Plu Celian/Netamo/Legrand

pipiche38, 2018-09-01:
The Celian/Netamo/Legrand connected plug can be managed with the plugin , however only the switch on/off is currently working.

It seems that there is one cluster (providing power information) not available . This would require an upgrade of the device firmware which unfortunately can only be done through Netatmo/Legrand gateway :-(

Smanar, 2018-09-01:
Someone have the procedure to "reset" devices, to launch a second time the pairing phase ?
I have the prb with a switch, but I think it's same for the plug.

pipiche38, 2018-09-01:
@Smanar here is the Legrand documentation. There is a reset button . Check "fonctionnement'

http://docdif.fr.grpleg.com/general/MEDIAGRP/NP-FT-GT/F02486FR-00.pdf

Smanar, 2018-09-01:
Bad luck.
It's the old model (not compatible with zigate), Legrand have stopped this model, now all device have "with netamo inside" and are compatible.
http://docdif.fr.grpleg.com/general/MEDIAGRP/NP-FT-GT/LE10635AA.pdf
http://docdif.fr.grpleg.com/general/MEDIAGRP/NP-FT-GT/LE10383AB.pdf

I know only 1 device with the reset button, the "micromodule"

Smanar, 2018-09-01:
I have broked my switch, but I have find. Front side, rly hard to access.