zigbeefordomoticz/Domoticz-Zigbee #148: Enhacement of DeviceConf.txt

pipiche38, 2018-09-01:
In order to speed up the acquiring and device creation process, it would be good that DeviceConf.txt contains all of the known/compatible Zigate plugin device.

Today the entry point in the DeviceConf.txt is the device Model. Unfortunately not all devices announced their device model. Some models give ProfileID and DeviceID

In such it would be great that DeviceCOnf.txt could be accessible via Model name, but also through ProfileId/DeviceID. 

The positive impacts are : 
- Less hardcoding in the onHearbeat() while creating device. As all information will be in the DeviceConf.txt
- DeviceConf.txt will contains defacto the list of known/confirmed devices.

pipiche38, 2018-09-02:
As we might have ProfileID+DeviceID for several devices (Ikea Tradfri / Netatmo ), it might be good that number of EP or list of EP could be also included in the Key search

In case we have duplicate, we should go through the process, and most-likely Cluster list could be a good differentiator

Smanar, 2018-09-02:
So you want to remove the hardcoded engine that detect device, for a Deviceconf.txt based database ?

pipiche38, 2018-09-02:
As much as we can

Smanar, 2018-09-02:
So to resume you want a engine who can use theses information

> Processed d7e2 end results is : {'RSSI': {}, 'RIA': '0', 'ProfileID': '0104', 'Status': '8043', 'Heartbeat': '0', 'Type': {}, 'Model': {}, 'MacCapa': '8e', 'ZDeviceID': '0100', 'IEEE': '00047400007a2666', 'Ep': {'01': {'0003': {}, '0004': {}, '0008': {}, '0006': {}, '000f': {}, '0005': {}, '0000': {}}}, 'Battery': {}}

To make device with adding type + name + (other thing ?) taked from the deviceconf.txt file ?

pipiche38, 2018-09-02:
Ok, so lets go further : 

For Xiaomi you get 0x004d for annoucement and you get a ClusterID 0x0000 , attrID = 0005
This provide : Model Name and then you can get all other information from DeviceConf.txt

I would like to use the similar process for all known devices. Which means that we would remove the complexity in the creation process inside onHearbeat when for example:
```
if status != "UNKNOW" and status != "DUP":
                                        if self.ListOfDevices[key]['MacCapa']=="8e" :  # Device sur secteur
```
Of course, if you get a new device, you will have to look after the debug and the frame and see what is the profileID, DeviceID ....

A new device is annouced ( 0x004d ) we get then : 
- saddr
- IEEE
- MacCapa

if the saddr/IEEE is not yet known lets continue, else address the issue

We request 0x0045 in order to get 0x8045

0x8045 will provide us : 
- a number of EndPoint

We continue the process by sending 0x0043 in order to get 0x8043

0x8043 will provide us : 
- For each EndPoint
  - Profile Id
  - Device Id
  - Cluster In 
  - Cluster Out

The idea would then be after the discovery process to look into DeviceConf with 
( ProfileID, DeviceID, Cluster, EP) and find a match .

`plug.Celiane.Legrand':{'ProfileID':'0104'}:{'ZdeviceID':'010a'}{'Ep': {'01':{'0000','0003','0004','0006','0005','fc01'}},'Type':'Switch/PowerMeter'},`

Smanar, 2018-09-02:
Something like that

https://pastebin.com/j22ViP2U

pipiche38, 2018-09-03:
Oui, je pense que c'est l'idée... Par contre je serais peut-etre un peu plus flexible sur les cas de rechreche/Update

Recherche+Update avec seulement le Model name -> Si on trouve et que cela est unique on update, sinon on retourne que ce n'est pas unique. A priori cela devrait etre unique puisque le Model est donné par le périphérique.

Recherche+Update avec ProfileID/ZDeviceID -Si on trouve et que cela est unqieu on update, sinon on retourne que ce n'est pas unique

Recherche+Update avec profileID/ZdeviceID/EP+Cluster par EP - Si on trouve et que cela est unique on update , sinon .... le device n'est pas connu et là soit via l'EP et Cluster on est capable d'identifier quel type de Device. Soit on repond non connu

- On peut imaginer que on reconnaisse par exemple le Type via Cluster
Cluster 0x0006 -> On/Off -> Switch
Cluster 0x0008 -> Level Control
....

Smanar, 2018-09-03:
Ben moi non justement ^^. Pour moi cette fonction est sensee etre fiable a 100%, et la difference entre un inter legrand et une lampe se fait juste avec les clusters, tout est identique.

1- si model name, ok > on retourne model name
2- test avec profileID/ZdeviceID/EP+Cluster par EP
3 - on retourne une erreur si pas trouve.

Le but du truc c'est d'avoir un fichier DeviceConf.txt en beton. On aura une liste precise de ce qui est supporte ou pas. Et si quelqu'un trouve un nouveau device, suffit de le rajouter dans le fichier. A quoi ca sert de laisser passer un device qui correspond "a peu pres" juste pour assurer le fonctionnement du plugin ?

pipiche38, 2018-09-03:
Ca ce plaide. 
On pourrait alors imaginer la fonction qui en cas d'erreur, dump effectivement tous ce qu'elle a collecté

Smanar, 2018-09-03:
Yep, et vu que c'est un fichier texte, n'importe qui pourrait les rajouter dedans pr tester.

closed 2019-01-14 by pipiche38