zigbeefordomoticz/Domoticz-Zigbee #152: MacCapa test

pipiche38, 2018-09-03:
In the plugin code, some tests against the value of 'MacCapa' are done against "8e". 

However this is not the right way to do as MacCapa is a bit field where bit2 indicates if it is Main Power or not.
```
MacCapa
Bit0 - 0 Node not able to act as Co-ordinator; 1 Node able to act as Co-ordinator
bit1 - 0 Reduced-Function Device ; 1 Full Function Device
bit2 - 0 Node is not mains-powered ; 1 Node is mains-powered
bit3 - 0 Receiver disabled during idle periods ; 1 Receiver enabled during idle period
bit4 - reserved
bit5 - reserved
bit6 - Security caability: 0 Standard ; 1 High security
bit7 - 0 Network address need not be allocated to node; 1 Network address shoul dbe allocated to node
```

So it would be more appropriate to test if the bit2 of the 'MacCapa' bit field is set in order to know if it is a Main Powered device or not !
Even more, it could be much faster if we add one boolean field to the  ListOfDevices indicating if this is Main Power or not . 

Suggestion:
 self.ListOfDevices[key]['Power'] = {} /* 1 for Main Power; 0 for Battery */

Smanar, 2018-09-03:
Ben, qu est ce qui nous dit que plus tard on ne va pas avoir besoin des autres bit ? > autant garder tout MaCapa
Pour savoir si c'est Main Power or  Batterie, suffit de regarder le niveau de batterie du device.
Pour le moment ca ne sert que pour identifier les devices, ce qui va sauter avec la future fonction qui va faire a 100% via le fichier deviceconf.txt (si cette amelioration est choisie)

pipiche38, 2018-09-03:
Bon point. A il nous faudrait alors une fonction qui permet de décoder ce bit field.

Smanar, 2018-09-03:
Ca je peux faire, mais vu que pour le moment ca nous sert pas et que le code est bloque, ce sera dans quelques jours.

pipiche38, 2018-11-22:
implemented in beta-3.2

closed 2018-11-22 by pipiche38