zigbeefordomoticz/Domoticz-Zigbee #153: Temperature refresh rate/threshold

mce35, 2018-09-03:
Hello,

I bought a Zigate a few weeks ago. I am using it with a few xiaomi v1 temp/humidity sensors (among other).

Sensors connected to the Zigate only report value change of at least 0.5°C (not sure about humidity change), whereas sensors connected to the original Xiaomi gateway report changes of 0.1°C. This leads to "hole" in graphs (there may be a few hours between 2 reports).

I have placed 2 sensors side by side and the graph of the one connected to the xiaomi gateway is far smoother than the one connected to the Zigate.

Graph from the sensor connected to the Zigate:
(No data from 10:10 (21.2°C) to 20:05 (21.7°C), because there were no change >=0.5°C I guess)
[image]

Graph from the sensor connected to the Xiaomi Gateway:
[image]

I am using the developement branch of the Zigate plugin.

The sensor connected to the Zigate is ~1 meter away, so I don't think it's a communication issue (RSSI 147).

I have attach a log of the Decode8102 messages for the a1c9 (short address of the sensor) device (grep -i Decode8102 /var/log/rasp3.log | grep a1c9)

[Decode8102_a1c9.log](https://github.com/sasu-drooz/Domoticz-Zigate/files/2346545/Decode8102_a1c9.log)

We can see some messages received every 55 minutes (no hole in SQN), but they doesn't contain temp/humidity measurement (ClusterID = 0000 AttrID = ff01 -> only battery measurement I think?).

Someone could help solving this issue please?

Regards.

pipiche38, 2018-09-04:
You right Cluster 0000 with attribute ff01 is for battery level.  But doesn't you get ClusterID "0402" as well 

Here are the Temperature measure that I can see in your log.
```

Sep  3 01:11:39 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = bf ClusterID = 0402 AttrID = 0000 Attribute Data = 08de
Sep  3 01:20:52 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = c2 ClusterID = 0402 AttrID = 0000 Attribute Data = 0911
Sep  3 02:17:59 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = c4 ClusterID = 0402 AttrID = 0000 Attribute Data = 08de
Sep  3 05:03:48 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = ca ClusterID = 0402 AttrID = 0000 Attribute Data = 08ab
Sep  3 07:37:08 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = ce ClusterID = 0402 AttrID = 0000 Attribute Data = 0879
Sep  3 09:10:51 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = d2 ClusterID = 0402 AttrID = 0000 Attribute Data = 0846
Sep  3 20:00:31 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = e0 ClusterID = 0402 AttrID = 0000 Attribute Data = 087b
Sep  3 20:10:19 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = e2 ClusterID = 0402 AttrID = 0000 Attribute Data = 08ae
Sep  3 20:17:36 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = e4 ClusterID = 0402 AttrID = 0000 Attribute Data = 08e1
Sep  3 20:26:43 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = e6 ClusterID = 0402 AttrID = 0000 Attribute Data = 0916
Sep  3 20:38:56 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = e8 ClusterID = 0402 AttrID = 0000 Attribute Data = 094a
Sep  3 20:55:10 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = eb ClusterID = 0402 AttrID = 0000 Attribute Data = 097d
Sep  3 21:46:32 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = ee ClusterID = 0402 AttrID = 0000 Attribute Data = 094a
Sep  3 22:13:53 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = f0 ClusterID = 0402 AttrID = 0000 Attribute Data = 0916
Sep  3 22:40:39 rasp3 domoticz: (Zigate) Decode8102 : Attribute Report from a1c9 SQN = f3 ClusterID = 0402 AttrID = 0000 Attribute Data = 08e2
```

Did you check if you have any checksum error in the log ?

I suspect that the Xiaomi gateway is sending a command to the sensor to get 0.1°C threshold. At least this is how it works on the OpenZwave devices.  This would require further investigation and especially we might have to send a command to the device to set this threshold. But as you know this is just about reverse engineering as we don't have much internal documentation from Xiaomi.

mce35, 2018-09-04:
I didn't find any checksum error. _grep -i checksum /var/log/rasp3.log | grep -i err_ doesn't return anything.

I also think that the gateway is sending some commands to set the threshold, but since I don't have a zigbee sniffer I don't know how to investigate further.

Do you know if temp/humidity sensors v2 (aqara) also report only 0.5°C change with the Zigate?

mce35, 2018-09-04:
From a comment here: http://faire-ca-soi-meme.fr/hack/2017/06/30/xiaomi-mi-gestion-tensions-de-piles/#comment-2044
It seems that the Cluster 0000/Attr FF01 contains temperature/humidity measurement (as well as battery level already decoded).
Decoding this would at least allow to have a measurement every ~55 minutes.

Do you think it would be worth decoding this?

mce35, 2018-09-05:
I will try to add 1 or 2 more temp sensors to see what happens.
I have only v1 sensors, I will also buy a v2 and try it, but it will take ~2weeks to be delivered.

pipiche38, 2018-09-05:
If you look to what I documented on #155, I'm wondering if what is written in the reference you put earlier is still valid, as I wonder how to make the difference between the BatteryLevel which is indeed decoded out of the 0000/ff01 and a potential Temperature/Humidity/Pressure. 

As I don't have such device, I would be interested if you could kindly put the plugin in Debug mode and capture few frame of Cluster 0000 with Attribute FF01, so we can re-check if this is still valid. On the latest version of the dev branch (today) I have added more debug

zaraki673, 2018-09-05:
For this problem, check battery level it could be the raison why the sensor
send less temperature changes values

pipiche38, 2018-09-05:
@zaraki673 and @mce35 I think if we can also decode the information here based on the http://faire-ca-soi-meme.fr/hack/2017/06/30/xiaomi-mi-gestion-tensions-de-piles/#comment-2044 we can see that : 

For 3290 
0121bd0b0421a81305211100062402000000006429e408652106180a210000
- (after 6429 tag)  Temperature is e408 = 58376
- (after 6521 tag ) Humidity is 0618 = 1560
Is that making sense ? This look high temp and dry 

For a1c9 
01219f0b0421a81305210b0006240200000000642965086521b61a0a210000
- (after 6429 tag)  Temperature is 6508= 25864
- (after 6521 tag ) Humidity is b61a= 46618

Values seems strange !

pipiche38, 2018-09-05:
@zaraki673 and @mce35 I think if we can also decode the information here based on the http://faire-ca-soi-meme.fr/hack/2017/06/30/xiaomi-mi-gestion-tensions-de-piles/#comment-2044 we can see that :

For 3290
0121bd0b0421a81305211100062402000000006429e408652106180a210000

    (after 6429 tag) Temperature is e408 = 2276 = 22.76°
    (after 6521 tag ) Humidity is 0618 = 61.50%

For a1c9
01219f0b0421a81305210b0006240200000000642965086521b61a0a210000

    (after 6429 tag) Temperature is 6508= 21.49°
    (after 6521 tag ) Humidity is b61a= 68.38%

Is that the figures you are expecting ?

mce35, 2018-09-05:
Yes it seems correct. I checked values for a few hours and it seems consistent.
```
$ grep -i Decode8102 /var/log/rasp3.log | grep a1c9 | grep 6429 | sed 's/\(.*\) rasp3.*6429\(..\)\(..\).*/\1 value \3\2/'
Sep  5 00:24:50 value 0882
Sep  5 00:24:50 value 0882
Sep  5 01:18:57 value 08d7
Sep  5 01:18:57 value 08d7
Sep  5 02:10:24 value 08c6
Sep  5 02:10:24 value 08c6
Sep  5 03:05:27 value 08d1
Sep  5 03:05:27 value 08d1
Sep  5 04:04:51 value 08cc
Sep  5 04:04:51 value 08cc
Sep  5 04:58:33 value 08bf
Sep  5 04:58:33 value 08bf
Sep  5 05:57:07 value 08b1
Sep  5 05:57:07 value 08b1
Sep  5 06:50:14 value 08a4
Sep  5 06:50:14 value 08a4
Sep  5 07:49:03 value 0880
Sep  5 07:49:03 value 0880
Sep  5 08:43:11 value 0865
Sep  5 08:43:11 value 0865
Sep  5 09:35:23 value 085a
Sep  5 09:35:23 value 085a
Sep  5 10:33:22 value 0854
Sep  5 10:33:22 value 0854
Sep  5 11:32:06 value 0850
Sep  5 11:32:06 value 0850
Sep  5 12:27:34 value 084c
Sep  5 12:27:34 value 084c
Sep  5 13:25:28 value 0847
Sep  5 13:25:28 value 0847
```

pipiche38, 2018-09-05:
Ok. Sound good. We are currently in the process of pushing a release to Master, so we will keep focus on that action, and then we will see how to integrate this change in the best way ... 
I think the Tag / Value approach in the 0000/ff01 looks pretty smart ...

Stay tune on #155

pipiche38, 2018-09-06:
This looks very strange as for @thiklop he got update at .1°C threshold.

Does this devices have been connected once to the Xiaomi gateway ?

pipiche38, 2018-09-10:
Stay tune on #155

closed 2018-09-10 by pipiche38