zigbeefordomoticz/Domoticz-Zigbee #156: If Zigate receiver miss an open - close frame of a Xiaomi windows sensor, the status of the door stay wrong in Domoticz forever.

dondipietro, 2018-09-05:
Hello,
With Xiaomi Windows sensor, if we open the windows, the sensor send the frame to zigate. If we miss that frame, the status of the windows in domoticz stay closed for days. It never update.

It only update if we close and open the windows again.

Does the Xiaomi windows sensor send for example every hour his open/close status or only if we activate it ?

I don't know if this bug is related to zigate or (domoticz zigate plugin) or Xiaomi device.

Is it possible to send an order to the windows sensor to ask for its open/close status ?

Thank you very much.

zaraki673, 2018-09-05:
>
> Hi,

> Does the Xiaomi windows sensor send for example every hour his open/close
> status or only if we activate it ?
>
Yes, only one time when it is activate, as temp/hum that will send data
only if their value have changed

> Is it possible to send an order to the windows sensor to ask for its
> open/close status?
>
Don't know if possible, we need to find/check if a command exist to ask for
status and if sensor send a response.

>

pipiche38, 2018-09-05:
@zaraki673 From the device announcement the Xiaomi get Cluster 0x0000 In and Out. So if I understood that means that we can send cluster 0x0000 as a command. In such we might get an answer to be decoded. Am I stupid to say that ?

Else we can try the readAttribute Command 0x0100. Basically this is what I'm doing with the Profalux Shutters
nn

Smanar, 2018-09-05:
Almost same problem for bulb/lamp.
If a toogle is skipped, the state is wrong, and impossible to make scenario like "if light is on" because not reliable.

thiklop, 2018-09-10:
I confirm the report but I think that it came from the device. When I push shortly the sync button, it does blink but the status don't refresh.

pipiche38, 2018-09-10:
Might be implemented via #155, as when sending the 0x0000 cluster , it seems that we could find the On/Off state. (At least for Plugs)

pipiche38, 2018-09-10:
@thiklop it would be good if you could check the Cluster 0000/ff01 messages in the log (with code-split branche), if you get this ?

Smanar, 2018-09-10:
And yep, I can confim, we can have on/off position, even dimmer position, I have tried it for bulb.

zaraki673, 2018-09-10:
> Je vous fais le retour asap demain

Je confirme ça fonctionne 😉

dondipietro, 2018-09-11:
Super ! 
Concrètement comment peut on récupérer après coup la position d'un switch ?

Merci beaucoup

thiklop, 2018-09-11:
Est-ce que cela pourrait fonctionner avec le #173 ?

Smanar, 2018-09-11:
@thiklop yep justement.
Si tu envois la commande z_output.ReadAttributeRequest(self,'17b8', '0006' , '0000')
Tu recuperes la position du switch 17b8. Et le code est deja en place dans la fonction ReadCluster()

```
	elif MsgClusterId=="0006" :  # (General: On/Off) xiaomi
		if MsgAttrID=="0000" or MsgAttrID=="8000":
			z_domoticz.MajDomoDevice(self, Devices, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgClusterData)
			self.ListOfDevices[MsgSrcAddr]['Ep'][MsgSrcEp][MsgClusterId]=MsgClusterData
Domoticz.Debug("ReadCluster - ClusterId=0006 - reception General: On/Off : " + str(MsgClusterData) )
```

pipiche38, 2018-09-11:
Ce sur quoi je bosse est de récupérer cette information sur la trame du Cluster 0x0000. On arrive a avoir Temperature, Humidity et eventuellement Pression Atmo.
Dans la mesure ou c'est un device sur pile, il faut éviter de faire du polling , car sinon on va vite vider la batterie..

Smanar, 2018-09-11:
Lol, j'imagine.
D'ailleurs par curiosite qu'est ce qui declenche ces trames, les remontes de temperature fonctionnaient avant, donc c'est un plus ?

pipiche38, 2018-09-11:
Absolument , c'est un plus ... Quand le device envoie l'etat de la batterie il envoie des infos en plus.
Comme Temp/Hum/OnOff/Pression atmospherique. Donc en 1 communiction on a toutes les infos qu'il peut envoyer

Smanar, 2018-09-11:
Et on sait comment ca marche ?
Une tempo ? quand la batterie baisse de 0.1V ? Quand venus passe devant mercure ?
J'ai des logs tout pourri, mais ca arrive tout les combien de temps a peu pres ?

pipiche38, 2018-09-11:
J'aime assez l'idée de Vénus passant de Mercure ;-)

Comme j'ai pas de trames et que j'utilise celle qu'on m'envoie ... J'ai pas du tout de tempo ... A voir le retour ...

Mais c'est la trame Cluster 0x0000/AtributeID FF01. Et j'utilise ta petite macro pour identifier les Tags

pipiche38, 2018-09-11:
Typiquement la prise Xiaomi m'envoie également un status si elle est on

pipiche38, 2018-09-11:
Implemented in 5796f1e66b3adf81ba897f446f2f362c248bb662

dondipietro, 2018-09-12:
Hello,
It seems that the function ReadAttributeRequest is not working with Xiaomi windows sensor.
I think that as it is a **sleeping device**, the device don't receive the request because it is sleeping.

May be we have to send the ReadAttributeRequest very quicly when the device wake up for hearth beat ?

Smanar, 2018-09-12:
Ha ?
Good to know, I haven't see yet how work this kind of device.

zaraki673, 2018-09-12:
Xiaomi device answer only for a few secondes/minutes (don t know how many times) when they are joining ZIGATE network, after they are in sleep mode until they send new values (this is true for battery device)

Smanar, 2018-09-12:
So impossible to send them commands in this state ?
But like have said @pipiche38 ReadAttributeRequest is not a pooling command, realy bad idea on device with batterie, especially that theses devices have their own pooling system.

zaraki673, 2018-09-12:
Yes, impossible to send them command or i don't find how to do it /when to do it (except when it join network)/ Witch command send ?
Pipiche had decode a message from thoses devices with information of state, battery ... this message is send each time you push device button or every few minutes

Smanar, 2018-09-12:
Arf, not good for me.
So if the device don't send a special information at connection, it's not possible for me to ask for it later.
I need to make test to know when it go to sleep mode and become unusable for me.

zaraki673, 2018-09-12:
no at connection, you can ask information about it, but don't know how many times,
At the moment, we ask 2 questions to new zigbee device, except if it send information need to be know by itself, that are : 
0x0045 Request EP list
0x0043 request Simple Descriptor for each EP
(you can see that in heartbeat)

Smanar, 2018-09-12:
Yep, but it's exactly my problem, "How many times" or perhaps there is some authorised commands or not. I don't see anything for that on documentation, but 121 pages in english ......

pipiche38, 2018-09-12:
I think it is not about ZigBee, but much more on how brand are implementing ZigBee.
As of now , the Xiaomi devices and especially the Temp/Humi/Baro send with the Batterylevel all of this information and we have now implemented this in the current dev.

If you look at Open-Z-Wave, this is much more the same pricniple. You get an update when there is significative change and on a regular basic. Which if I'm not mistaken is every 50' on Xiaomi.

closed 2018-09-12 by pipiche38