zigbeefordomoticz/Domoticz-Zigbee #157: No data history from Zigate

DebugBill, 2018-09-05:
Hi,

I have just installed zigate in my otherwise working Domoticz installation using mainly Zwave and Teleinfo. It worked almost immediatly and I can now see my xiaomi temp+baro+humid sensor. This is domotiz V4.9992.

The only problem is that I get no historical data when clicking on the "log" button in the devices. For all others I can see daily, monthly, yearly trends, but for Zigate devices.... nothing.

Is that a known bug/feature or limitation of the python framework, or I my experiencing someting new?

As the developer of the Teleinfo module I have some knowledge of Domoticz code base in C++ but none of the python framework whatsoever. I could have a look at it, but I would like to know if I am looking for a bug to fix here or a feature to develop.

Yours,

pipiche38, 2018-09-05:
This looks strange as we are currently working on an issue which is reporting missing data, but not about no data at all!

See #153 

I would be interested if you could have a look of the investigation we have done on #153 and if you can check on your configuration side

IN short you should have the graph, but you might have some missing point

DebugBill, 2018-09-06:
Hi, 
Thanks for your answers. The problem is suprising. I reported the issue not mentionning I am using Chrome Version 68.0.3440.106 (Build officiel) (64 bits).
I checked IE and Firefox..... all the the graphs are there. 
In Chrome, only the zwave ones are there, but not the Zigate"s. I attach a screenshot showing the empty page. 
I cleared the cache, no difference
I tried a "private browsing" window. Bingo, it works. 
Looks like it is indeed a cache issue, and the problem lies in Domoticz itself rather than in Zigate

Thanks mce35 for the pointer
[image]

pipiche38, 2018-09-06:
Great, I'm glad that you find a way to solev your issue. I know that Domoticz is battling with the cache and quiet often cleaning it make it work.

pipiche38, 2018-09-07:
So I guess we can close the issue ?

DebugBill, 2018-09-07:
Sure

closed 2018-09-07 by DebugBill