zigbeefordomoticz/Domoticz-Zigbee #158: Lost some messages with xiaomi sensors (door sensor and Temp/Hum/Press)

ricky074, 2018-09-06:
I've Xiaomi sensors, 4 doorssensors (22f0 - 80c2 - f64b - 6a4c) and 1 Temp/Hum/Pres (63a3)
For information, the temperature sensor is very closed to the zigate (less than 10cm)
80c2 and 6a4c are outdoor
f64b and 22f0 are indoor

Since last week, I monitored them and more particulary SQN number from domoticz log file.

I 've noticed some mistakes.
Some SQN increments doesn't appear in log file.
You will see my log file en attached file 
[domoticz.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2355872/domoticz.txt)

Here is the results :

Device ID = 63a3 (Temp/Hum/Pres)
  - 2018-09-03 15:50:05.581 LastSQN = 239 ==> 2018-09-03 16:26:33.762 SQN = 241
  ==> 1 error(s) found on 995 messages

Device ID = 22f0 (Door sensor)
  - 2018-08-30 19:02:38.074 LastSQN =  30 ==> 2018-08-30 19:07:47.727 SQN =  33
  ==> 1 error(s) found on 247 messages

Device ID = 80c2 (Door sensor)
  - 2018-08-31 02:18:20.102 LastSQN =  70 ==> 2018-08-31 03:59:07.266 SQN =  72
  - 2018-08-31 17:07:27.251 LastSQN =  92 ==> 2018-08-31 18:48:02.636 SQN =  94
  - 2018-09-03 15:19:50.372 LastSQN = 192 ==> 2018-09-03 17:00:08.331 SQN = 194
  - 2018-09-03 17:12:02.045 LastSQN = 196 ==> 2018-09-03 17:12:02.897 SQN = 198
  - 2018-09-03 17:12:02.897 LastSQN = 198 ==> 2018-09-03 17:12:04.649 SQN = 200
  - 2018-09-03 17:12:37.743 LastSQN = 203 ==> 2018-09-03 17:12:41.503 SQN = 205
  - 2018-09-04 13:17:16.923 LastSQN = 229 ==> 2018-09-04 14:57:29.890 SQN = 231
  - 2018-09-04 21:28:49.608 LastSQN = 240 ==> 2018-09-04 23:09:31.882 SQN = 242
  - 2018-09-05 00:50:00.019 LastSQN = 244 ==> 2018-09-05 02:30:33.792 SQN = 247
  - 2018-09-05 20:04:10.132 LastSQN =  13 ==> 2018-09-05 21:44:50.534 SQN =  15
  - 2018-09-06 02:46:08.508 LastSQN =  21 ==> 2018-09-06 04:26:49.435 SQN =  23
  ==> 11 error(s) found on 210 messages

Device ID = f64b (Door sensor)
  - 2018-09-03 15:11:19.604 LastSQN =  43 ==> 2018-09-03 16:51:35.004 SQN =  45
  ==> 1 error(s) found on 189 messages

Device ID = 6a4c (Door sensor)
  - 2018-08-31 01:02:36.370 LastSQN = 203 ==> 2018-08-31 02:43:02.185 SQN = 205
  - 2018-08-31 03:33:28.273 LastSQN = 206 ==> 2018-08-31 05:13:53.854 SQN = 208
  - 2018-09-04 12:13:55.200 LastSQN =  85 ==> 2018-09-04 14:44:30.267 SQN =  88
  - 2018-09-04 21:26:08.317 LastSQN =  97 ==> 2018-09-04 23:06:41.220 SQN =  99
  - 2018-09-05 03:17:55.073 LastSQN = 104 ==> 2018-09-05 04:58:22.321 SQN = 106
  ==> 5 error(s) found on 198 messages

Is it normal ? Is it soft's mistake or data transmission mistake ?

thiklop, 2018-09-06:
Hi,

Do you have any crc error in your log ?

ricky074, 2018-09-06:
Hi,
No CRC error in  my log file.

Smanar, 2018-09-07:
I think the prb is hardware.
For the first device you have 2 command with the same SQN

> 2018-09-03 15:50:05.576  (Zigate) Decode8102 : Attribute Report from 63a3 SQN = ef ClusterID = 0403 AttrID = 0014 Attribute Data = ff
> 2018-09-03 15:50:05.581  (Zigate) Decode8102 : Attribute Report from 63a3 SQN = ef ClusterID = 0403 AttrID = 0010 Attribute Data = 2554

And you go ef > f1

For the second device you have probably a disconnection during this phase

> 018-08-30 19:02:38.074  (Zigate) Decode8102 : Attribute Report from 22f0 SQN = 1e ClusterID = 0006 AttrID = 0000 Attribute Data = 00
> 2018-08-30 19:02:38.113  (Zigate) Update v2 0:'Closed' (Garage)
> 2018-08-30 19:03:10.871  (Zigate) Creation process for 22f0 Info: {'Status': '004d', 'ZDeviceID': {}, 'RSSI': 114, 'Model': {}, 'RIA': '24', 'Heartbeat': '1', 'IEEE': {}, 'Ep': {'01': {'0006': '00', '0000': {}}}, 'Battery': 152, 'Type': {}, 'MacCapa': {}, 'ProfileID': {}}
> 2018-08-30 19:04:40.933  (Zigate) Creation process for 22f0 Info: {'Status': '004d', 'ZDeviceID': {}, 'RSSI': 114, 'Model': {}, 'RIA': '24', 'Heartbeat': '1', 'IEEE': {}, 'Ep': {'01': {'0006': '00', '0000': {}}}, 'Battery': 152, 'Type': {}, 'MacCapa': {}, 'ProfileID': {}}
> 2018-08-30 19:06:10.967  (Zigate) Creation process for 22f0 Info: {'Status': '004d', 'ZDeviceID': {}, 'RSSI': 114, 'Model': {}, 'RIA': '24', 'Heartbeat': '1', 'IEEE': {}, 'Ep': {'01': {'0006': '00', '0000': {}}}, 'Battery': 152, 'Type': {}, 'MacCapa': {}, 'ProfileID': {}}
> 2018-08-30 19:06:51.090  Status: (Zigate) Decode8009: Network state UP - PAN Id = d773 on Channel = 0b
> 2018-08-30 19:07:40.976  (Zigate) Creation process for 22f0 Info: {'Status': '004d', 'ZDeviceID': {}, 'RSSI': 114, 'Model': {}, 'RIA': '24', 'Heartbeat': '1', 'IEEE': {}, 'Ep': {'01': {'0006': '00', '0000': {}}}, 'Battery': 152, 'Type': {}, 'MacCapa': {}, 'ProfileID': {}}
> 2018-08-30 19:07:47.727  (Zigate) Decode8102 : Attribute Report from 22f0 SQN = 21 ClusterID = 0006 AttrID = 0000 Attribute Data = 01

pipiche38, 2018-10-02:
https://github.com/fairecasoimeme/ZiGate/issues/38

pipiche38, 2019-01-06:
Linked to Zigate implementation.
I'm closing for now.

closed 2019-01-06 by pipiche38