zigbeefordomoticz/Domoticz-Zigbee #159: Xcube producing "unknown message

pipiche38, 2018-09-06:
When playing with an Xcube and generating horinzotal rotation, there are message logged 

`(Zigate) ReadCluster - ClusterID=000c - unknown message - SAddr = df07 EP = 03 MsgAttrID = 0055 Value = c317d47b`

From the investigation it looks like when you are doing an horizontal rotation  clock or reverse, you get 2 messages : ( https://zigate.fr/xiaomi-magic-cube-cluster/ ). For now when receiving 000C/FF05 with update as a Rotation and we just log the 2nd message in the log.

- ClusterID=000C with Attribute ff05 on Ep 03 
- ClusterID=000C with Attribute 0055 on Ep 03

Smanar, 2018-09-06:
And you haven't different values according to the angle ?

pipiche38, 2018-09-06:
Might be , but for now I think this is enough. You get a rotation. Capturing clock or reverse might be a good one.

Smanar, 2018-09-18:
Cluster 000c/Attribut 0055 > angle = str(struct.unpack('f',struct.pack('I',int(MsgClusterData,16)))[0])

closed 2018-10-08 by pipiche38