zigbeefordomoticz/Domoticz-Zigbee #168: the plugin expect unique model definition in DeviceConf.tx

pipiche38, 2018-09-09:
In case you have a duplicate entry in DeviceConf.txt, you get such error in the log
Sep 09 10:33:30 pi3 domoticz[7251]: 2018-09-09 10:33:30.107  Error: (Zigate) 'onStart' failed 'SyntaxError'.
Sep 09 10:33:30 pi3 domoticz[7251]: 2018-09-09 10:33:30.107  Error: (Zigate) ----> Line 666 in '/var/lib/domoticz/plugins/Domoticz-Zigate/plugin.py', function onStart
Sep 09 10:33:30 pi3 domoticz[7251]: 2018-09-09 10:33:30.107  Error: (Zigate) ----> Line 105 in '/var/lib/domoticz/plugins/Domoticz-Zigate/plugin.py', function onStart
Sep 09 10:33:30 pi3 domoticz[7251]: 2018-09-09 10:33:30.109  Status: ASyncSerial: port open! Baud 115200 0

and it reset the DeviceList.txt

Smanar, 2018-09-09:
Are you try with 
tuple = json.loads(str)
instead of
tuple_List = eval(str)

pipiche38, 2018-11-11:
Not a problem
Wont' fix

closed 2018-11-11 by pipiche38