zigbeefordomoticz/Domoticz-Zigbee #169: Extract the manufacturerCode during Pairing process

pipiche38, 2018-09-10:
It would be great to get the manufacturerCode during the pairing/creation process and store it for the corresponding device in DeviceOfList and finally in DeviceList.txt

Smanar, 2018-09-10:
I m justly on it atm. Give me 2/3 days.

Smanar, 2018-09-11:
Are you sure it s the manufacterer code you want ? 
"IKEA of sweden" for exemple ?

pipiche38, 2018-09-11:
If this is what we get, I think this is still an important information from the Device to be stored in the plugin dictionary , no ?

Smanar, 2018-09-11:
As you want, easy to put in place, but we can't display it on domoticz, so it's only for the plugin.
Model/Type is usefull yes, because for exemple for clusters gestion(to differenciate devices) but I don't see where to use manufacture, I think it's more accurate to use model/type instead.

pipiche38, 2018-10-17:
DOne via Node Descriptor request 0x0042/0x8042 

Implemented during IEEE dev and b46f32f4033d394ad4bd7a6b5d4b041b5257c957

closed 2018-10-17 by pipiche38