zigbeefordomoticz/Domoticz-Zigbee #176: Firmware Version is not initialized

pipiche38, 2018-09-11:
It looks like due to a timing issue Firmware version is not initialized.

Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.187  (Zigate) Connected successfully
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.189  Error: (Zigate) Cannot request Get List of Device due to low firmware level
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.290  (Zigate) Decode8000_v2 - MsgData lenght is : 8 out of 8
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.291  (Zigate) Decode8010 - Reception Version list : 0001030d
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.291  Status: (Zigate) Major Version Num: 0001
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.291  Status: (Zigate) Installer Version Number: 030d
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.292  (Zigate) Decode8000_v2 - MsgData lenght is : 8 out of 8
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.293  (Zigate) Decode8000_v2 - MsgData lenght is : 8 out of 8
Sep 11 23:29:33 pi3 domoticz[22896]: 2018-09-11 23:29:33.293  (Zigate) Start Network status : Success

pipiche38, 2018-09-11:
Workaround in place, the request to get the List of Device is always done despite the Firmware version. If the firmware is not capable to handle it, a status KO will be sent, but doesn't affect the plugin

Smanar, 2018-09-12:
For me it's no a problem, because this command is used only at startup.
Not good version > no devicelist > no upddate with her > plugin continu working without problem.

zaraki673, 2018-09-12:
Confirm, I forget to upgrade my ZIGATE and it's working fine

pipiche38, 2018-09-12:
@zaraki673 I beleive you are on 03b or above, which is compatible !

pipiche38, 2018-09-12:
@Smanar as stated in the comment from here https://github.com/sasu-drooz/Domoticz-Zigate/issues/176#issuecomment-420435149

The workaround is in place. 
So if you are with a firmware above 030b, then the devicelist should be get.
However, the result is pending on timing. And it is hard to predict when it will be given.

closed 2018-09-12 by pipiche38