zigbeefordomoticz/Domoticz-Zigbee #177: onMessage error on (re)start

thiklop, 2018-09-12:
When the plugin (re)start, sometimes Domoticz send an error on erro log when executing onMessage function :

(Zigate) Plugin did not stop after 30 seconds, flushing event queue...
2018-09-10 18:27:31.466 Error: (Zigate) 'onMessage' failed 'error':'unpack requires a bytes object of length 6'.
2018-09-10 18:27:31.466 Error: (Zigate) ----> Line 678 in /home/olivier/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-09-10 18:27:31.466 Error: (Zigate) ----> Line 186 in /home/olivier/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage 

NB : line 191 on code-split
`
Zero1, MsgType, Length, ReceivedChecksum = struct.unpack ('>BHHB', BinMsg[0:6])`

pipiche38, 2018-10-08:
any chance to duplicate this ?

pipiche38, 2019-01-14:
I suggest that we close that issue. If it pop-up again, please re-open

closed 2019-01-14 by pipiche38