zigbeefordomoticz/Domoticz-Zigbee #187: Implementing   xiaomi-aqara-vibration-sensor

pipiche38, 2018-09-15:
http://faire-ca-soi-meme.fr/domotique/2018/09/03/test-xiaomi-aqara-vibration-sensor/#Compatibilite_ZiGate

cf. 15a41dd05a36f8a916c1bc45094e83f1b62977c5

pipiche38, 2018-10-03:
Implementation started in z_input as part of Cluster 0x01010 ..

However we need to get the corresponding Model name, what are the available clusters and how do we want to map it into domoticz

pipiche38, 2018-10-11:
Model = lumi.sensor_cube
At least 2 EPs

pipiche38, 2018-11-12:
in progress in 3f30295fca6950e9c8d88171c7d06aa5bf5b8f02

LouiS22, 2018-12-09:
Have one of these sensor, so if you need tests, pls let me know

pipiche38, 2018-12-09:
@LouiS22 Good news.
So what I would suggest:
- you should be on 'stable' branche
- set Log to Verbose
- and then do a pairing 
- then sending me the log with what came 

They are some code already in ...

Attached the log here

pipiche38, 2018-12-09:
@LouiS22 thanks and very sorry , this was a stupid mistake in the code (unable to test it).
Find here attached a fix. Please drop this file (after unziping)
[z_readClusters.py.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2660991/z_readClusters.py.zip)

LouiS22, 2018-12-10:
2018-12-10 18:41:45.802 (Zigate) ReadCluster - ClusterId=0006 - reception General: On/Off: 00
2018-12-10 18:41:45.802 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-10 18:41:46.854 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-10 18:41:46.856 (Zigate) HB count = 1
2018-12-10 18:41:49.813 (Zigate) receiveData - new Data coming
2018-12-10 18:41:49.813 (Zigate) receiveData - MsgType: 8102
2018-12-10 18:41:49.813 (Zigate) ZigateRead - decoded data : 018102000e8442c40e010008000000200001fe5703 lenght : 42
2018-12-10 18:41:49.813 (Zigate) ZigateRead - MsgType: 8102, MsgLength: 000e, MsgCRC: 84, Data: 42c40e010008000000200001fe; RSSI: 57
2018-12-10 18:41:49.813 (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000e8442c40e010008000000200001fe5703
2018-12-10 18:41:49.813 (Zigate) Decode8102 - Individual Attribute response : [c40e:01] ClusterID: 0008 AttributeID: 0000 Status: 00 Type: 20 Size: 0001 ClusterData: >fe<
2018-12-10 18:41:49.813 (Zigate) DeviceExist - Found in ListOfDevices with status = inDB
2018-12-10 18:41:49.813 (Zigate) Decode8102 : Attribute Report from c40e SQN = 42 ClusterID = 0008 AttrID = 0000 Attribute Data = fe
2018-12-10 18:41:49.814 (Zigate) Device : c40e MacCapa : 8e updating SQN to 42
2018-12-10 18:41:49.814 (Zigate) ReadCluster - MsgData lenght is: 26 out of 24+
2018-12-10 18:41:49.814 (Zigate) DeviceExist - Found in ListOfDevices with status = inDB
2018-12-10 18:41:49.814 (Zigate) ReadCluster - ClusterId=0008 - Level Control: fe
2018-12-10 18:41:49.814 (Zigate) MajDomoDevice - Device ID : 90fd9ffffe29a333 - Device EP : 01 - Type : 0008 - Value : fe - Hue :
2018-12-10 18:41:49.814 (Zigate) MajDomoDevice - Type = LvlControl
2018-12-10 18:41:49.814 (Zigate) MajDomoDevice - NWKID = c40e IEEE = 90fd9ffffe29a333 Unit = 6335
2018-12-10 18:41:49.814 (Zigate) MajDomoDevice - {'0000': {}, '0008': {}, '1000': {}, '0006': '00', '0003': {}, 'Type': 'LvlControl', '0004': {}, '0005': {}, '0b05': {}, 'ClusterType': {'6335': 'LvlControl'}}
2018-12-10 18:41:49.815 (Zigate) MajDomoDevice - We have 1 EPs with ClusterType
2018-12-10 18:41:49.815 (Zigate) MajDomoDevice - NWKID: c40e SwitchType: 7, DeviceType: LvlControl, ClusterType: LvlControl, old_nVal: 0 , old_sVal: Off
2018-12-10 18:41:49.815 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-10 18:41:54.162 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-10 18:41:54.164 (Zigate) HB count = 2
2018-12-10 18:41:55.722 (Zigate) receiveData - new Data coming
2018-12-10 18:41:55.722 (Zigate) receiveData - MsgType: 8102
2018-12-10 18:41:55.722 (Zigate) ZigateRead - decoded data : 018102001f4305ee9f0100000005004200126c756d692e766962726174696f6e2e617131b103 lenght : 76
2018-12-10 18:41:55.722 (Zigate) ZigateRead - MsgType: 8102, MsgLength: 001f, MsgCRC: 43, Data: 05ee9f0100000005004200126c756d692e766962726174696f6e2e617131; RSSI: b1
2018-12-10 18:41:55.722 (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102001f4305ee9f0100000005004200126c756d692e766962726174696f6e2e617131b103
2018-12-10 18:41:55.722 (Zigate) Decode8102 - Individual Attribute response : [ee9f:01] ClusterID: 0000 AttributeID: 0005 Status: 00 Type: 42 Size: 0012 ClusterData: >6c756d692e766962726174696f6e2e617131<
2018-12-10 18:41:55.723 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-10 18:41:55.735 (Zigate) receiveData - new Data coming
2018-12-10 18:41:55.735 (Zigate) receiveData - MsgType: 8102
2018-12-10 18:41:55.735 (Zigate) ZigateRead - decoded data : 018102003bb505ee9f010000ff010042002e0121c70b0328260421a81305210a0006240000000000082108030a21000098212800992100009a251000e2ffbe04b103 lenght : 132
2018-12-10 18:41:55.735 (Zigate) ZigateRead - MsgType: 8102, MsgLength: 003b, MsgCRC: b5, Data: 05ee9f010000ff010042002e0121c70b0328260421a81305210a0006240000000000082108030a21000098212800992100009a251000e2ffbe04; RSSI: b1
2018-12-10 18:41:55.735 (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102003bb505ee9f010000ff010042002e0121c70b0328260421a81305210a0006240000000000082108030a21000098212800992100009a251000e2ffbe04b103
2018-12-10 18:41:55.735 (Zigate) Decode8102 - Individual Attribute response : [ee9f:01] ClusterID: 0000 AttributeID: ff01 Status: 00 Type: 42 Size: 002e ClusterData: >0121c70b0328260421a81305210a0006240000000000082108030a21000098212800992100009a251000e2ffbe04<
2018-12-10 18:41:55.736 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-10 18:41:55.723 Error: (Zigate) Decode8102 - Receiving a message from unknown device : ee9f with Data : 05ee9f0100000005004200126c756d692e766962726174696f6e2e617131
2018-12-10 18:41:55.735 Error: (Zigate) Decode8102 - Receiving a message from unknown device : ee9f with Data : 05ee9f010000ff010042002e0121c70b0328260421a81305210a0006240000000000082108030a21000098212800992100009a251000e2ffbe04
2018-12-10 18:41:56.493 (Zigate) receiveData - new Data coming
2018-12-10 18:41:56.493 (Zigate) receiveData - MsgType: 8102
2018-12-10 18:41:56.493 (Zigate) ZigateRead - decoded data : 018102001f4506ee9f0100000005004200126c756d692e766962726174696f6e2e617131b403 lenght : 76
2018-12-10 18:41:56.493 (Zigate) ZigateRead - MsgType: 8102, MsgLength: 001f, MsgCRC: 45, Data: 06ee9f0100000005004200126c756d692e766962726174696f6e2e617131; RSSI: b4
[... cut ...]
2018-12-10 18:41:55.735 Error: (Zigate) Decode8102 - Receiving a message from unknown device : ee9f with Data : 05ee9f010000ff010042002e0121c70b0328260421a81305210a0006240000000000082108030a21000098212800992100009a251000e2ffbe04
2018-12-10 18:41:56.493 Error: (Zigate) Decode8102 - Receiving a message from unknown device : ee9f with Data : 06ee9f0100000005004200126c756d692e766962726174696f6e2e617131
2018-12-10 18:41:56.506 Error: (Zigate) Decode8102 - Receiving a message from unknown device : ee9f with Data : 06ee9f010000ff010042002e0121c70b0328260421a81305210a0006240000000000082108030a21000098212800992100009a251000e2ffbe04

pipiche38, 2018-12-10:
@LouiS22 , I'm reachable on Keybase.io (you can see my username on Github). Might be easier to chart ;-)

The error message you have, says that the device is sending messages, but is not known by the plugin. 

So basically device 'ee9f' doesn't have any entry in DeviceList-xx.txt .

LouiS22, 2018-12-10:
> @LouiS22 , I'm reachable on Keybase.io (you can see my username on Github). Might be easier to chart ;-)
> 
> The error message you have, says that the device is sending messages, but is not known by the plugin.
> 
> So basically device 'ee9f' doesn't have any entry in DeviceList-xx.txt .

Installed and sent you a PM via keybase :)

pipiche38, 2018-12-11:
Firs implementation in dev- 6a9f8b73943a1c0eab49919c14eeed7c8289b3f6

pipiche38, 2018-12-15:
@LouiS22 could you kindly confirm if this is work ?

LouiS22, 2018-12-15:
Looks fine!

pipiche38, 2018-12-15:
Thanks we close the issue

closed 2018-12-15 by pipiche38