zigbeefordomoticz/Domoticz-Zigbee #188: Implement ZigBee group to make Ikea Tradfri command working

pipiche38, 2018-09-15:
Based on the wokr done by Akila, it might be good to think an implement a way to group ZigBee devices under a group. This would help to make use of the Ikea Tradfri command

http://faire-ca-soi-meme.fr/domotique/2018/03/14/une-telecommande-zigbee-pour-les-controler-tous/

thiklop, 2018-10-08:
I bought one some days ago, how can I help ?

Smanar, 2018-10-08:
The big problem for me is to find how to implement this in domoticz ...
In other application why not, but the Domoticz  UI is too poor to make a system for group.

Or the user need to make all the procedure itself, outside Domoticz, but after ... We can read group from device, but I don't see how to attribute an action to a group in Domoticz.

pipiche38, 2018-10-08:
Open-Zwave does outside of Domoticz as part of there plugin.

Smanar, 2018-10-08:
Yep, but it's not a python plugin (coded in domoticz core) ?
For exemple, they have button on configuration hardware panel, IDK if it's possible with python plugin.

We can use the WWW folder for special page, but for you it will be hard to implement that in plugin update process.

pipiche38, 2018-10-09:
I think that is much better to do it from the plugin.

This might be a 2 steps approach.(1) configuring the grouping, (2) UI to handle the group.

For (1) we might use An approach with a conf file as we have DeviceConf , we might have DeviceGroup where we could describe the groups .

Then the group could be created in the context of the Ikea command. So we can identify the list of devices in one group and which device manage the group..

This needs further thoughts but could be a way forward

pipiche38, 2018-10-09:
@Smanar 

> Yep, but it's not a python plugin (coded in domoticz core) ?
> For exemple, they have button on configuration hardware panel, IDK if it's possible with python plugin.

I think for Open-Zwave it is even a bit more complex.  Open-Zwave have there own application, but they can also be integrated into Domoticz and indeed comes with a lot of UI like the Network Topology, The group/Scene.

For having OZW devices @Home, what I see on my domoticz folder, is several XML files related to OWZ:
- zwscene.xml and ozwcp.scenes.XXXXX which I beleive are for Scene configuration.

So as mentioned in the previous comment, I think we should decouple the configuration to the UI part and may be focus more on the configuration aspects.

Smanar, 2018-10-09:
The configuration aspect is the bigger problem for me.
I have searching an user friendly approch without suceed, the better way I have found is custom page and use the web server from domoticz, but IDK if I can interact directly with the zigate if there is already connected to another application.
Conf file is a good idea but it s more for power user for me. And I don't have idea for rest, create a fake device for a designed group ? And if I want to remove a device from a group ....

pipiche38, 2018-10-12:
Interesting idea 
https://github.com/doudz/zigate/issues/29

deennoo, 2018-10-25:
I'm using Ikea plugin for Ikea gateway on domoticz.

When groupe are create on Ikea App this plugin juste create a dimmer Switch.

As nos de Can read cluster, it Can be done thé same way.

Parse groupe id for each device (bulb, remote, plug)

And create it.

For adding device to a group : using commande Line on plugin conf like it's done for rflink. This need to create a simple dictionnaire for élémentaire action 

Ex : addgroupe xx device yyy

pipiche38, 2018-12-09:
We will be able to re-use the group created via the remote controller, but the remote control so far cannot be use by Zigate

pipiche38, 2018-12-10:
As of known, we can use the Ikea Remote Command to create a group of ZLL. 
During the discovery phase, the plugin will detect the group created by the remote control and will create a widget in Domoticz. So it will be possible to manipulate the Group from Domoticz or from the Remote Control.

https://github.com/fairecasoimeme/ZiGate/issues/64
https://github.com/fairecasoimeme/ZiGate/issues/6

Implemented in dev-grpmgt

closed 2018-12-10 by pipiche38