zigbeefordomoticz/Domoticz-Zigbee #191: Update on windows issue

seblang, 2018-09-19:
Hello I want to do an update at last version without to lost all my materials but I am under windows. I have done a full copy from zip file but with this solution I have lost all device! What file a have to copy to have the last version and keep all my device? Regards

zaraki673, 2018-09-19:
Hi, you have to keep your devicelist.txt.
Don't know if it will work without creating new device in Domoticz, for that you have to edit devicelist.txt and replace status ='inDb' with status ='' for all devices

pipiche38, 2018-09-20:
First , you need to have a backup of the Domoticz database and then the plugin informations as mentionned by @zaraki673 which is the DeviceList.txt

If you have lost all devices, I'm afraid this is more on the Domoticz side, rather than the Zigate plugin one.

pipiche38, 2018-09-20:
Also , make sure that you have the latest version 3.0.3. We have fixed an issue 2 days ago. And with that fix, if you have the Domoticz database, I guess that you can restart and DeviceLIst.txt will be re-populated .

pipiche38, 2018-10-02:
@seblang is the problem solved ?

pipiche38, 2018-11-11:
Please re-open the issue if the problem is still pending

closed 2018-11-11 by pipiche38