zigbeefordomoticz/Domoticz-Zigbee #207: Xiaomi Aqara cube recognized by zigate but cannot trigger http request

nx345, 2018-10-07:
When allowing new device, the aqara cube is recognized by zigate. All movements (shaking, 90 degrees, 180 degrees, and so on) are working well and appeared in the log. 

But when adding a http request for a movement (for instance flip 90 degres) and saving this under domoticz. Even before triggering this movement, I got this error just by saving the http request wanted : 

2018-10-07 12:56:32.631 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-10-07 12:56:32.631 Error: (zigate) ----> Line 280 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-10-07 12:56:32.631 Error: (zigate) ----> Line 242 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-10-07 12:56:32.631 Error: (zigate) ----> Line 427 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function ResetDevice

pipiche38, 2018-10-07:
@nx345 which plugin branch are you ? Master or Beta ?
Identify what are 
IDX of your Zigate Hardware in the Hardware list
IDX of your Cube device in the Device list

and do a sqllite3 query like that one : 
select DeviceID, Name, Options  from DeviceStatus where HardwareID=3 and ID=541;

nx345, 2018-10-07:
@pipiche38 Found out how to do the query here is the output : 
d3a9|Cube|LevelActions:fHx8fHx8fHx8;LevelNames:T2ZmfFNoYWtlfFdha2V1cHxEcm9wfDkwwrB8MTgwwrB8UHVzaHxUYXB8Um90YXRpb24=;LevelOffHidden:dHJ1ZQ==;SelectorStyle:MA==;TypeName:QXFhcmE=;Zigate:eydSU1NJJzogMjQzLCAnWkRldmljZUlEJzoge30sICdUeXBlJzogJ0FxYXJhJywgJ1N0YXR1cyc6ICdpbkRCJywgJ1JJQSc6ICczJywgJ01hY0NhcGEnOiAnODAnLCAnQmF0dGVyeSc6IHt9LCAnSGVhcnRiZWF0JzogJzAnLCAnTW9kZWwnOiAnbHVtaS5zZW5zb3JfY3ViZS5hcWdsMDEnLCAnSUVFRSc6ICcwMDE1OGQwMDAyNzk4NGFlJywgJ1Byb2ZpbGVJRCc6IHt9LCAnRXAnOiB7JzAxJzogeycwMDAwJzoge30sICcwMDBjJzoge30sICcwMDEyJzoge319fX0=

pipiche38, 2018-10-07:
Ok. I need to see how to decode those datas and to check if we have TypeName in the Zigate dict.
I suspect not ...

pipiche38, 2018-10-07:
@zaraki673 it is simply base64 ... why I didn't think about :-(

Zigate: {'RSSI': 243, 'ZDeviceID': {}, 'Type': 'Aqara', 'Status': 'inDB', 'RIA': '3', 'MacCapa': '80', 'Battery': {}, 'Heartbeat': '0', 'Model': 'lumi.sensor_cube.aqgl01', 'IEEE': '00158d00027984ae', 'ProfileID': {}, 'Ep': {'01': {'0000': {}, '000c': {}, '0012': {}}}}

So line 427 in z_domoticz.py (of latest version ) is the following
`Dtypename=DOptions['TypeName']`

pipiche38, 2018-10-07:
@nx345 I beleive you have one Device where the TypeName is not correct.

Could you do : 
select DeviceID, Name, Options from DeviceStatus where HardwareID=12;

and copy paste the result. I suspect that you have One device, where the Options is not correct

pipiche38, 2018-10-07:
Thanks. Nothing bad in this output. All TypeName are good and the Zigate field also !

'''
{'ZDeviceID': '0100', 'ProfileID': 'c05e', 'RIA': '0', 'MacCapa': '8e', 'Model': 'Ampoule.LED1622G12.Tradfri', 'Status': 'inDB', 'Ep': {'01': {'0003': {}, '0004': {}, '0006': {}, '0005': {}, '0000': {}, '1000': {}, '0b05': {}, '0008': {}}}, 'Battery': {}, 'Heartbeat': '1', 'IEEE': '000b57fffee3ec43', 'RSSI': {}, 'Type': {}}

{'RSSI': {}, 'ZDeviceID': '0100', 'Type': {}, 'Status': 'inDB', 'RIA': '0', 'MacCapa': '8e', 'Battery': {}, 'Heartbeat': '1', 'Model': 'Ampoule.LED1622G12.Tradfri', 'IEEE': '000b57fffeda82f2', 'ProfileID': 'c05e', 'Ep': {'01': {'0000': {}, '0005': {}, '0008': {}, '0b05': {}, '1000': {}, '0006': {}, '0004': {}, '0003': {}}}}

{'ZDeviceID': '0100', 'ProfileID': 'c05e', 'RIA': '0', 'MacCapa': '8e', 'Model': 'Ampoule.LED1622G12.Tradfri', 'Status': 'inDB', 'Ep': {'01': {'0003': {}, '0004': {}, '0006': {}, '0005': {}, '0000': {}, '1000': {}, '0b05': {}, '0008': {}}}, 'Battery': {}, 'Heartbeat': '1', 'IEEE': '000b57fffeb99dd7', 'RSSI': {}, 'Type': {}}

{'RSSI': 243, 'ZDeviceID': {}, 'Type': 'Aqara', 'Status': 'inDB', 'RIA': '3', 'MacCapa': '80', 'Battery': {}, 'Heartbeat': '0', 'Model': 'lumi.sensor_cube.aqgl01', 'IEEE': '00158d00027984ae', 'ProfileID': {}, 

'''

nx345, 2018-10-07:
When adding the cube device it is okay. But when you add something in the action field of the cube and especially when you save it, the error appears.

pipiche38, 2018-10-07:
I duplicate the problem !

pipiche38, 2018-10-07:
Duplicated on Dev and Beta branch.

pipiche38, 2018-10-07:
I feel it is a bug in Domoticz, which is overwritting the Zigate and ClusterType fields !

pipiche38, 2018-10-07:
@nx345 well done !

https://github.com/domoticz/domoticz/issues/2763

They need to fix the issue on Domoticz. We cannot do more on our side !

pipiche38, 2018-10-07:
In between I suggest to use LUA script script_device_xxxx to manage device notifications

nx345, 2018-10-07:
> In between I suggest to use LUA script script_device_xxxx to manage device notifications

I was thinking of the same solution. Nevertheless, typing the path script in the cube menu will return the same error. The only way to bypass this bug is to launch a script that is referring to the state of the cube.

pipiche38, 2018-10-07:
> I was thinking of the same solution. Nevertheless, typing the path script in the cube menu will return the same error. The only way to bypass this bug is to launch a script that is referring to the state of the cube.

Not sure I understood. I'm suggesting a script under script/lua/script_device_xxxx. 
I'm sure that inside the script which is going to be executed everytime there is an update to the Cube/Device, you can inside the script retreive value

otherdevices_lastupdate['Aqara'] give you the time when the switch name 'Aqara' has been touch
otherdevices_svalues['Aqara'] give you the Value ( 10, 20,30, 40, 50,60 ...)

pipiche38, 2018-10-07:
/var/lib/domoticz/scripts/lua/script_device_aqara.lua

```
commandArray = {}

if devicechanged['Zigate - Aqara-7b6a-01']  then
    print( 'Date when update : ' .. otherdevices_lastupdate['Zigate - Aqara-7b6a-01'] )
    print( 'Action       : ' .. devicechanged['Zigate - Aqara-7b6a-01'] )
    print( 'Action Level : ' ..otherdevices_svalues['Zigate - Aqara-7b6a-01'] )
end

return commandArray
```

nx345, 2018-10-07:
> /var/lib/domoticz/scripts/lua/script_device_aqara.lua
>
> ```
> commandArray = {}
>
> if devicechanged['Zigate - Aqara-7b6a-01']  then
>     print( 'Date when update : ' .. otherdevices_lastupdate['Zigate - [...]

Thank you my friend I will try it asap

pipiche38, 2018-10-08:
Good news @nx345 , but still we have a potential issue which is that Domoticz override the Zigate information, so we will park this issue until Domoticz solve their problem : domoticz/domoticz#2763

But the workaround to use lua script seems to be a good alternative !

nx345, 2018-10-08:
> Good news @nx345 , but still we have a potential issue which is that Domoticz override the Zigate information, so we will park this issue until Domoticz solve their problem : [...]

Of course, but I don't think that it will be solved until the next stable version of domoticz (meaning next year in july).

pipiche38, 2018-10-08:
Domoticz considers that we **must not** use Options to store plugin information.
Bug on our side 

Will be track against #217

closed 2018-10-08 by pipiche38

nx345, 2018-10-08:
> Domoticz considers that we **must not** use Options to store plugin information.
> Bug on our side
> 
> Will be track against #217

Good news then ! Even letting a blank option in the cube and saving it leads to an error.

pipiche38, 2018-10-08:
Yes, because it has override the Plugin information ! So you must re-create the Device :-(