zigbeefordomoticz/Domoticz-Zigbee #229: 3.0.0 or beta-3.1.0 / Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.

lboue, 2018-10-11:
**Describe the bug**
I tried to upgrade Domoticz-Zigate plugin versions Version 3.0.0 and beta-3.1.0.
I have Domoticz V4.10059.

I am getting many time the following error:
`2018-10-11 20:12:22.413  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.`

**To Reproduce**
Steps to reproduce the behavior:
1. Install Domoticz V4.100
2. Install  Domoticz-Zigate plugin  beta-3.1.0 or  3.0.0
3. Start Domoticz service
4. See error

`2018-10-11 20:12:22.413  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.`

**Expected behavior**
No error.

**Screenshots**
```
cd domoticz/plugins/
cd Domoticz-Zigate/
mv Domoticz-Zigate/ /tmp/
wget https://github.com/sasu-drooz/Domoticz-Zigate/archive/beta-3.1.0.zip
unzip beta-3.1.0.zip 

pi@raspberrypi:~/domoticz $ sudo service domoticz restart

sudo tailf /tmp/domoticz.txt
2018-10-11 20:12:01.561  Status: Domoticz V4.10059 (c)2012-2018 GizMoCuz
2018-10-11 20:12:01.561  Status: Build Hash: 06741959, Date: 2018-10-11 08:59:01
2018-10-11 20:12:01.562  Status: Startup Path: /home/pi/domoticz/
2018-10-11 20:12:01.603  Sunrise: 08:16:00 SunSet: 19:23:00
2018-10-11 20:12:01.603  Day length: 11:08:00 Sun at south: 13:05:00
2018-10-11 20:12:01.603  Civil twilight start: 07:45:00 Civil twilight end: 19:54:00
2018-10-11 20:12:01.603  Nautical twilight start: 07:09:00 Nautical twilight end: 20:30:00
2018-10-11 20:12:01.603  Astronomical twilight start: 06:34:00 Astronomical twilight end: 21:05:00
2018-10-11 20:12:01.603  Status: EventSystem: reset all events...
2018-10-11 20:12:01.781  Status: PluginSystem: Started, Python version '3.5.3'.
2018-10-11 20:12:01.792  Active notification Subsystems: email, pushover (2/13)
2018-10-11 20:12:01.794  Status: WebServer(HTTP) started on address: :: with port 8080
2018-10-11 20:12:01.806  Status: WebServer(SSL) started on address: :: with port 8443
2018-10-11 20:12:01.809  Status: Proxymanager started.
2018-10-11 20:12:01.880  Starting shared server on: :::6144
2018-10-11 20:12:01.881  Status: TCPServer: shared server started...
2018-10-11 20:12:01.881  Status: RxQueue: queue worker started...
2018-10-11 20:12:03.246  Status: PROXY: Authenticate result: success.
2018-10-11 20:12:03.882  Status: Wunderground: Worker started...
2018-10-11 20:12:03.883  Status: XiaomiGateway (ID=10): Delaying worker startup...
2018-10-11 20:12:08.883  Status: XiaomiGateway (ID=10): Selected as main Gateway
2018-10-11 20:12:08.884  Status: XiaomiGateway (ID=10): Worker started...
2018-10-11 20:12:08.884  Status: (Zigate) Started.
2018-10-11 20:12:08.884  Status: Netatmo: Worker started...
2018-10-11 20:12:08.885  Status: (Pots_Fleurs) Started.
2018-10-11 20:12:08.886  Status: (ECO-DEVICES): Worker started...
2018-10-11 20:12:08.886  Status: EventSystem: reset all events...
2018-10-11 20:12:08.887  Status: EventSystem: reset all device statuses...
2018-10-11 20:12:09.106  Status: PluginSystem: Entering work loop.
2018-10-11 20:12:09.174  Status: Python EventSystem: Initalizing event module.
2018-10-11 20:12:09.180  Status: EventSystem: Started
2018-10-11 20:12:09.181  Status: EventSystem: Queue thread started...
2018-10-11 20:12:09.397  Status: (Zigate) Entering work loop.
2018-10-11 20:12:09.397  Status: (Zigate) Initialized version 3.9.99999, author 'zaraki673  pipiche38'
2018-10-11 20:12:09.624  Status: (Pots_Fleurs) Entering work loop.
2018-10-11 20:12:09.624  Status: (Pots_Fleurs) Initialized version 1.0.0, author 'blauwebuis'
2018-10-11 20:12:09.628  Status: (Zigate) onStart called - Zigate plugin V 3.9.999 (dev branch)
2018-10-11 20:12:09.633  Status: (Zigate) Zigate Structure V2
2018-10-11 20:12:09.633  Error: (Zigate) 'onStart' failed 'ValueError':'invalid literal for int() with base 10: 'Debug''.
2018-10-11 20:12:09.633  Error: (Zigate) ----> Line 302 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onStart
2018-10-11 20:12:09.633  Error: (Zigate) ----> Line 75 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onStart
2018-10-11 20:12:22.413  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:12:22.413  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:12:22.413  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:12:23.207  Status: Incoming connection from: 192.168.1.59
2018-10-11 20:12:32.383  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:12:32.383  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:12:32.384  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:12:39.162  Status: Incoming connection from: 127.0.0.1
2018-10-11 20:12:42.354  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:12:42.354  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:12:42.354  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:12:52.374  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:12:52.375  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:12:52.375  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:02.347  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:13:02.348  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:02.348  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:12.369  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:13:12.369  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:12.370  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:22.388  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:13:22.389  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:22.389  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:32.358  Error: (Zigate) 'onHeartbeat' failed 'AttributeError':''str' object has no attribute 'Connected''.
2018-10-11 20:13:32.358  Error: (Zigate) ----> Line 330 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
2018-10-11 20:13:32.358  Error: (Zigate) ----> Line 291 in '/home/pi/domoticz/plugins/Domoticz-Zigate-beta-3.1.0/plugin.py', function onHeartbeat
```

**Plugin (please complete the following information):**
 - OS: Raspbian GNU/Linux 9
 - Plugin version: beta-3.1.0
 - Plugin branch: beta-3.1.0

**Additional context**
Add any other context about the problem here.

pipiche38, 2018-10-11:
Please do consider the Beta as unstable and make sure that you have a proper backup of the Domoticz.db and DeviceList.txt.

could you kindly cleanup the python cache after having stop domoticz service

rm Domoticz-Zigate/__pycache__/*

and then restart it.

pipiche38, 2018-10-12:
Excellent, was it the removal of __pycache__ doing the trick ? or did you do something else ?