zigbeefordomoticz/Domoticz-Zigbee #233: Double Order

deennoo, 2018-10-12:
When pressing button order are recorded twice : 

2018-10-13 01:43:15 | 1 Click
-- | --
2018-10-13 01:43:15 |  
2018-10-13 01:43:12 | 1 Click
2018-10-13 01:43:12 |  
2018-10-13 01:43:07 | 3 Click
2018-10-13 01:43:04 | 1 Click
2018-10-13 01:43:04 |  
2018-10-13 01:18:32 | 4 Click
2018-10-13 01:16:56 | 1 Click
2018-10-13 01:16:56 |  
2018-10-13 01:15:49 | 1 Click
2018-10-13 01:15:49 |  
2018-10-13 01:12:04 | 1 Click
2018-10-13 01:12:04 |  
2018-10-13 01:11:46 | 1 Click
2018-10-13 01:11:46 |  
2018-10-13 01:11:38 | 1 Click
2018-10-13 01:11:38 |  
2018-10-13 01:11:30 | 1 Click
2018-10-13 01:11:30 |  
2018-10-13 01:09:30 | 1 Click
2018-10-13 01:09:30 |  
2018-10-13 01:07:34 | 3 Click
2018-10-13 01:07:23 | 2 Click
2018-10-13 01:07:09 | 1 Click

When use it on simple event like : 

"if devicechanged['Zigate - MSwitch-8af6-01'] then
    commandArray['Zigate - Plug-143f-09']='Toggle'
end"

1 Click and blank one trigger event

This only happenning on 1 Click

Using Domoticz 

Version: 4.9701
Build Hash: b47a877f
Compile Date: 2018-06-23 16:27:56
dzVents Version: 2.4.6
Python Version: 3.5.2 (default, Jun 26 2018, 20:36:17) [GCC 4.9.2]

Zigate FW 3.0d & Xiaomi AQ2 Switch (square one) and Salus wall plug

pipiche38, 2018-10-13:
On which branch ? Master ?

deennoo, 2018-10-13:
Using master

pipiche38, 2018-10-13:
@deennoo do you have the possibility to enable Debug (just Python Only) in the hardware, to see exactly what is happening ?

pipiche38, 2018-10-14:
@deennoo could you let me know what should be exepcted ?

Here from the log I see 

Off (00) and then 1:10 ... Where is the double Order ?

pipiche38, 2018-10-15:
@deennoo : so the right Selector must be : 
"SelectorStyle:0;LevelNames:Off|Switch 1|Switch 2|Both_Click"

Now, what I'm not sure is when looking at the log, this is the Device itself sending the Off and then the On, so I don't think we can do mcuh on that !

pipiche38, 2018-10-22:
Could you configure that one is the Type = MSwitch

Are they those one ?
'lumi.sensor_switch.aq2':{'Ep':{'01':{'0006'}},'Type':'MSwitch'},
'lumi.sensor_switch':{'Ep':{'01':{'0006'}},'Type':'MSwitch'},

closed 2018-10-25 by deennoo