zigbeefordomoticz/Domoticz-Zigbee #234: Lumi.dwallswitch.fire

deennoo, 2018-10-13:
There some information about Aqara 230 double switch working with Phase and Neutre also know as double fire switch.

2 Switch with power consumtion for each one, simple On/Off

Usefull endpoint are 01 for Left switch and 02 for right one

inter double 230v fire xiaomi

Type: 0x8000 (Status)
  Length: 5
  Status: 0x00 (Success)
  SQN: 0xBB
  Message: 
Type: 0x8045 (Active Endpoints Response)
  SQN: 0xBB
  Status: 0x00
  Short Address: 0xECC8
  Endpoint Count: 4
  Endpoint List: 
    Endpoint 0: 0x01
    Endpoint 1: 0x02
    Endpoint 2: 0x03
    Endpoint 3: 0x04

	
	
	0x01
	Type: 0x8043 (Simple Descriptor Response)
  SQN: 0xBC
  Status: 0x00
  Short Address: 0xECC8
  Length: 30
  EndPoint: 0x01
  Profile ID: 0x0104 (ZigBee HA)
  Device ID: 0x0051 (Generic - Smart Plug)
  Input Cluster Count: 9
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0x0004 (General: Groups)
    Cluster 2:  Cluster ID: 0x0003 (General: Identify)
    Cluster 3:  Cluster ID: 0x0006 (General: On/Off)
    Cluster 4:  Cluster ID: 0x0010 (Unknown)        ---> power consumtion as lumi.plug
    Cluster 5:  Cluster ID: 0x0005 (General: Scenes)
    Cluster 6:  Cluster ID: 0x000A (General: Time)
    Cluster 7:  Cluster ID: 0x0001 (General: Power Config)
    Cluster 8:  Cluster ID: 0x0002 (General: Temperature Config)
  Output Cluster Count: 2
    Cluster 0:  Cluster ID: 0x0019 (General: OTA)
    Cluster 1:  Cluster ID: 0x000A (General: Time)

	
	0x02
	  Length: 5
  Status: 0x00 (Success)
  SQN: 0xBE
  Message: 
Type: 0x8043 (Simple Descriptor Response)
  SQN: 0xBE
  Status: 0x00
  Short Address: 0xECC8
  Length: 12
  EndPoint: 0x02
  Profile ID: 0x0104 (ZigBee HA)
  Device ID: 0x0051 (Generic - Smart Plug)
  Input Cluster Count: 2
    Cluster 0:  Cluster ID: 0x0006 (General: On/Off)
    Cluster 1:  Cluster ID: 0x0010 (Unknown)  ---> power consumtion as lumi.plug
  Output Cluster Count: 0
  
  0x03
  Type: 0x8000 (Status)
  Length: 5
  Status: 0x00 (Success)
  SQN: 0xBF
  Message: 
Type: 0x8043 (Simple Descriptor Response)
  SQN: 0xBF
  Status: 0x00
  Short Address: 0xECC8
  Length: 14
  EndPoint: 0x03
  Profile ID: 0x0104 (ZigBee HA)
  Device ID: 0x0009 (Generic - Mains Power Outlet)
  Input Cluster Count: 1
    Cluster 0:  Cluster ID: 0x000C (Unknown)
  Output Cluster Count: 2
    Cluster 0:  Cluster ID: 0x000C (Unknown)
    Cluster 1:  Cluster ID: 0x0004 (General: Groups)

	
	0x04
	
Type: 0x8000 (Status)
  Length: 5
  Status: 0x00 (Success)
  SQN: 0xC0
  Message: 
Type: 0x8043 (Simple Descriptor Response)
  SQN: 0xC0
  Status: 0x00
  Short Address: 0xECC8
  Length: 12
  EndPoint: 0x04
  Profile ID: 0x0104 (ZigBee HA)
  Device ID: 0x0053 (Unknown)
  Input Cluster Count: 1
    Cluster 0:  Cluster ID: 0x000C (Unknown)
  Output Cluster Count: 1
    Cluster 0:  Cluster ID: 0x000C (Unknown)
	
	
	
On/off left Button : ep origin 1 ep dest 1
Type: 0x8000 (Status)
  Length: 5
  Status: 0x00 (Success)
  SQN: 0x09
  Message: 
Type: 0x8101 (Default Response)
  SQN: 0x09
  EndPoint: 0x01
  Cluster ID: 0x0006 (General: On/Off)
  Command: 0x01
  Status: 0x00
Type: 0x8102 (Attribute Report)
  SQN: 0x93
  Src Addr: 0xECC8
  Src Ep: 0x01
  Cluster ID: 0x0006 (General: On/Off)
  Attribute ID: 0x0000
  Attribute Size: 0x0001
  Attribute Type: 0x10 (Boolean)
  Attribute Data: 0x01
  Status: 0x00
Type: 0x8102 (Attribute Report)
  SQN: 0x93
  Src Addr: 0xECC8
  Src Ep: 0x01
  Cluster ID: 0x0006 (General: On/Off)
  Attribute ID: 0xF000
  Attribute Size: 0x0004
  Attribute Type: 0x23 (UINT32)
  Attribute Data: 0x07000009
  Status: 0x00
  
  
  On/off right button : ep origin 1  ep dest 2
    Command: 0x01
  Status: 0x00
Type: 0x8102 (Attribute Report)
  SQN: 0x0A
  Src Addr: 0xECC8
  Src Ep: 0x02
  Cluster ID: 0x0006 (General: On/Off)
  Attribute ID: 0x0000
  Attribute Size: 0x0001
  Attribute Type: 0x10 (Boolean)
  Attribute Data: 0x01
  Status: 0x00
Type: 0x8102 (Attribute Report)
  SQN: 0x0A
  Src Addr: 0xECC8
  Src Ep: 0x02
  Cluster ID: 0x0006 (General: On/Off)
  Attribute ID: 0xF000
  Attribute Size: 0x0004
  Attribute Type: 0x23 (UINT32)
  Attribute Data: 0x0700000C
  Status: 0x00

Hopping this help

pipiche38, 2018-10-13:
Do you own one ?
It would be interesting in that case that you pull the 'beta' branch.
(be aware that we will do a DB upgrade and in such you must do a copy of your Domoticz Database before) if you want to come back to Master branch at a time.

There is a parameter in the Beta branch which can allow to track all of the discovery frame in order to integrate much easily the device later on.

deennoo, 2018-10-13:
I'have one before zigate was born, i have some neutral 230v double switch too, and some other brand zigbee devices (Salus full hvac setup, Jiawen bulb and led strip)

I 'll setup a dev station to help you, i start to help at the begining, easy code, easy to follow, but now i have to update my mind to understand how the plugin works and were to act to add some devices for PR

pipiche38, 2018-10-13:
Sound great .... If you want you can setup a Keybase.io in order to chat ... My keybase.io account is available on my github page ;-)

deennoo, 2018-10-15:
As there is 2 diff EP, can't you create 2 simples on off switch ?

pipiche38, 2018-10-16:
Need to think about it ...

pipiche38, 2018-10-24:
@deennoo , could  you kindly confirm that it works on Dev branch and so we can close that one ?
0bc63353a348d8341dc6ad4eddaec4732ab6c307

closed 2018-10-25 by deennoo