zigbeefordomoticz/Domoticz-Zigbee #236: Power metering for Salus Plug

pipiche38, 2018-10-16:
It requires the implementation of cluster 0x0702 /

need to understand how to decode the MsgClusterData

It might also required to send a ReadAttributeRequest for this Cluster on 0x0000 and 0x0400 attribute

pipiche38, 2018-10-16:
@deennoo for follow-up

deennoo, 2018-10-16:
ok thx

pipiche38, 2018-10-17:
First tentative implemented on dev : b350dd3c1a876821d34ef28000606973e095b0df

pipiche38, 2018-10-19:
Main issue is to understand which Attribute the salus plug is storing information
 Instantaneous demand
Current Summation Delivered

pipiche38, 2018-10-22:
ConfigureReporting implemnted let see what is the outcome 38a42a49a5f046e0aeda7539f80adad926032236

pipiche38, 2018-11-11:
Implementation . Working , but required a firmware update of Zigate in order to handle 24bits

closed 2018-11-11 by pipiche38