zigbeefordomoticz/Domoticz-Zigbee #239: DeviceList-HWID.txt must be backup at startup

pipiche38, 2018-10-17:
As the primary database of the plugin, it might be good to backup the file at startup time .

zaraki673, 2018-10-22:
And check if file exist on first run of the plugin : 
`2018-10-21 14:04:12.252  (zigate) LoadDeviceList - DeviceList filename : /opt/domoticz/plugins/Domoticz-Zigate/DeviceList-9.txt
2018-10-21 14:04:12.252  Error: (zigate) 'onStart' failed 'FileNotFoundError'.
2018-10-21 14:04:12.252  Error: (zigate) ----> Line 286 in '/opt/domoticz/plugins/Domoticz-Zigate/plugin.py', function onStart
2018-10-21 14:04:12.252  Error: (zigate) ----> Line 119 in '/opt/domoticz/plugins/Domoticz-Zigate/plugin.py', function onStart
2018-10-21 14:04:12.252  Error: (zigate) ----> Line 21 in '/opt/domoticz/plugins/Domoticz-Zigate/z_database.py', function LoadDeviceList`

pipiche38, 2018-10-22:
Fix in 6c41ce7c7acde4ae13e77160ba971112f788943b

closed 2018-10-22 by pipiche38

pipiche38, 2018-10-22:
DeviceList-3.txt  is now backup as DeviceList-3.txt_2018-10-22-14:38:33