zigbeefordomoticz/Domoticz-Zigbee #240: How to switch from master branch to dev-x.x.x

BornTKill, 2018-10-20:
Hello.

I have read on zigate github that dev branch may solved the Lost Xiaomi devices in domoticz. I would like to test that. Can you explain me how to switch branch without losing configuration ? (DeviceList.txt I suppose)

Thanks in advance.

pipiche38, 2018-10-20:
Hi,

Not sure that will solved the lost of Xiaomi devices, as my understanding is more related to the Zigate firmware.
However in Dev, we have better way to handle the NetworkID versus IEEE.

But dev is dev, so you might get some disturbance and un-stability, however would be happy to have you also using it.

https://github.com/sasu-drooz/Domoticz-Zigate/blob/dev-x.x.x/Upgrade-IEEE.md  is a good start to get most of the information.

In case you wanted to go back to Master branch, then you'll have to restore DeviceList.txt and domoticz.db and do a get checkout master.

pipiche38, 2018-10-20:
I you wait few days, we will most-likely move this branch to Beta !

BornTKill, 2018-10-20:
Thank you for your quick answers!
I will wait until it’s in bêta ;)

closed 2018-10-22 by pipiche38