zigbeefordomoticz/Domoticz-Zigbee #246: Problem after changing switch value

Cortexlegeni, 2018-10-31:
Hello,
I'm along time user of the plugin but I had to leave it for a long period due to family issues.
Today I tried to change a xiaomi switch value and after this I got an error.

in the log : 

> 2018-10-31 17:36:36.199 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-10-31 17:36:36.199 Error: (Zigate) ----> Line 280 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-10-31 17:36:36.199 Error: (Zigate) ----> Line 242 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-10-31 17:36:36.199 Error: (Zigate) ----> Line 427 in '/home/pi/domoticz/plugins/Zigate/z_domoticz.py', function ResetDevice

> 

I can look at it if you want but If you already have the answer there's no need. Just tell me.
Thanks
Cortex

pipiche38, 2018-10-31:
@Cortexlegeni , welcome back ;-)

I believe you are on the latest version of master ?

if yes, Did you set any Selector Action to this particular switch ? as the error your are reporting refer to a missing element in the Domoticz Database

Cortexlegeni, 2018-10-31:
I tried ton remove the blocky script but I still have errors and even when I try direct commands 👍 

> 2018-10-31 21:58:59.190 Status: User: Admin initiated a switch command (1/Inter 8b/Set Level)
2018-10-31 21:58:59.211 Error: (Zigate) 'onCommand' failed 'KeyError'.
2018-10-31 21:58:59.211 Error: (Zigate) ----> Line 272 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onCommand
2018-10-31 21:58:59.211 Error: (Zigate) ----> Line 220 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onCommand
2018-10-31 21:58:59.211 Error: (Zigate) ----> Line 35 in '/home/pi/domoticz/plugins/Zigate/z_command.py', function mgtCommand

>

pipiche38, 2018-10-31:
Not sure if we are talking about the same thing. What I'm referring is the possibility to add a On or Off action to trigger a script. What I think you mentioned is more that you have a blocky event which is trigger when there is an event to this switch ?

If you know sqlite, it would be great if you could have a look to the domoticz.db database and check the switch .

`select Name, Options from DeviceStatus where DeviceID="xxxx" ;`

To find DeviceID, you can have in the Devices list on the Domoticz -> Setup

[image]

Last, if that is the real issue, you can switch to the current beta-3.1.x branch which doesn't have this compatibility issue with Domoticz. However you need either to remove all devices from Domoticz and start from a fresh plugin or to go through an upgrade process which is documented here : https://github.com/sasu-drooz/Domoticz-Zigate/blob/beta-3.1.x/Upgrade-IEEE.md

pipiche38, 2018-10-31:
@Cortexlegeni if you have done what I think you have done, the domoticz database has been overwritten, and the error will persist. In order to come back to a good state is to remove the device and create (pair) it again. In such it will create the Device with the right record in the database.

To remove the Device in the Master branch is a bit tricky, as the end-2-end removal was not yet implemented.
```
(1) remove the Device in Domoticz
(2) stop Domoticz
(3) Edit the file DeviceList.txt and remove the corresponding device entry 
(4) restart Domoticz

```

pipiche38, 2018-11-01:
> I tried ton remove the blocky script but I still have errors and even when I try direct commands 👍

This is the same problem, the database record for this device has been overwritten by Domoticz and some Zigate plugin information or not available anymore. The only chance to come back to normal is to remove the device and create it again (as mentioned in my previous response).

pipiche38, 2018-11-01:
@Cortexlegeni ,
At least it confirmed my initial thought, so if you want to switch to the beta-3.1.x branch this will allow you to do whatever you want. (but in all cases, you will have to remove this device from Domoticz and create it again)

Cortexlegeni, 2018-11-01:
I did the remove but the device seems to still be store somewhere:

> 2018-11-01 08:46:42.745 Status: (Zigate) Initialized version 3.0.3, author 'zaraki673 pipiche38'
2018-11-01 08:46:42.753 Status: (Zigate) onStart called - Zigate plugin V 3.0.3
2018-11-01 08:46:42.813 Status: (Zigate) onConnect called
2018-11-01 08:46:42.815 Status: (Zigate) Zigate enter in discover mode for 254 Secs
2018-11-01 08:46:42.967 Status: (Zigate) Major Version Num: 0001
2018-11-01 08:46:42.967 Status: (Zigate) Installer Version Number: 030c
2018-11-01 08:46:42.754 Error: (Zigate) Error loading Device ID: 1, Name: 'Inter 8b', nValue: 1, sValue: '10' not loaded int Zigate Plugin!
2018-11-01 08:46:43.019 (Zigate) Decode8015 : Number of devices known in Zigate = 3
2018-11-01 08:46:43.019 (Zigate) Decode8015 : [ 0] DevID = 00 Addr = b2d3 IEEE = 00158d0002135130 not found in ListOfDevice
2018-11-01 08:46:43.020 (Zigate) Decode8015 : [ 1] DevID = 01 Addr = 9ce5 IEEE = 00158d0002272ed6 RSSI = 156 Power = 00 found in ListOfDevice
2018-11-01 08:46:43.020 (Zigate) Decode8015 : [ 2] DevID = 32 Addr = af9d IEEE = 84182600000ca4d5 RSSI = 169 Power = 01 found in ListOfDevice
2018-11-01 08:46:44.418 Status: Incoming connection from: 192.168.1.36
2018-11-01 08:46:52.931 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-01 08:46:52.931 Error: (Zigate) ----> Line 280 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-11-01 08:46:52.931 Error: (Zigate) ----> Line 242 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-11-01 08:46:52.931 Error: (Zigate) ----> Line 427 in '/home/pi/domoticz/plugins/Zigate/z_domoticz.py', function ResetDevice
2018-11-01 08:47:02.752 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-01 08:47:02.753 Error: (Zigate) ----> Line 280 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-11-01 08:47:02.753 Error: (Zigate) ----> Line 242 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-11-01 08:47:02.753 Error: (Zigate) ----> Line 427 in '/home/pi/domoticz/plugins/Zigate/z_domoticz.py', function ResetDevice
2018-11-01 08:47:12.775 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-01 08:47:12.775 Error: (Zigate) ----> Line 280 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-11-01 08:47:12.775 Error: (Zigate) ----> Line 242 in '/home/pi/domoticz/plugins/Zigate/plugin.py', function onHeartbeat
2018-11-01 08:47:12.775 Error: (Zigate) ----> Line 427 in '/home/pi/domoticz/plugins/Zigate/z_domoticz.py', function ResetDevice

>

Cortexlegeni, 2018-11-01:
Do you want me to switch to beta or dev because if this problem is solved already I don't mind changing, it my testing Domoticz for Zigate tests so I don't mind. The objective is to investigate and solve new problems and not dig old ones that won't be in future versions.

pipiche38, 2018-11-01:
Yes, in the Beta or Dev, it is up to you. Beta will be a bit more stable than Dev.
in both cases beta or dev, you need either to start from a clean DB .

If you are registered to keybase.io you can reach me directly on @pipiche

pipiche38, 2018-11-08:
Fix on the Beta version

closed 2018-11-08 by pipiche38