zigbeefordomoticz/Domoticz-Zigbee #247: How can I stop this log spamming after trying to add Tradfri remote

karstenbakker, 2018-11-10:
I got problems with adding the remote control from Tradfri.
I press the button 4 times quick and after that my console starts spamming and it keeps going on while adding nothing.

I am using the latest beta version

2018-11-10 17:06:11.502 (Zigate stick) CreateDomoDevice - Device ID : d0cf5efffe049a76 Device EP : 01 Type :
2018-11-10 17:06:11.502 (Zigate stick) processNotinDBDevices - final step for creation of : b45e => {'SQN': '00', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': {}, 'Ep': {}, 'ZDeviceID': {}, 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '0', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': {}, 'Status': '0045'}
2018-11-10 17:06:11.503 (Zigate stick) onHeartbeat - creating device id : b45e with : {'SQN': '00', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': {}, 'Ep': {}, 'ZDeviceID': {}, 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '0', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': {}, 'Status': '0045'}
2018-11-10 17:06:11.503 (Zigate stick) processNotinDBDevices - final step for creation of : 4f59 => {'SQN': '25', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '23', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0045', 'LogicalType': 'End Device', 'ReceiveOnIdle': 'Off'}
2018-11-10 17:06:11.503 (Zigate stick) onHeartbeat - creating device id : 4f59 with : {'SQN': '25', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '23', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0045', 'LogicalType': 'End Device', 'ReceiveOnIdle': 'Off'}
2018-11-10 17:06:11.504 (Zigate stick) CreateDomoDevice - Device ID : d0cf5efffe049a76 Device EP : 01 Type :
2018-11-10 17:06:11.504 (Zigate stick) processNotinDBDevices - final step for creation of : e6c4 => {'SQN': '44', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '2', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0042', 'LogicalType': 'End Device', 'ReceiveOnIdle': 'Off'}
2018-11-10 17:06:11.504 (Zigate stick) onHeartbeat - creating device id : e6c4 with : {'SQN': '44', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '2', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0042', 'LogicalType': 'End Device', 'ReceiveOnIdle': 'Off'}
2018-11-10 17:06:11.504 (Zigate stick) CreateDomoDevice - Device ID : d0cf5efffe049a76 Device EP : 01 Type :
2018-11-10 17:06:11.505 (Zigate stick) processNotinDBDevices - final step for creation of : 0c26 => {'SQN': '00', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': {}, 'Ep': {}, 'ZDeviceID': {}, 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '22', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': {}, 'Status': '0042'}
2018-11-10 17:06:11.505 (Zigate stick) onHeartbeat - creating device id : 0c26 with : {'SQN': '00', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': {}, 'Ep': {}, 'ZDeviceID': {}, 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '22', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': {}, 'Status': '0042'}
2018-11-10 17:06:11.505 (Zigate stick) onHeartbeat - new device discovered 0x8045 received 693c
2018-11-10 17:06:11.505 (Zigate stick) onHeartbeat - new device discovered request Simple Descriptor 0x0043 and wait for 0x8043 for EP 01, of : 693c
2018-11-10 17:06:11.506 (Zigate stick) processNotinDBDevices - final step for creation of : 693c => {'SQN': '35', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '0', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0043', 'ReceiveOnIdle': 'Off', 'LogicalType': 'End Device', 'NbEp': '1'}
2018-11-10 17:06:11.506 (Zigate stick) onHeartbeat - creating device id : 693c with : {'SQN': '35', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '0', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0043', 'ReceiveOnIdle': 'Off', 'LogicalType': 'End Device', 'NbEp': '1'}
2018-11-10 17:06:11.507 (Zigate stick) CreateDomoDevice - Device ID : d0cf5efffe049a76 Device EP : 01 Type :
2018-11-10 17:06:11.507 (Zigate stick) processNotinDBDevices - final step for creation of : ab16 => {'SQN': '01', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': {}, 'Ep': {}, 'ZDeviceID': {}, 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '2', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': {}, 'Status': '0042'}
2018-11-10 17:06:11.507 (Zigate stick) onHeartbeat - creating device id : ab16 with : {'SQN': '01', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': {}, 'Ep': {}, 'ZDeviceID': {}, 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '2', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': {}, 'Status': '0042'}
2018-11-10 17:06:11.507 (Zigate stick) processNotinDBDevices - final step for creation of : 3a9e => {'SQN': '13', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '6', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0043', 'ReceiveOnIdle': 'Off', 'LogicalType': 'End Device', 'NbEp': '1'}
2018-11-10 17:06:11.508 (Zigate stick) onHeartbeat - creating device id : 3a9e with : {'SQN': '13', 'Version': '3', 'Type': {}, 'Battery': {}, 'Model': 'TRADFRI remote control', 'Manufacturer': '117c', 'Ep': {'01': {'ClusterType': {}, '0000': {}}}, 'ZDeviceID': {}, 'PowerSource': 'Battery', 'DeviceType': 'RFD', 'RIA': '10', 'IEEE': 'd0cf5efffe049a76', 'Heartbeat': '6', 'MacCapa': '80', 'ProfileID': {}, 'RSSI': 75, 'Status': '0043', 'ReceiveOnIdle': 'Off', 'LogicalType': 'End Device', 'NbEp': '1'}
2018-11-10 17:06:11.508 (Zigate stick) CreateDomoDevice - Device ID : d0cf5efffe049a76 Device EP : 01 Type :

pipiche38, 2018-11-10:
It looks like you are trying to pair the IKEA Remote control, which doesn't work! After a while the device will be put in an UNKNOWN state in the plugin database and the log will stop.

So I would suggest:
Don't leave the Zigate in Permit Join for ever. When you have finished to add your objects, simply return the Permit Join time to 0

karstenbakker, 2018-11-10:
All right, I did put the Zigate on 0. The log is still getting larger tho.

pipiche38, 2018-11-10:
So I suggest you to stop the plugin and edit the DeviceList.txt ( I suspect that you are on the Master branch)
Then remove that line