zigbeefordomoticz/Domoticz-Zigbee #249: Zigate not joining after update to fw 3.0e

Gemies, 2018-11-11:
Hi Guys,

Ik hope you can help me out. This evening I updated my Zigate firmware to the latest version 3.0e. Now when I permitted joining in domoticz for 254 seconds. The zigate is not recieving any devices. And it is not blinking as before. In my case tradfri bulbs who were working before.
I downloaded the software (NXP JN51XX FLash program) and update ( ZiGate_Coordinator_v3.0e.bin). I followed the update instructions as descibed on the website of Zigate. After updating firmware i also flashed (JennicSniffer_JN5168_1000000.bin) to get it in sniffer modus again.
When this did not work i did the flashing proces again, but nothing happens when i permit join in hardware (254 or 255)

I also reinstalled the domoticz zigate python plugin. But nothing joins nor the zigate device leds are blinking.

Does anyone have a clue what to do?

Gemies, 2018-11-12:
I am on the master, but I did not put pdm on true. I will try tonight.

Gemies, 2018-11-12:
@pipiche38 I tried starting the zigate with erase PM on and permit join time on 255. activated the plugin when zigate was connected. But the Zigate did not flash (red and blue) as in join mode. So i reflashed it with coordinator 3.0e and sniffer. cleared the Eeprom and restarted the plugin again. Still not joining. 

The thing that bothers me is that i cannot see the FW version in de Domo log only what you can see in the text file below . 

[zigatelog.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2573531/zigatelog.txt)

pipiche38, 2018-11-13:
If you have a windows system, I would like you to use this tool https://github.com/fairecasoimeme/ZiGate/blob/master/Tools/TestGUI_ZiGate_NXP-4.zip and see if you can control the Zigate and put it to join mode. In order to see if the issue is on Zigate or on the plugin

You will need those drivers for Windows10: https://github.com/fairecasoimeme/ZiGate/blob/master/Drivers/Windows10/pilotes_PL2303.zip

Gemies, 2018-11-13:
@pipiche38 

This is what I tried to today: 
I installed the nxp test-tool and went to setting put the baud rate from 9600 to 100.000 as the usb to tll HX version can handle 12mbps. Also in device management i put the serial port tot 11520. After that I opened the port and went to the tab discover devices, in Raw data came this code: 
17:08:42.564 -> 01 02 10 4E 02 10 02 13 4D 02 10 02 10 02 10 03  
I started joining the tradfri bulb (6x on and off). But nothing happened. It did not recieve any messages.

[image]

After that I re-flashed the zigate and got this log with a lot of faults.
[FlashGUIlog13-11.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2577071/FlashGUIlog13-11.txt) 
The usb connector on the pl-2303hd device is moving a little bit. Maybe it is faulty. 

Maybe you have some special insights. 

Thanks for your help so far. I really appreciate  it

pipiche38, 2018-11-13:
The right speed in 115200 (sorry for not mentioned earlier).
What you should do is, from the "Management" tab, there is Permit Join

Put FFFC in he Address and FF in the Interval

This must put the Zigate to Joint mode

If not, then the issue is on Zigate side

If yes, then we have problem ;-)

Gemies, 2018-11-13:
If think this is what you mean, (see screenshot below) but nothing happens. Should the red en blue light blink when it is in join state? They are steady red an blue and the second led on the usb -ttl is also doing nothing (blank). I tried joining some tradfri products, but nothing happens!

[image]

closed 2018-11-13 by Gemies

pipiche38, 2018-11-13:
What looks strange is you have nothing on the Receive Message view. And that is not normal at all.

Gemies, 2018-11-13:
I have sent a mail to Frederic Doubois from Zigate with a link to this issue thread.  Hopefully we can work things out. Thanks so much for your time and effort. If you have new thoughts on the issue please let me know.

pipiche38, 2018-11-13:
I had a similar problem when I tried with the GUI just after flashing the firmware. I removed the ZIgate and plugin again and it works.

It is somehow when you put the ZIgate in Write mode (for firmware update), it might be protected and you cannot access it without the upgrad tool.

pipiche38, 2018-11-14:
The GUI has nothing to do with python it is a pure binary code

Gemies, 2018-11-14:
Right now i am running raspbian Jessie with python 3.4.2.

pipiche38, 2018-11-14:
@Gemies seems that you are progressing. From the messages I saw , it means that the Zigate is receiving message. is now the led flashing as expected to allow Pairing ?

As python, I suggest you to have a look to https://www.domoticz.com/wiki/Using_Python_plugins
 and indeed python 3.4 seems to be under condition

Gemies, 2018-11-16:
@pipiche
Sorry, was upgrading my OS, but this went wrong, and had flash a new image. Now I am on python 3.5.3.
My question now is,  do I need to flash Jennifer sniffer or should it work without it. 
Because join with domoticz isn't working yet.

pipiche38, 2018-11-16:
If the Zigate is on the 3.0e firmware nothing to do. 
You need to add new hardware in Domoticz to add Zigate plugin , select the channel you want or put 0 . Put software reset (for the first time)  , and set permit to join to 254.

Then you need to pair your devices

Gemies, 2018-11-18:
Somehow the Zigate is not switching to joinmode. Also when I Flashed FW 3.0d it does not join anymore. Getting a bit frustrated. In GUI everything works. Also with this FW version. But with a Flashed jennic sniffer it does not work.

 But when I do not flash the Jennic sniffer domoticz gives this issue.
Framesize is bad and Frame CRC is bad. This stop's after flashing sniffer. As posted before.
Would installing the development plugin be an option. Git checkout developement does not work for me. 

I am a bit desperate right now. I really appreciate any help.

pipiche38, 2018-11-18:
You should use the 3.0e firmware and then after upgrading the Zigate you should start the plugin with Erase PDM option.

Are you on WiFi Zigate ? Or is that a USB-TTL one ?

The Crc /frame length are linked to WiFi 

I don’t understand what you do with flashing the Jennic sniffer.

There is no development branch at that time, only beta-3.2

Gemies, 2018-11-18:
This is what the installation guide on the zigate website says about JennicSniffer. I have the USB TTL version. 

Mode Sniffer
Pour utiliser votre ZiGate en mode sniffer ZigBee, vous pouvez suivre les instructions précédente. Seul le fichier binaire (le Firmware) est à changer. Vous pourrez le trouver ici:

JennicSniffer_JN5168_1000000.bin

Une fois installé, vous pourrez utiliser Ubiqua protocol analyzer ou même Wireshark (à développer) pour sniffer toutes les trames ZigBee.

Gemies, 2018-11-19:
Could it be plugin thing? Is it possible to get the log from the python plugin? Because joining is still not working. I followed your instructions step by step.

pipiche38, 2018-11-20:
Can you join me on keybase.io @pipiche

Gemies, 2018-11-20:
Joined keybase!

pipiche38, 2018-11-20:
Update to 3.0.4