zigbeefordomoticz/Domoticz-Zigbee #250: DeviceList being cleared after editing PluginConf delay

karstenbakker, 2018-11-12:
Today I tried to adjust the send delay, because everytime I switch on a group or scene it is going one by one and very slow. So I wanted to change this.

So I stopped domoticz, opened the PluginConf.txt and edited senddelay to 0.5 only the script doesnt take floats. After that I started domoticz and it seems my DeviceList was cleared out because of this. If I set it to 0 unfortunately some lamps don't go on. 

Is there any option to make the group go on at once? Just like if I control a remote from Tradfri.

Also I am running on beta.3.1.xx

pipiche38, 2018-11-13:
I'm very sorry for this bad experience,

For now the plugin doesn't manage group nor scene!

Today we cannot have float and .5 for instance due to Domoticz restriction in the Python framework.

I'm going to fix the issue of the float to avoid corrupting the DeviceList. As you might see in Beta-3.2 we make a backup of the DeviceList prior each start.

Are you on USB-TTL or Wifi ?

karstenbakker, 2018-11-13:
No problem.
Is the support for groups or scenes coming tho? Because this has been the mean reason for buying the Zigate so I can get rid of all the gateways.

I am on USB-TTL.

pipiche38, 2018-11-13:
For now the support of Group/scene is in the pipeline, but not much developer and we have still a number of issue to fix. 
What is on my TO DO is basically to address this delay stuff, and get a much more reactive way of processing commands

pipiche38, 2018-11-13:
`So I stopped domoticz, opened the PluginConf.txt and edited senddelay to 0.5 only the script doesnt take floats. ` is now fixed in the latest Beta-3.2

pipiche38, 2018-11-26:
@karstenbakker just push Beta-3.3, if you want to give a try, performance on your switch On/Off would be much better.

karstenbakker, 2018-11-26:
@pipiche38 I have switched to beta-3.3, indeed the switching of single lights is really fast. Unfortunately If I switch a group right now, it goes really really slow and it sometimes misses the lights .

karstenbakker, 2018-11-26:
2018-11-26 21:10:09.547 (Zigate stick) Decode8120 - Configure Reporting response - ClusterID: 0300, MsgSrcAddr: 367b, MsgSrcEp:01 , Status: 86 - The End Device bind request was unsuccessful due to a failure to match any suitable clusters.
2018-11-26 21:10:11.590 (Zigate stick) Decode8120 - Configure Reporting response - ClusterID: 0300, MsgSrcAddr: 367b, MsgSrcEp:01 , Status: 86 - The End Device bind request was unsuccessful due to a failure to match any suitable clusters.
2018-11-26 21:10:11.809 (Zigate stick) Decode8120 - Configure Reporting response - ClusterID: 0300, MsgSrcAddr: 367b, MsgSrcEp:01 , Status: 87 - No free entries in the extended address table (resource error) - this table is configured in the ZPS Configuration Editor
2018-11-26 21:10:17.025 (Zigate stick) Update v2 Values 1:'On:' (WK-LMP-TV)

I do receive this somehow?

karstenbakker, 2018-11-26:
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Management Network Update. SQN: 00, Total Transmit: 25 , Transmit Failures: 7 , Status: Success)
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Management Network Update
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 11 Interference: : 192
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 12 Interference: : 168
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 13 Interference: : 192
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 14 Interference: : 166
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 15 Interference: : 157
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 16 Interference: : 157
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 17 Interference: : 158
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 18 Interference: : 157
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 19 Interference: : 156
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 20 Interference: : 157
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 21 Interference: : 156
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 22 Interference: : 158
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 23 Interference: : 182
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 24 Interference: : 157
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 25 Interference: : 156
2018-11-26 21:15:01.724 Status: (Zigate stick) Decode804A - Channel: 26 Interference: : 156

karstenbakker, 2018-11-26:
Well I deleted everything by accident. So I will start to add everything fresh again.

pipiche38, 2018-11-26:
Not a big issue. We are still on Beta and those messages are in the Log. Looks like you have a ColorControl Bulbe with Network Address 367b, and that one doesn't accept Configure Reporting on Cluster 0x0300.
On the switch of a group, I would be interested on the Log , as I would like to understand if the slowness  if coming from Domoticz, the Plugin or the Zigate itself.

karstenbakker, 2018-11-26:
I don't have any color bulb, only the white with different white colors. I just tried adding the devices, but I now get all RGB colors which isn't my bulb.

karstenbakker, 2018-11-26:
This is what I collected, the shutdown is perfect everything goes out same time but on not:
`0
2018-11-26 23:16:19.774 (Zigate stick) ListOfDevices : {'3de3': {'SQN': {}, 'IEEE': 'd0cf5efffe13bec9', 'LogicalType': 'Router', 'Stack Version': '87', 'MacCapa': '8e', 'Version': '3', 'Heartbeat': '15', 'Battery': {}, 'RSSI': 48, 'Type': 'ColorControl/Switch/LvlControl/', 'NbEp': '1', 'Ep': {'01': {'0000': {}, '0008': {}, '0003': {}, '0b05': {}, '1000': {}, 'ClusterType': {'3': 'ColorControl'}, '0004': {}, '0300': {}, '0005': {}, '0006': '00'}}, 'RIA': '1', 'PowerSource': 'Main', 'Manufacturer': '117c', 'HW Version': '1', 'Status': 'inDB', 'App Version': '17', 'DeviceType': 'FFD', 'Model': 'TRADFRI bulb E27 WS opal 980lm', 'ReceiveOnIdle': 'On', 'Stamp': {'Time': '2018-11-26 23:16:04', 'MsgType': '8102'}, 'ZDeviceID': '0220', 'ProfileID': 'c05e', 'ColorInfos': {'X': '30138', 'ColorTemperatureMireds': '370', 'Y': '26909', 'ColorMode': 2}}, 'b51f': {'SQN': {}, 'IEEE': 'd0cf5efffe1d53c8', 'LogicalType': 'Router', 'Stack Version': '87', 'MacCapa': '8e', 'Version': '3', 'Heartbeat': 0, 'Battery': {}, 'RSSI': 93, 'Type': 'ColorControl/LvlControl/Switch/', 'NbEp': '1', 'Ep': {'01': {'0000': {}, '0008': {}, '0003': {}, '0b05': {}, '1000': {}, 'ClusterType': {'1': 'ColorControl'}, '0004': {}, '0300': {}, '0005': {}, '0006': '01'}}, 'RIA': '5', 'PowerSource': 'Main', 'Manufacturer': '117c', 'HW Version': '1', 'Status': 'inDB', 'App Version': '17', 'DeviceType': 'FFD', 'Model': 'TRADFRI bulb E27 WS opal 980lm', 'ReceiveOnIdle': 'On', 'Stamp': {'Time': '2018-11-26 23:16:15', 'MsgType': '8102'}, 'ZDeviceID': '0220', 'ProfileID': 'c05e', 'ColorInfos': {'X': '30138', 'ColorTemperatureMireds': '370', 'Y': '26909', 'ColorMode': 2}}, 'e246': {'SQN': {}, 'IEEE': 'd0cf5efffe1382e3', 'LogicalType': 'Router', 'Stack Version': '87', 'MacCapa': '8e', 'Version': '3', 'Heartbeat': 0, 'Battery': {}, 'RSSI': 93, 'Type': 'ColorControl/LvlControl/Switch/', 'NbEp': '1', 'Ep': {'01': {'0000': {}, '0008': {}, '0003': {}, '0b05': {}, '1000': {}, 'ClusterType': {'2': 'ColorControl'}, '0004': {}, '0300': {}, '0005': {}, '0006': '00'}}, 'RIA': '1', 'PowerSource': 'Main', 'Manufacturer': '117c', 'HW Version': '1', 'Status': 'inDB', 'App Version': '17', 'DeviceType': 'FFD', 'Model': 'TRADFRI bulb E27 WS opal 980lm', 'ReceiveOnIdle': 'On', 'Stamp': {'Time': '2018-11-26 23:16:09', 'MsgType': '8102'}, 'ZDeviceID': '0220', 'ProfileID': 'c05e', 'ColorInfos': {'X': '30138', 'ColorTemperatureMireds': '370', 'Y': '26909', 'ColorMode': 2}}}
2018-11-26 23:16:19.775 (Zigate stick) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 1 normalQ: 2
2018-11-26 23:16:19.775 (Zigate stick) checkTOwaitForStatus - 8002s enter at: 1543270575 delta: 4
2018-11-26 23:16:19.775 (Zigate stick) waitForData - Timeout 4 on 8002
2018-11-26 23:16:19.775 (Zigate stick) waitForData - waitQ: 0 dataQ: 0 normalQ: 2
2018-11-26 23:16:19.775 (Zigate stick) sendData - Cmd: 0081 waitQ: 0 dataQ: 0 normalQ: 1
2018-11-26 23:16:19.775 (Zigate stick) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 1
2018-11-26 23:16:19.775 (Zigate stick) sendData - Cmd: 00C0 waitQ: 0 dataQ: 0 normalQ: 0
2018-11-26 23:16:19.775 (Zigate stick) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-11-26 23:16:19.775 (Zigate stick) _sendData 00C0
2018-11-26 23:16:19.776 (Zigate stick) ZigateEncode - Encodind data: 00C0
2018-11-26 23:16:19.776 (Zigate stick) Transcode in: 00C0 / out:0210C0
2018-11-26 23:16:19.776 (Zigate stick) ZigateEncode - Encodind data: 0009
2018-11-26 23:16:19.776 (Zigate stick) Transcode in: 0009 / out:02100219
2018-11-26 23:16:19.776 (Zigate stick) ZigateEncode - Encodind data: b9
2018-11-26 23:16:19.776 (Zigate stick) Transcode in: b9 / out:b9
2018-11-26 23:16:19.776 (Zigate stick) ZigateEncode - Encodind data: 02b51f010101d90000
2018-11-26 23:16:19.776 (Zigate stick) Transcode in: 02b51f010101d90000 / out:0212b51f021102110211d902100210
2018-11-26 23:16:19.777 (Zigate stick) addCmdToSend: cmd: 0081 data: 02e246010101830000 reTransmit: 0
2018-11-26 23:16:19.777 (Zigate stick) ResetDevice - Found ClusterType in EP[01] : ColorControl for Device :1
2018-11-26 23:16:19.778 (Zigate stick) ResetDevice - Found ClusterType in EP[01] : ColorControl for Device :2
2018-11-26 23:16:19.778 (Zigate stick) HB count = 31
2018-11-26 23:16:19.878 (Zigate stick) onMessage called on Connection Name: 'ZiGate', Transport: 'Serial', Protocol: 'None', Address: '/dev/ttyUSB0', Port: '', Baud: 115200, Bytes: 2391, Connected: True, Last Seen: 2018-11-26 23:16:19 Data = 'b'\x01\x80\x02\x10\x02\x10\x02\x15\x02\x1f\x02\x10J\x02\x10\xc0\x02\x10\x03\x01\x81\x02\x11\x02\x10\x02\x17\x98J\x02\x11\x02\x13\x02\x10\x02\x1a\x02\x10]\x03''
2018-11-26 23:16:19.879 (Zigate stick) onMessage called on Connection b'\x01\x80\x02\x10\x02\x10\x02\x15\x02\x1f\x02\x10J\x02\x10\xc0\x02\x10\x03\x01\x81\x02\x11\x02\x10\x02\x17\x98J\x02\x11\x02\x13\x02\x10\x02\x1a\x02\x10]\x03'
2018-11-26 23:16:19.879 (Zigate stick) onMessage incoming data : '0180021002100215021f02104a0210c00210030181021102100217984a021102130210021a02105d03'
2018-11-26 23:16:19.879 (Zigate stick) receiveData - new Data coming
2018-11-26 23:16:19.879 (Zigate stick) receiveData - MsgType: 8000
2018-11-26 23:16:19.879 (Zigate stick) receivedData - MsgType: 8000 PacketType: 00c0
2018-11-26 23:16:19.879 (Zigate stick) receiveStatusCmd - waitQ: 1 dataQ: 1 normalQ: 1
2018-11-26 23:16:19.880 (Zigate stick) receiveData - new Data coming
[... cut ...]
2018-11-26 23:16:19.880 (Zigate stick) ZigateRead - MsgType: 8101, MsgLength: 0007, MsgCRC: 98, Data: 4a0103000a00; RSSI: 5d
2018-11-26 23:16:19.880 (Zigate stick) ZigateRead - MsgType 8101 - Default Response: 0181010007984a0103000a005d03
2018-11-26 23:16:19.880 (Zigate stick) Decode8101 - reception Default response : SQN : 4a, EP : 01, Cluster ID : 0300 , Command : 0a, Status : Success
2018-11-26 23:16:19.880 (Zigate stick) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 1 normalQ: 1
2018-11-26 23:16:19.880 (Zigate stick) checkTOwaitForStatus - 8002s enter at: 1543270579 delta: 0`

pipiche38, 2018-11-27:
Could you please attach as a file, as there is not a single end of line and it is quiet an heavy work to decode it.
Could you kindly do the same, but without enabling Debug, so remains at Debug = None

pipiche38, 2018-11-27:
Could you reach me on keybase.io @pipiche , it will be easier to communicate over chat