zigbeefordomoticz/Domoticz-Zigbee #251: Error out of range after switching to beta 3.2

karstenbakker, 2018-11-12:
After switching to beta 3.2 I am receiving the following errors in my logs and the Zigate isn't working.
Also I discovered that there were 8 devices in my DeviceConf but somehow the Zigate says it got only 6.

The underneath starts first with sending a message to all device with speed 000 sec then after a while the following comes back.

2018-11-12 23:00:10.209 (Zigate stick) Decode8000_v2 - status: ZigBee Error Code The supplied request type was invalid. SEQ: 00 Packet Type: 0030 Queue = 24
2018-11-12 23:00:14.120 (Zigate stick) Decode8000_v2 - status: ZigBee Error Code A parameter value was invaid or out of range SEQ: 00 Packet Type: 0030 Queue = 21

Sorry for spamming the issue's but this is what I walked in to.

pipiche38, 2018-11-13:
@karstenbakker , I wouldn't worry too much of this issue. Basically it is telling you that some command ( a bind request ) were not successful, during the start-up phase.

Eventually to dig down , I would appreciate if you could post the Startup Log with setting "Verbose and Debuging" level to "Verbose"

karstenbakker, 2018-11-13:
Well, Somehow it seems to works right now.

Don't know what changed, I also noticed that alot of devicelist-2.txt-2018... ...  files were made somehow.

Also the Devices known in zigate in the logs says it sees 6 but the list is 8.
2018-11-13 19:14:38.751 Status: (Zigate stick) onStart called - Zigate plugin V beta-3.2
2018-11-13 19:14:38.754 Status: (Zigate stick) load PluginConf
2018-11-13 19:14:38.758 Status: (Zigate stick) load ListOfDevice
2018-11-13 19:14:38.786 Status: (Zigate stick) Entries loaded from /home/pi/domoticz/plugins/Domoticz-Zigate/DeviceList-2.txt : 8
2018-11-13 19:14:38.793 Status: (Zigate stick) onConnect called
2018-11-13 19:14:38.795 Status: (Zigate stick) Start network
2018-11-13 19:14:38.897 Status: (Zigate stick) Major Version Num: 0001
2018-11-13 19:14:38.897 Status: (Zigate stick) Installer Version Number: 030d
2018-11-13 19:14:39.149 Status: (Zigate stick) Decode8015 : Number of devices known in Zigate = 6

And the console keeps spamming this with the verbose:

018-11-13 19:14:48.677 (Zigate stick) ResetDevice - Found ClusterType in EP[01] : LvlControl for Device :8
2018-11-13 19:14:48.677 (Zigate stick) ResetDevice - No ClusterType found for this device : 8 in {'Battery': {}, 'Status': 'inDB', 'SQN': {}, 'RSSI': 99, 'Manufacturer': '117c', 'Heartbeat': '2', 'Version': '3', 'Ep': {'02': {'ClusterType': {}, '1000': {}}, 'f2': {'ClusterType': {}, '0021': {}}, '01': {'0003': {}, 'fc7c': {}, '0004': {}, '0005': {}, '0008': {}, '0000': {}, 'ClusterType': {'8': 'LvlControl'}, '0006': '01'}}, 'RIA': '10', 'PowerSource': 'Main', 'NbEp': '3', 'DeviceType': 'FFD', 'MacCapa': '8e', 'ProfileID': '0104', 'ZDeviceID': '010a', 'Model': 'plug.legrand.netamo', 'IEEE': 'd0cf5efffecb7e8f', 'Type': 'Switch/LvlControl/', 'ReceiveOnIdle': 'On', 'LogicalType': 'Router'}

pipiche38, 2018-11-13:
> Well, Somehow it seems to works right now.
> 
> Don't know what changed, I also noticed that alot of devicelist-2.txt-2018... ... files were made somehow.

Yes, we have an auto-backup at each start of the plugin.

> Also the Devices known in zigate in the logs says it sees 6 but the list is 8.
> 2018-11-13 19:14:38.751 Status: (Zigate stick) onStart called - Zigate plugin V beta-3.2
> 2018-11-13 19:14:38.754 [...]

This is normal as this is not really the number of device if it is the number of most recent active device on the Zigate

> And the console keeps spamming this with the verbose:
>
> 018-11-13 19:14:48.677 (Zigate stick) ResetDevice - Found ClusterType in EP[01] : LvlControl for Device :8
> 2018-11-13 19:14:48.677 (Zigate [...]
Did you migrate from Master to Beta branch ? Or did you start from an empty plugin Database ?
Could you kindly post the DeviceList-2.txt ? Something looks wrong here.

> I also noticed when editing the verbose to normall it keeps giving te same errors again while the zigate doesnt switch any device:
> 2018-11-13 19:21:31.367 (Zigate stick) identifySend - send an Identify Message to: be58 for 0000 seconds
> 2018-11-13 19:21:31.373 (Zigate stick) identifySend - send an Identify Message to: 8171 for 0000 seconds
> 2018-11-13 19:21:31.373 (Zigate stick) identifySend - send an Identify Message to: 8171 for 0000 seconds
> 2018-11-13 19:21:31.379 (Zigate stick) identifySend - send an Identify Message to: 8171 for 0000 seconds
> 2018-11-13 19:21:31.379 (Zigate stick) identifySend - send an Identify Message to: 367b for 0000 seconds
> 2018-11-13 19:21:31.384 (Zigate stick) identifySend - send an Identify Message to: 0641 for 0000 seconds
> 2018-11-13 19:21:31.390 (Zigate stick) identifySend - send an Identify Message to: e299 for 0000 seconds
> 2018-11-13 19:21:31.395 (Zigate stick) identifySend - send an Identify Message to: 2446 for 0000 seconds
> 2018-11-13 19:21:31.400 (Zigate stick) identifySend - send an Identify Message to: 661b for 0000 seconds
> 2018-11-13 19:21:31.405 (Zigate stick) identifySend - send an Identify Message to: aee6 for 0000 seconds
> 2018-11-13 19:21:39.342 (Zigate stick) Decode8000_v2 - status: ZigBee Error Code The supplied request type was invalid. SEQ: 00 Packet Type: 0030 Queue = 34
> 2018-11-13 19:21:47.130 (Zigate stick) Decode8000_v2 - status: ZigBee Error Code The supplied request type was invalid. SEQ: 00 Packet Type: 0030 Queue = 32

Yes, we are still in Beta and the plugin is still a bit verbose ...

pipiche38, 2018-11-13:
What do you mean 'migrate'. If you were on Beta-3.1, you don't have to do more that restarting the plugin (domoticz)

pipiche38, 2018-11-15:
so could you clarify what is the issue ?

from the log above you are on beta-3.2
2```
018-11-13 19:14:38.751 Status: (Zigate stick) onStart called - Zigate plugin V beta-3.2
2018-11-13 19:14:38.754 Status: (Zigate stick) load PluginConf
2018-11-13 19:14:38.758 Status: (Zigate stick) load ListOfDevice
```

karstenbakker, 2018-11-15:
The whole issue started when I switched to beta 3.2, It keeps sendings this weird things and the Zigate doesnt send out any switches towards the bulbs
2018-11-13 19:21:31.367 (Zigate stick) identifySend - send an Identify Message to: be58 for 0000 seconds
2018-11-13 19:21:31.373 (Zigate stick) identifySend - send an Identify Message to: 8171 for 0000 seconds
2018-11-13 19:21:31.373 (Zigate stick) identifySend - send an Identify Message to: 8171 for 0000 seconds
2018-11-13 19:21:31.379 (Zigate stick) identifySend - send an Identify Message to: 8171 for 0000 seconds
2018-11-13 19:21:31.379 (Zigate stick) identifySend - send an Identify Message to: 367b for 0000 seconds
2018-11-13 19:21:31.384 (Zigate stick) identifySend - send an Identify Message to: 0641 for 0000 seconds
2018-11-13 19:21:31.390 (Zigate stick) identifySend - send an Identify Message to: e299 for 0000 seconds
2018-11-13 19:21:31.395 (Zigate stick) identifySend - send an Identify Message to: 2446 for 0000 seconds
2018-11-13 19:21:31.400 (Zigate stick) identifySend - send an Identify Message to: 661b for 0000 seconds
2018-11-13 19:21:31.405 (Zigate stick) identifySend - send an Identify Message to: aee6 for 0000 seconds

pipiche38, 2018-11-15:
These are normal messages, that is not an issue . What do you mean keep sending it ? I don't see any repeat here.
This message should by the way not appear in the recent beta-3.2 .

Lats, when you say, Zigate doesn't send out any switches towards bulbs, do you mean allways ! or only during the startup phase ( 2 first minutes). As indeed during the few minutes after start, the plugin is busy to setup a number of things.

pipiche38, 2018-11-15:
Here is what I wrote about your DeviceList-2.txt:
What do you mean 'migrate'. If you were on Beta-3.1, you don't have to do more that restarting the plugin (domoticz) and the devicelist seems to miss a lot of information. 

They are also very strange caracters like : 'V$ ' don't know if they came form the copy-paste or if they are in the file itself. 
Last what do you have on the log, this will help to understand what is happing

karstenbakker, 2018-11-17:
Seems everything is working after the latest commits you made, I am going to run the beta-3.2 for a while to see if I encounter any errors.

closed 2018-11-17 by karstenbakker