zigbeefordomoticz/Domoticz-Zigbee #252: domoticz not able to restart

BornTKill, 2018-11-18:
**Describe the bug**
I've just successfully upgraded zigate fw to 3.0e (to avoid xiaomi lost devices) and everything was OK.
So I decided to upgrade the plugin to beta 3.2. IEEE upgrade process was OK.
Because I had a lot of lost devices (xiaomi temperature sensor), I completely deleted my zigate devices list in domoticz. I check and de DeviceList-13.txt files was also empty after that.

Then I join all my devices without problem. My DeviceList-13.txt looks like :

feec : {'App Version': '5', 'Stack Version': '2', 'RIA': '0', 'RSSI': 64, 'HW Version': '1', 'PowerSource': 'Battery', 'Version': '3', 'Model': 'lumi.sensor_cube.aqgl01', 'MacCapa': '80', 'Manufacturer': '1037', 'ProfileID': {}, 'SQN': '1c', 'Status': 'inDB', 'Heartbeat': '159', 'NbEp': '3', 'Ep': {'02': {'0012': '0048'}, '03': {'000c': '80'}, '01': {'ClusterType': {'67': 'Aqara'}, '000c': {}, '0012': {}, '0000': {}}}, 'LogicalType': 'End Device', 'DeviceType': 'RFD', 'ReceiveOnIdle': 'Off', 'Type': 'Aqara', 'IEEE': '00158d00028f7824', 'Stamp': {'Time': '2018-11-18 21:23:00', 'MsgType': '8102'}, 'ZDeviceID': {}, 'Battery': {}}
0fa7 : {'App Version': '3', 'Stack Version': '2', 'RIA': '0', 'RSSI': 116, 'HW Version': '30', 'PowerSource': 'Battery', 'Version': '3', 'Model': 'lumi.weather', 'MacCapa': '80', 'Manufacturer': '1037', 'ProfileID': {}, 'SQN': '13', 'Status': 'inDB', 'Heartbeat': '182', 'NbEp': '1', 'Ep': {'01': {'ClusterType': {'60': 'Humi', '58': 'Temp+Hum', '59': 'Temp', '61': 'Baro', '57': 'Temp+Hum+Baro'}, '0003': {}, '0403': 1013.2, '0402': 21.0, '0000': {}, '0405': 58.1, 'ffff': {}, 'Type': 'Temp/Humi/Baro'}}, 'LogicalType': 'End Device', 'DeviceType': 'RFD', 'ReceiveOnIdle': 'Off', 'Type': '', 'IEEE': '00158d000233f95f', 'Stamp': {'Time': '2018-11-18 21:26:09', 'MsgType': '8102'}, 'ZDeviceID': {}, 'Battery': {}}
f333 : {'App Version': '3', 'Stack Version': '2', 'RIA': '0', 'RSSI': 62, 'HW Version': '30', 'PowerSource': 'Battery', 'Version': '3', 'Model': 'lumi.weather', 'MacCapa': '80', 'Manufacturer': '1037', 'ProfileID': {}, 'SQN': '1d', 'Status': 'inDB', 'Heartbeat': '196', 'NbEp': '1', 'Ep': {'01': {'ClusterType': {'53': 'Temp+Hum', '56': 'Baro', '54': 'Temp', '52': 'Temp+Hum+Baro', '55': 'Humi'}, '0003': {}, '0403': 1013.1, '0402': 21.4, '0000': {}, '0405': 59.9, 'ffff': {}, 'Type': 'Temp/Humi/Baro'}}, 'LogicalType': 'End Device', 'DeviceType': 'RFD', 'ReceiveOnIdle': 'Off', 'Type': '', 'IEEE': '00158d000237160b', 'Stamp': {'Time': '2018-11-18 21:26:12', 'MsgType': '8102'}, 'ZDeviceID': {}, 'Battery': {}}
6321 : {'App Version': '3', 'Stack Version': '2', 'RIA': '0', 'RSSI': 45, 'HW Version': '30', 'PowerSource': 'Battery', 'Version': '3', 'Model': 'lumi.weather', 'MacCapa': '80', 'Manufacturer': '1037', 'ProfileID': {}, 'SQN': '11', 'Status': 'inDB', 'Heartbeat': '171', 'NbEp': '1', 'Ep': {'01': {'ClusterType': {'66': 'Baro', '63': 'Temp+Hum', '64': 'Temp', '62': 'Temp+Hum+Baro', '65': 'Humi'}, '0003': {}, '0403': 1013.1, '0402': 23.2, '0000': {}, '0405': 55.0, 'ffff': {}, 'Type': 'Temp/Humi/Baro'}}, 'LogicalType': 'End Device', 'DeviceType': 'RFD', 'ReceiveOnIdle': 'Off', 'Type': '', 'IEEE': '00158d000233f92c', 'Stamp': {'Time': '2018-11-18 21:18:51', 'MsgType': '8102'}, 'ZDeviceID': {}, 'Battery': {}}

I decided to disable joining mode and after that, domoticz was not able to restart.
Here is the log :

2018-11-18 22:27:10.869  Status: RxQueue: queue worker started...
2018-11-18 22:27:12.869  Status: (Zigate) Started.
2018-11-18 22:27:12.870  Status: EventSystem: reset all events...
2018-11-18 22:27:12.870  Status: EventSystem: reset all device statuses...
2018-11-18 22:27:12.995  Status: Python EventSystem: Initalizing event module.
2018-11-18 22:27:12.996  Status: EventSystem: Started
2018-11-18 22:27:12.996  Status: EventSystem: Queue thread started...
2018-11-18 22:27:13.256  Status: PluginSystem: Entering work loop.
2018-11-18 22:27:13.471  Status: (Zigate) Entering work loop.
2018-11-18 22:27:13.472  Status: (Zigate) Initialized version beta-3.2, author 'zaraki673  pipiche38'
2018-11-18 22:27:13.477  Status: (Zigate) onStart called - Zigate plugin V beta-3.2
2018-11-18 22:27:13.478  (Zigate) Debug logging mask set to: PYTHON PLUGIN IMAGE DEVICE CONNECTION MESSAGE 
2018-11-18 22:27:13.478  (Zigate) 'Mode5':'11'
2018-11-18 22:27:13.478  (Zigate) 'Version':'beta-3.2'
2018-11-18 22:27:13.478  (Zigate) 'DomoticzBuildTime':'2018-06-23 16:24:51'
2018-11-18 22:27:13.478  (Zigate) 'HardwareID':'13'
2018-11-18 22:27:13.478  (Zigate) 'Mode3':'False'
2018-11-18 22:27:13.478  (Zigate) 'Port':'9999'
2018-11-18 22:27:13.478  (Zigate) 'Author':'zaraki673  pipiche38'
2018-11-18 22:27:13.478  (Zigate) 'Mode6':'126'
2018-11-18 22:27:13.478  (Zigate) 'DomoticzVersion':'4.9700'
2018-11-18 22:27:13.478  (Zigate) 'Mode4':'False'
2018-11-18 22:27:13.478  (Zigate) 'Mode1':'USB'
2018-11-18 22:27:13.478  (Zigate) 'SerialPort':'/dev/ttyUSB1'
2018-11-18 22:27:13.478  (Zigate) 'HomeFolder':'/home/pi/domoticz/plugins/Domoticz-Zigate/'
2018-11-18 22:27:13.479  (Zigate) 'Address':'0.0.0.0'
2018-11-18 22:27:13.479  (Zigate) 'Mode2':'255'
2018-11-18 22:27:13.479  (Zigate) 'Key':'Zigate'
2018-11-18 22:27:13.479  (Zigate) 'Name':'Zigate'
2018-11-18 22:27:13.479  (Zigate) 'DomoticzHash':'a3a45906'
2018-11-18 22:27:13.479  (Zigate) Device count: 16
2018-11-18 22:27:13.479  (Zigate) Device:           1 - ID: 52, Name: 'Zigate - Temp+Hum+Baro-00158d000237160b-01', nValue: 0, sValue: '21.4;59.9;1;1013.1;3'
2018-11-18 22:27:13.479  (Zigate) Device ID:       '52'
2018-11-18 22:27:13.479  (Zigate) Device Name:     'Zigate - Temp+Hum+Baro-00158d000237160b-01'
2018-11-18 22:27:13.479  (Zigate) Device nValue:    0
2018-11-18 22:27:13.479  (Zigate) Device sValue:   '21.4;59.9;1;1013.1;3'
2018-11-18 22:27:13.479  (Zigate) Device LastLevel: 0
2018-11-18 22:27:13.479  (Zigate) Device Options: {}
2018-11-18 22:27:13.479  (Zigate) Device:           2 - ID: 53, Name: 'Zigate - Temp+Hum-00158d000237160b-01', nValue: 0, sValue: '21.4;59.9;1'
2018-11-18 22:27:13.479  (Zigate) Device ID:       '53'
2018-11-18 22:27:13.479  (Zigate) Device Name:     'Zigate - Temp+Hum-00158d000237160b-01'
2018-11-18 22:27:13.479  (Zigate) Device nValue:    0
2018-11-18 22:27:13.479  (Zigate) Device sValue:   '21.4;59.9;1'
2018-11-18 22:27:13.479  (Zigate) Device LastLevel: 0
2018-11-18 22:27:13.479  (Zigate) Device Options: {}
2018-11-18 22:27:13.480  (Zigate) Device:           3 - ID: 54, Name: 'Zigate - Température Salon', nValue: 21, sValue: '21.4'
2018-11-18 22:27:13.480  (Zigate) Device ID:       '54'
2018-11-18 22:27:13.480  (Zigate) Device Name:     'Zigate - Température Salon'
2018-11-18 22:27:13.480  (Zigate) Device nValue:    21
2018-11-18 22:27:13.480  (Zigate) Device sValue:   '21.4'
2018-11-18 22:27:13.480  (Zigate) Device LastLevel: 0
2018-11-18 22:27:13.480  (Zigate) Device Options: {}
2018-11-18 22:27:13.480  (Zigate) Device:           4 - ID: 55, Name: 'Zigate - Humidité Salon', nValue: 59, sValue: '0'
2018-11-18 22:27:13.480  (Zigate) Device ID:       '55'
2018-11-18 22:27:13.480  (Zigate) Device Name:     'Zigate - Humidité Salon'
[... 77 lines cut ...]
2018-11-18 22:27:13.484  (Zigate) Device nValue:    0
2018-11-18 22:27:13.484  (Zigate) Device sValue:   '1013.1;3'
2018-11-18 22:27:13.484  (Zigate) Device LastLevel: 0
2018-11-18 22:27:13.484  (Zigate) Device Options: {}
2018-11-18 22:27:13.484  (Zigate) Device:           16 - ID: 67, Name: 'Zigate - Cube Salon', nValue: 0, sValue: '0'
2018-11-18 22:27:13.484  (Zigate) Device ID:       '67'
2018-11-18 22:27:13.484  (Zigate) Device Name:     'Zigate - Cube Salon'
2018-11-18 22:27:13.484  (Zigate) Device nValue:    0
2018-11-18 22:27:13.485  (Zigate) Device sValue:   '0'
2018-11-18 22:27:13.485  (Zigate) Device LastLevel: 0
2018-11-18 22:27:13.485  (Zigate) Device Options: {'LevelActions': '|||||||||', 'SelectorStyle': '0', 'LevelOffHidden': 'true', 'LevelNames': 'Off|Shake|Wakeup|Drop|90°|180°|Push|Tap|Rotation'}
2018-11-18 22:27:13.485  Status: (Zigate) load PluginConf
2018-11-18 22:27:13.485  (Zigate) PluginConf.txt = {'CrcCheck':'On','RemoveDevice':'0','sendDelay':'1','storeDiscoveryFrames':'0','LQI':'0','ForceCreationDevice':'0','NetworkScan':'21600','logFORMAT':'0'}
2018-11-18 22:27:13.486  (Zigate) CrcCheck: 1 sendDelay: 1 storeDiscoveryFrames: 0 logFORMAT: 0 LQI: 0 RemoveDevice: 0 ForceCreationDevice: 0 NetworkScan: 21600
2018-11-18 22:27:13.492  Status: (Zigate) load ListOfDevice
[... cut ...] ': '00158d00028f7824', 'NbEp': '3', 'App Version': '5'}
2018-11-18 22:27:13.543  Error: Domoticz received fatal signal 11 !...
2018-11-18 22:27:13.544  Error:   /home/pi/domoticz/domoticz() [0x262154]
2018-11-18 22:27:13.545  Error:   /home/pi/domoticz/domoticz(_Z14signal_handleri+0x4c) [0x26220c]
2018-11-18 22:27:13.545  Error:   /lib/arm-linux-gnueabihf/libc.so.6(__default_sa_restorer_v2+0) [0x76c4f180]

I can only restart domoticz when DeviceList-13.txt file is empty.
Thank you for your help.

pipiche38, 2018-11-18:
This looks pretty strange ... I copy-paste your devicelist-13.txt and use it on my system without any trouble.

From the log you attached, it looks like it failed with the 1st line of the file. Did you try to remove the first line ( which correspond to the entry 'feec' )

What is your python version ? as well as your Domoticz version ?

BornTKill, 2018-11-18:
Hello,

I try to restart domoticz and délétions one line by one line between each restart. But same issue :/

Python is 3.4.2
Domoticz is Version: 4.9700

I delete all devices again. I erased zigate. I delete plugin. I git clone bêta 3.2 again (no need to Upgrade IEEE). I join devices that are listed in domo but no DeviceList-13.txt is created now. Restart ok but devices unknow in plugin DB.

pipiche38, 2018-11-19:
Not sure I fully understand what you mean by start from scratch. 
Did you also delete the Domoticz DB ? 

I believe not. If that is the case, what happens if you put the file DeviceList-13.txt back ? As if you sill have the devices on Domoticz you need also the Plugin DB in order to make the connection between the 2 worlds.

If that works with the file back, then you need to re-pair each of the devices to Zigate and the plugin should be able to identify that those devices were already existing and make the connection

pipiche38, 2018-11-19:
@BornTKill  I have sent you a Keybase.io message

BornTKill, 2018-11-19:
Maybe we can speak in french ?

From scratch is only for plugin of course : delete devices in domo, persistent delete the zigate : in order to have the same state (no devices) in both world.
I leave the hardware in domoticz.
Then i delete plugin dir, then git clone beta 3.2 again.
I skip upgrade IEEE because it’s not a migration (and my domoticz.db was already migrate before).
I pair devices successfully but de DeviceList-13.txt (plugin DB if I have well understand) is not

pipiche38, 2018-11-19:
@could you reach me on keybase ? @pipiche

BornTKill, 2018-11-22:
Hello,

Python 3.5 solved the issue. Yes 1/3 price but chinese cloud powered.

pipiche38, 2018-11-22:
@phfu 

> Hello
> Same bug here : domoticz crashes after restart.
> 
> Configuration :
> Raspberry pi 2,
> Domoticz Version : 4.9700
> Python Version: 3.4.2
> Zigate firmware : ZiGate_Coordinator_v3.0e.bin
> 
> How it appeared :
> I installed beta beta-3.2 Zigate plugin. No previous material / devices in Domoticz. No database upgrade.
> I was able to add 5 Xiaomi devices, without any problem.
> Domoticz was working fine with zigate and the sensors were updating.
> 
> Problem :
> When I restart the Raspberry (sudo reboot), Domoticz restarts, but crashes shortly.
> Here is the log of the crash :
> 2018-11-21 23:29:31.597 Error: Domoticz received fatal signal 11 !...
> 2018-11-21 23:29:31.606 Error: /home/pi/domoticz/domoticz() [0x262154]
> 2018-11-21 23:29:31.606 Error: /home/pi/domoticz/domoticz(_Z14signal_handleri+0x4c) [0x26220c]
> 2018-11-21 23:29:31.606 Error: /lib/arm-linux-gnueabihf/libc.so.6(__default_sa_restorer_v2+0) [0x76c7d180]
> 
> See also complete log file attached.
> 
> When I suppress the plugin directory and restart the Raspberry, it starts fine again (with Error: Zigate hardware (23) thread seems to have ended unexpectedly), which is normal since I suppressed the directory but not the material.
> 
> The previous version was working better, although I had loss of devices after a few days.
> 
> [domoticz_log.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2609283/domoticz_log.txt)
> 
> Just order an official Xiaomi Gateway... Black friday prices made it 1/3 of the zigate price...

Do you have other Python based plugin running ?

Could you also have a look to : HTTP://www.domoticz.com/forum/viewtopic.php?p=198354#p198354

Please cross-check the Python environment as 3.4 seems to have some specials requirements !

pipiche38, 2018-11-23:
Similar problem on a different plugin https://github.com/domoticz/domoticz/issues/2879

phfu, 2018-11-23:
Hello.
I installed python v3.6 on my raspberry (compiled and all)
python3 -V gives me Python 3.6.5 

I tried everything, looked on many forums, but I cannot change the Python version shown in the status of Domoticz...  3 hours lost this evening...

Any hint ?
is the python version directly linked at Domoticz compilation ?  I used standard packaged installation

Thanks in advance
Philippe

BornTKill, 2018-11-24:
Same here. It seems that you have to compile with python 3.6

https://www.domoticz.com/forum/viewtopic.php?t=23829

zaraki673, 2018-11-24:
You don't have to compile, standard package is enough,

What is our OS ? (rasbian, debian ... windows :( )
For many linux system you could follow this how to, who explain how choose
default python on our system :
https://linuxconfig.org/how-to-change-default-python-version-on-debian-9-stretch-linux

BornTKill, 2018-11-24:
Already tried but not working.

BornTKill, 2018-12-02:
Hello.

As post mortem, here what has been done to be able to use bêta 3.x

Migrate to a newer distribution with python 3.5.1 available in main repository.
Dedicate USB port (no USB hub) to avoid lost communication between Zigate and plugin.

Now everything work like a charm.
Thank to @pipiche and the community.

pipiche38, 2018-12-02:
Thanks we will advise Python 3.5 and above (as I would tend to agree that Python 3.4 seems to be more challenging and most-likely due to recent Domoticz version)

Closing the issue

closed 2018-12-02 by pipiche38

phfu, 2018-12-05:
Hello
Just to let you know that I took some time to re-install all my Raspberry Domoticz system.
Python 3.5+ and latest Zigate Plugin version are running fine since a few days now.
This issue is definitively closed,   problem was linked with Python 3 version.

Merci beaucoup, Best regards

pipiche38, 2018-12-05:
@phfu  thanks for the update

BornTKill, 2018-12-05:
Glad for you @phfu