zigbeefordomoticz/Domoticz-Zigbee #254: Handle : lumi.remote.b186acn01

pipiche38, 2018-11-21:
**Describe the bug**
it would be great to handle the new Aqara Wireless Remote Switch Single devices

pipiche38, 2018-11-21:
2018-11-21 21:38:01.978 (Zigate) Decode8043 - Reception Simple descriptor response : EP : 01, Profile : 0104, Device Id : 5f01, Bit Field : 01
2018-11-21 21:38:01.979 (Zigate) Decode8043 - Reception Simple descriptor response : In Cluster Count : 05
2018-11-21 21:38:01.979 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0000
2018-11-21 21:38:01.980 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0003
2018-11-21 21:38:01.980 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0019
2018-11-21 21:38:01.981 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: ffff
2018-11-21 21:38:01.982 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0012
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Out Cluster Count : 07
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0000
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0003
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0019
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: ffff
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0012
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0700
2018-11-21 21:38:01.983 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0000
2018-11-21 21:38:05.206 (Zigate) Decode8043 - Reception Simple descriptor response : EP : 03, Profile : 0104, Device Id : 5f03, Bit Field : 01
2018-11-21 21:38:05.206 (Zigate) Decode8043 - Reception Simple descriptor response : In Cluster Count : 02
2018-11-21 21:38:05.206 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0003
2018-11-21 21:38:05.207 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 000c
2018-11-21 21:38:05.208 (Zigate) Decode8043 - Reception Simple descriptor response : Out Cluster Count : 04
2018-11-21 21:38:05.208 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0003
2018-11-21 21:38:05.209 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 000c
2018-11-21 21:38:05.209 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0400
2018-11-21 21:38:05.209 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0400

pipiche38, 2018-11-21:
8f1 => {
'Heartbeat': '1', 
'ZDeviceID': '5f03', 
'IEEE': '00158d00028cad60', 
'NbEp': '3', 
'LogicalType': 'End Device', 
'DeviceType': 'RFD',
 'Manufacturer': '1037', 
'RSSI': 144, 
'Status': '8043', 
'MacCapa': '80', 
'Type': {}, 
'ProfileID': '0104', 
'Ep': {
          '01': {'ClusterType': {}, '0019': {}, '0700': {}, '0003': {}, '0000': {}, 'ffff': {}, '0012': {}}, 
          '02': {'ClusterType': {}}, 
          '03': {'ClusterType': {}, '0003': {}, '0400': {}, '000c': {}}}, 
'App Version': '8', 
'ReceiveOnIdle': 'Off', 
'PowerSource': 'Battery', 
'Battery': {}, 
'RIA': '1', 
'Stamp': {'MsgType': '8102', 'Time': '2018-11-21 21:37:06'}, '
SQN': '01', 
'Version': '3', 
'Model': 'lumi.remote.b186acn01'}

pipiche38, 2018-11-21:
Seems that it is the same as lumi.sensor_86sw1

hkhk12, 2018-11-23:
Created a verbose log while pushing the lumi.remote.b186acn01 switch
[lumi.remote.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2611716/lumi.remote.txt)

When pressing the button, the state does not change in Domoticz. Scripts (dzvents) aren't triggered.

pipiche38, 2018-11-27:
@hkhk12 I have analysed your log and here are the current status.

(1) The lumi.remote.b186acn01 is correctly created.
'be54': {'SQN': '08', 'Ep': {'03': {}, '02': {}, '01': {'0012': '0002', '0000': {}, 'ClusterType': {'701': 'Button'}, '0006': {}}}, 'App Version': '8', 'PowerSource': 'Battery', 'Version': '3', 'Type': 'Button', 'MacCapa': '80', 'DeviceType': 'RFD', 'NbEp': '3', 'ProfileID': {}, 'IEEE': '00158d00028cad60', 'RSSI': 123, 'Battery': {}, 'ZDeviceID': {}, 'LogicalType': 'End Device', 'Heartbeat': '73', 'RIA': '0', 'Manufacturer': '1037', 'Stamp': {'MsgType': '8102', 'Time': '2018-11-23 21:48:52'}, 'ReceiveOnIdle': 'Off', 'Status': 'inDB', 'Model': 'lumi.remote.b186acn01'}

(2) It looks like it is using the 0x0012 Cluster to communicate its state instead of Cluster 0x0006

hkhk12, 2018-11-27:
@pipiche38 

installed beta3.3 and added button and Ikea led.
1. Ikea led now identifies as 2 devices (Light/Switch, Switch, On/Off and Light/Switch, Switch, Dimmer)
2. Aquara button is identified but doesn't trigger the button in Domoticz not dzvents scripts.
There is a log:
_ReadCluster - 0012 NwkId: 6352 Ep: 01 AttrId: 0055 AttyType: 21 Attsize: 0002 Status: 00 AttrValue: 0001_

Attached the log but if you need verbose log just let me know.

[zigate-aquara-switch.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2621722/zigate-aquara-switch.txt)

pipiche38, 2018-11-27:
Ikea led issue is now fixed on the latest version.

I have added some more debug in order to capture what is happening. Indeed We are receiving a Signal

hkhk12, 2018-11-27:
Ikea led indeed fixed.

No change on the button though

[zigate-aquara-switch.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2621897/zigate-aquara-switch.txt)

hkhk12, 2018-11-28:
Button now fixed too!
Thanks.

pipiche38, 2018-11-28:
Thanks for providing the logs.

closed 2018-11-28 by pipiche38