zigbeefordomoticz/Domoticz-Zigbee #259: 'onHeartbeat' failed 'KeyError' when adding a device

davidllerena, 2018-11-27:
I have update plugin to the last version at today (27/11/2018) 
When trying to add a new Philips Hue White Bulb I receive the 
`Error: (ZiGate) 'onHeartbeat' failed 'KeyError'.`

```
2018-11-27 22:54:41.101 Error: (ZiGate) 'onHeartbeat' failed 'KeyError'.
2018-11-27 22:54:41.101 Error: (ZiGate) ----> Line 280 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-11-27 22:54:41.101 Error: (ZiGate) ----> Line 240 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 276 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processListOfDevices
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 247 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processNotinDBDevices
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 42 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function CreateDomoDevice
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 519 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function GetType

```
---------------------------------
Complete log of 2 heartbeats:
-----------
2018-11-27 22:54:31.040 (ZiGate) Pushing 'onHeartbeatCallback' on to queue
2018-11-27 22:54:31.067 (ZiGate) Processing 'onHeartbeatCallback' message
2018-11-27 22:54:31.067 (ZiGate) Calling message handler 'onHeartbeat'.
2018-11-27 22:54:31.068 (ZiGate) ListOfDevices : {'670c': {'RSSI': 108, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '86', 'Type': 'Temp/Humi/Baro', 'Battery': 93, 'RIA': '13', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d000245b4b3', 'MacCapa': '80', 'Ep': {'01': {'0405': 62.5, '0403': 946.0, '0000': {}, '0402': 31.0}}}, 'a6eb': {'RSSI': 99, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '86', 'Type': 'Temp/Humi/Baro', 'Battery': {}, 'RIA': '7', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d000245f2d8', 'MacCapa': '80', 'Ep': {'01': {'0403': 950.2, '0000': {}, '0402': 22.6, '0405': 40.5}}}, 'c85d': {'Type': {}, 'Model': 'Ampoule.LED1622G12.Tradfri', 'Status': '8043', 'Heartbeat': '58', 'RSSI': {}, 'Battery': {}, 'RIA': '0', 'ProfileID': 'c05e', 'ZDeviceID': '0100', 'IEEE': '0017880104676979', 'MacCapa': '8e', 'Ep': {'f2': {'0021': {}}, '0b': {'0004': {}, '0008': {}, '0006': {}, '0005': {}, '0000': {}, '0003': {}, '1000': {}}}}, 'c75c': {'Type': {}, 'Model': {}, 'Status': '8045', 'Heartbeat': '2', 'RSSI': {}, 'Battery': {}, 'RIA': '0', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00178801040357cd', 'MacCapa': '8e', 'Ep': {'f2': {}, '0b': {}}}, '57c0': {'RSSI': 88, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '28', 'Type': 'Temp/Humi/Baro', 'Battery': {}, 'RIA': '2', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d0002320ac3', 'MacCapa': '80', 'Ep': {'01': {'0403': {}, '0000': {}, '0402': {}, '0405': {}}}}, '901f': {'RSSI': 147, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '28', 'Type': 'Temp/Humi/Baro', 'Battery': 91, 'RIA': '3', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d0001fa75d6', 'MacCapa': '80', 'Ep': {'01': {'0405': 39.4, '0000': {}, '0402': 22.8, '0403': 950.3}}}, '35bb': {'RSSI': 153, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '28', 'Type': 'Temp/Humi/Baro', 'Battery': 93, 'RIA': '8', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d000245dc78', 'MacCapa': '80', 'Ep': {'01': {'0403': 943.8, '0000': {}, '0402': 29.5, '0405': 72.8}}}}
2018-11-27 22:54:31.069 (ZiGate) onHeartbeat - creating device id : c85d with : {'Type': {}, 'Model': 'Ampoule.LED1622G12.Tradfri', 'Status': '8043', 'Heartbeat': '59', 'RSSI': {}, 'Battery': {}, 'RIA': '0', 'ProfileID': 'c05e', 'ZDeviceID': '0100', 'IEEE': '0017880104676979', 'MacCapa': '8e', 'Ep': {'f2': {'0021': {}}, '0b': {'0004': {}, '0008': {}, '0006': {}, '0005': {}, '0000': {}, '0003': {}, '1000': {}}}}
2018-11-27 22:54:31.069 Error: (ZiGate) 'onHeartbeat' failed 'KeyError'.
2018-11-27 22:54:31.069 Error: (ZiGate) ----> Line 280 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-11-27 22:54:31.069 Error: (ZiGate) ----> Line 240 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-11-27 22:54:31.069 Error: (ZiGate) ----> Line 276 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processListOfDevices
2018-11-27 22:54:31.069 Error: (ZiGate) ----> Line 247 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processNotinDBDevices
2018-11-27 22:54:31.069 Error: (ZiGate) ----> Line 42 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function CreateDomoDevice
2018-11-27 22:54:31.070 Error: (ZiGate) ----> Line 519 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function GetType
2018-11-27 22:54:41.044 (ZiGate) Pushing 'onHeartbeatCallback' on to queue
2018-11-27 22:54:41.090 (ZiGate) Processing 'onHeartbeatCallback' message
2018-11-27 22:54:41.092 (ZiGate) Calling message handler 'onHeartbeat'.
2018-11-27 22:54:41.099 (ZiGate) ListOfDevices : {'670c': {'RSSI': 108, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '87', 'Type': 'Temp/Humi/Baro', 'Battery': 93, 'RIA': '13', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d000245b4b3', 'MacCapa': '80', 'Ep': {'01': {'0405': 62.5, '0403': 946.0, '0000': {}, '0402': 31.0}}}, 'a6eb': {'RSSI': 99, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '87', 'Type': 'Temp/Humi/Baro', 'Battery': {}, 'RIA': '7', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d000245f2d8', 'MacCapa': '80', 'Ep': {'01': {'0403': 950.2, '0000': {}, '0402': 22.6, '0405': 40.5}}}, 'c85d': {'Type': {}, 'Model': 'Ampoule.LED1622G12.Tradfri', 'Status': '8043', 'Heartbeat': '59', 'RSSI': {}, 'Battery': {}, 'RIA': '0', 'ProfileID': 'c05e', 'ZDeviceID': '0100', 'IEEE': '0017880104676979', 'MacCapa': '8e', 'Ep': {'f2': {'0021': {}}, '0b': {'0004': {}, '0008': {}, '0006': {}, '0005': {}, '0000': {}, '0003': {}, '1000': {}}}}, 'c75c': {'Type': {}, 'Model': {}, 'Status': '8045', 'Heartbeat': '2', 'RSSI': {}, 'Battery': {}, 'RIA': '0', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00178801040357cd', 'MacCapa': '8e', 'Ep': {'f2': {}, '0b': {}}}, '57c0': {'RSSI': 88, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '28', 'Type': 'Temp/Humi/Baro', 'Battery': {}, 'RIA': '2', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d0002320ac3', 'MacCapa': '80', 'Ep': {'01': {'0403': {}, '0000': {}, '0402': {}, '0405': {}}}}, '901f': {'RSSI': 147, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '28', 'Type': 'Temp/Humi/Baro', 'Battery': 91, 'RIA': '3', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d0001fa75d6', 'MacCapa': '80', 'Ep': {'01': {'0405': 39.4, '0000': {}, '0402': 22.8, '0403': 950.3}}}, '35bb': {'RSSI': 153, 'Model': 'lumi.weather', 'Status': 'inDB', 'Heartbeat': '28', 'Type': 'Temp/Humi/Baro', 'Battery': 93, 'RIA': '8', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d000245dc78', 'MacCapa': '80', 'Ep': {'01': {'0403': 943.8, '0000': {}, '0402': 29.5, '0405': 72.8}}}}
2018-11-27 22:54:41.101 (ZiGate) onHeartbeat - creating device id : c85d with : {'Type': {}, 'Model': 'Ampoule.LED1622G12.Tradfri', 'Status': '8043', 'Heartbeat': '60', 'RSSI': {}, 'Battery': {}, 'RIA': '0', 'ProfileID': 'c05e', 'ZDeviceID': '0100', 'IEEE': '0017880104676979', 'MacCapa': '8e', 'Ep': {'f2': {'0021': {}}, '0b': {'0004': {}, '0008': {}, '0006': {}, '0005': {}, '0000': {}, '0003': {}, '1000': {}}}}
2018-11-27 22:54:41.101 Error: (ZiGate) 'onHeartbeat' failed 'KeyError'.
2018-11-27 22:54:41.101 Error: (ZiGate) ----> Line 280 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-11-27 22:54:41.101 Error: (ZiGate) ----> Line 240 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 276 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processListOfDevices
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 247 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py, function processNotinDBDevices
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 42 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function CreateDomoDevice
2018-11-27 22:54:41.102 Error: (ZiGate) ----> Line 519 in /home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py, function GetType

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

 - OS: Linux raspi 4.9.35-v7+
 - Version [e.g. 22]

**Plugin (please complete the following information):**
 - OS: [e.g. Linux Fedora 28]
 - Plugin version
 - Plugin branch

**Additional context**
Add any other context about the problem here.

pipiche38, 2018-11-27:
You mentioned that you are trying to add a new Philips Hue White Bulb, but I see the plugin reporting the creation of IKEA Tradfri LED ?

Did you touch the DeviceConf.txt file ? If so , could you kindly provide it ?

davidllerena, 2018-11-27:
There are 2 new devices, both are Philips Hue White.
'IEEE': '0017880104676979' and 'IEEE': '00178801040357cd'

I haven't modify the DeviceConf.txt. 
Plugin detected them as Ampoule.LED1622G12.Tradfri

davidllerena, 2018-11-27:
`1929 sep 23 00:18 DeviceConf.txt`

davidllerena, 2018-11-27:
**DeviceConf.txt**
```
{
'lumi.weather':{'Ep':{'01':{'0402','0405','0403'}},'Type':'Temp/Humi/Baro'},
'lumi.sensor_ht':{'Ep':{'01':{'0402','0405'}},'Type':'Temp/Humi'},
'lumi.sensor_magnet.aq2':{'Ep':{'01':{'0006'}},'Type':'Door'},
'lumi.sensor_magnet':{'Ep':{'01':{'0006'}},'Type':'Door'},
'lumi.sensor_motion':{'Ep':{'01':{'0006'}},'Type':'Motion'},
'lumi.sensor_switch.aq2':{'Ep':{'01':{'0006'}},'Type':'MSwitch'},
'lumi.sensor_switch':{'Ep':{'01':{'0006'}},'Type':'MSwitch'},
'lumi.sensor_86sw2':{'Ep':{'01':{'0006'},'02':{'0006'}},'Type':'DButton'},
'lumi.sensor_smoke':{'Ep':{'01':{'0006'}},'Type':'Smoke'},
'lumi.sensor_wleak.aq1':{'Ep':{'01':{'0006'}},'Type':'Water'},
'lumi.sensor_86sw1':{'Ep':{'01':{'0006'}},'Type':'Button'},
'lumi.sensor_motion.aq2':{'Ep':{'01':{'0406','0400'}},'Type':'Motion/Lux'},
'lumi.sensor_cube':{'Ep':{'01':{'0012','000c'}},'Type':'XCube'},
'lumi.sensor_cube.aqgl01':{'Ep':{'01':{'0012','000c'}},'Type':'Aqara'},
'lumi.plug':{'Ep':{'01':{'0006','000c'}},'Type':'Plug/PowerMeter'},
'Ampoule.LED1545G12.Tradfri':{'Ep':{'01':{'0006','0008','0300'}},'Type':'Switch/LvlControl/ColorControl'},
'Ampoule.LED1622G12.Tradfri':{'Ep':{'01':{'0006','0008'}},'Type':'Switch/LvlControl'},
'Ampoule.phillips.hue':{'Ep':{'01':{'0006','0008'}},'Type':'Switch/LvlControl'},
'plug.Osram':{'Ep':{'03':{'0006'}},'Type':'Plug'},
'plug.Salus':{'Ep':{'09':{'0006'}},'Type':'Plug'},
'shutter.Profalux':{'Ep':{'01':{'0006','0008'}},'Type':'LvlControl'},
'Ampoule.LED1624G9.Tradfri':{'Ep':{'01':{'0006','0008','0300'}},'Type':'Switch/LvlControl/ColorControl'},
'Command.30338849.Tradfri':{'Ep':{'01':{'0000','0001','0009','0b05','1000'}},'Type':'Command'},
'PST03A-v2.2.5':{'Ep':{'01':{'0406':'','0500':'','0400':'','0402':'','Type':'Door/Switch/Lux/Temp'},'02':{'0500':'','Type':'Motion'}},'Type':'Motion/Door/Switch/Lux/Temp'},
'plug.Celiane.Legrand':{'Ep': {'01':{'0000','0003','0004','0006','0005','fc01'}},'Type':'Switch/PowerMeter'},
}

```

pipiche38, 2018-11-27:
Can you try to put Debug level to Verbose (**Python Only**) ?
And as the logs might be long, could you simply drop the file .

From what I suspect, is that this Bulbe provides 2 EndPoints while the Tradfri has only one configured.

davidllerena, 2018-11-27:
Which file do you need? domoticz.log?

pipiche38, 2018-11-28:
Yes, maybe not the full file, an extract would be enought from the last start.

pipiche38, 2018-11-28:
@davidllerena , could you kindly unzip that file and put it under Domoticz-Zigate folder then restart Domoticz.

[z_domoticz.py.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2624098/z_domoticz.py.zip)

yvin, 2018-11-28:
Hi,

I have the same kind of problem

Domoticz Version: 4.10234 (it was the same with previews version)

I add the zigate hardware
I clear the device on the zigate

I push the button of a xiaomi temperature sensor

I have a loop with this message

2018-11-28 20:04:14.737 (Zigate) processNotinDBDevices - ready for creation: {'Battery': {}, 'ReadAttributes': {'Ep': {'01': {'0001': {'0007': '86', '0021': '86', '0020': '00', '0002': '86', '0001': '86', '0005': '86', '0006': '86', '0003': '86', '0004': '86', '0000': '86'}}}}, 'IEEE': '00158d0002273acf', 'Type': '', 'Version': '3', 'App Version': '3', 'ZDeviceID': '5f01', 'NbEp': '1', 'Model': 'lumi.weather', 'RSSI': 118, 'RIA': '0', 'Status': '8043', 'SQN': '0b', 'Ep': {'01': {'Type': 'Temp/Humi/Baro', '0403': 1007.8, '0000': {}, '0001': {}, 'ffff': {}, '0402': 21.4, '0003': {}, 'ClusterType': {}, '0405': 51.4}}, 'Heartbeat': '39', 'ProfileID': '0104', 'Stamp': {'Time': '2018-11-28 20:01:34', 'MsgType': '8102'}, 'MacCapa': '80'}
2018-11-28 20:04:14.737 Status: (Zigate) [0] NEW OBJECT: 6833 Ox8043 received Infos
2018-11-28 20:04:14.737 Status: (Zigate) [0] NEW OBJECT: 6833 Trying to create Domoticz device(s)
2018-11-28 20:04:14.738 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-28 20:04:14.738 Error: (Zigate) ----> Line 290 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 20:04:14.738 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 20:04:14.738 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 20:04:14.738 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
2018-11-28 20:04:14.738 Error: (Zigate) ----> Line 96 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py', function CreateDomoDevice
2018-11-28 20:04:19.748 (Zigate) processNotinDBDevices - ready for creation: {'Battery': {}, 'ReadAttributes': {'Ep': {'01': {'0001': {'0007': '86', '0021': '86', '0020': '00', '0002': '86', '0001': '86', '0005': '86', '0006': '86', '0003': '86', '0004': '86', '0000': '86'}}}}, 'IEEE': '00158d0002273acf', 'Type': '', 'Version': '3', 'App Version': '3', 'ZDeviceID': '5f01', 'NbEp': '1', 'Model': 'lumi.weather', 'RSSI': 118, 'RIA': '0', 'Status': '8043', 'SQN': '0b', 'Ep': {'01': {'Type': 'Temp/Humi/Baro', '0403': 1007.8, '0000': {}, '0001': {}, 'ffff': {}, '0402': 21.4, '0003': {}, 'ClusterType': {}, '0405': 51.4}}, 'Heartbeat': '40', 'ProfileID': '0104', 'Stamp': {'Time': '2018-11-28 20:01:34', 'MsgType': '8102'}, 'MacCapa': '80'}
2018-11-28 20:04:19.747 Status: (Zigate) [0] NEW OBJECT: 6833 Ox8043 received Infos
2018-11-28 20:04:19.748 Status: (Zigate) [0] NEW OBJECT: 6833 Trying to create Domoticz device(s)
2018-11-28 20:04:19.749 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-28 20:04:19.749 Error: (Zigate) ----> Line 290 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 20:04:19.749 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 20:04:19.749 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 20:04:19.749 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
2018-11-28 20:04:19.749 Error: (Zigate) ----> Line 96 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py', function CreateDomoDevice
2018-11-28 20:04:24.762 (Zigate) processNotinDBDevices - ready for creation: {'Battery': {}, 'ReadAttributes': {'Ep': {'01': {'0001': {'0007': '86', '0021': '86', '0020': '00', '0002': '86', '0001': '86', '0005': '86', '0006': '86', '0003': '86', '0004': '86', '0000': '86'}}}}, 'IEEE': '00158d0002273acf', 'Type': '', 'Version': '3', 'App Version': '3', 'ZDeviceID': '5f01', 'NbEp': '1', 'Model': 'lumi.weather', 'RSSI': 118, 'RIA': '0', 'Status': '8043', 'SQN': '0b', 'Ep': {'01': {'Type': 'Temp/Humi/Baro', '0403': 1007.8, '0000': {}, '0001': {}, 'ffff': {}, '0402': 21.4, '0003': {}, 'ClusterType': {}, '0405': 51.4}}, 'Heartbeat': '41', 'ProfileID': '0104', 'Stamp': {'Time': '2018-11-28 20:01:34', 'MsgType': '8102'}, 'MacCapa': '80'}
2018-11-28 20:04:24.761 Status: (Zigate) [0] NEW OBJECT: 6833 Ox8043 received Infos
2018-11-28 20:04:24.761 Status: (Zigate) [0] NEW OBJECT: 6833 Trying to create Domoticz device(s)
2018-11-28 20:04:24.762 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-28 20:04:24.762 Error: (Zigate) ----> Line 290 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 20:04:24.762 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 20:04:24.762 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 20:04:24.762 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
2018-11-28 20:04:24.763 Error: (Zigate) ----> Line 96 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py', function CreateDomoDevice

I reboot domoticz
Impossible to access to the WEBGUI
Domoticz stays offline
I need to replace the database by a backup one to restart Domoticz

pipiche38, 2018-11-28:
@yvin which version of the Zigate plugin are you ?
Do you have any Zigate related devices already in Domoticz ?

Could you put the plugin in 'debug = verbose" mode and attached the corresponding logs ?

davidllerena, 2018-11-28:
> @davidllerena , could you kindly unzip that file and put it under Domoticz-Zigate folder then restart Domoticz.
> 
> [z_domoticz.py.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2624098/z_domoticz.py.zip)

Hi,

The patch worked fine. Good job!!!!!

Only one thing about the description of the bulbs. Remember that my bulbs where detected as Tradfri. This is because in DeviceConf.txt Tradfri is before Philips in the list of devices. As workaround I have move Ampoule.LED1622G12.Tradfri under Ampoule.phillips.hue.

**diff --git a/DeviceConf.txt b/DeviceConf.txt**
```
index 1e56983..4fb784c 100644
--- a/DeviceConf.txt
+++ b/DeviceConf.txt
@@ -15,8 +15,8 @@
 'lumi.sensor_cube.aqgl01':{'Ep':{'01':{'0012','000c'}},'Type':'Aqara'},
 'lumi.plug':{'Ep':{'01':{'0006','000c'}},'Type':'Plug/PowerMeter'},
 'Ampoule.LED1545G12.Tradfri':{'Ep':{'01':{'0006','0008','0300'}},'Type':'Switch/LvlControl/ColorControl'},
-'Ampoule.LED1622G12.Tradfri':{'Ep':{'01':{'0006','0008'}},'Type':'Switch/LvlControl'},
 'Ampoule.phillips.hue':{'Ep':{'01':{'0006','0008'}},'Type':'Switch/LvlControl'},
+'Ampoule.LED1622G12.Tradfri':{'Ep':{'01':{'0006','0008'}},'Type':'Switch/LvlControl'},
 'plug.Osram':{'Ep':{'03':{'0006'}},'Type':'Plug'},
 'plug.Salus':{'Ep':{'09':{'0006'}},'Type':'Plug'},
 'shutter.Profalux':{'Ep':{'01':{'0006','0008'}},'Type':'LvlControl'},

```

**Before**
```
...
creating device id : c85d with : {'Type': {}, 'Model': 'Ampoule.LED1622G12.Tradfri', 'Status': '8043', 'Heartbeat': '60', 'RSSI': {}, 'Battery': {}, 'RIA': '0', 'ProfileID': 'c05e', 'ZDeviceID': '0100', 'IEEE': '0017880104676979', 'MacCapa': '8e', 'Ep': {'f2': {'0021': {}}, '0b': {'0004': {}, '0008': {}, '0006': {}, '0005': {}, '0000': {}, '0003': {}, '1000': {}}}}
```

**After**
```
...
'c85d': {'Ep': {'0b': {'0005': {}, '0006': {}, '0000': {}, '0004': {}, '1000': {}, '0008': {}, '0003': {}}, 'f2': {'0021': {}}}, 'RSSI': 132, 'IEEE': '0017880104676979', 'ZDeviceID': '0061', 'RIA': '0', 'Status': 'inDB', 'MacCapa': '8e', 'Heartbeat': '68', 'Model': 'Ampoule.phillips.hue', 'ProfileID': 'a1e0', 'Battery': {}, 'Type': 'Switch/LvlControl/'}}
```

Many thanks!!!

Regards

pipiche38, 2018-11-28:
Thanks for the report. Not sure about DeviceConf order, as it is hard coded in the code to treat the hue as a tradfri ;-)

I’ll push the patch to the official branch.

pipiche38, 2018-11-28:
6b4fe1651959793572e9a764e590339407d4e0bb 
@yvin please open a different issue so we can track that one independently.

closed 2018-11-28 by pipiche38