zigbeefordomoticz/Domoticz-Zigbee #263: Error message when a new device is added

yvin, 2018-11-28:
Domoticz Version: 4.10234 (it was the same with previews version)
Zigate Beta-3.3 

I add the zigate hardware
I clear the device on the zigate
I don't have device in Domoticz

I push the button of a xiaomi temperature sensor

I have a loop with error message

only after Domoticz restart the device is added in DeviceList-16.txt
Domoticz stays offline I need to use a backup database

```
2018-11-28 22:03:33.220 (Zigate) Decode8102 - Individual Attribute response : [2bd5:01] ClusterID: 0403 AttributeID: 0014 Status: 00 Type: 28 Size: 0001 ClusterData: >ff<
2018-11-28 22:03:33.220 (Zigate) DeviceExist - Found in ListOfDevices with status = 8043
2018-11-28 22:03:33.220 (Zigate) Decode8102 : Attribute Report from 2bd5 SQN = 05 ClusterID = 0403 AttrID = 0014 Attribute Data = ff
2018-11-28 22:03:33.220 (Zigate) Device : 2bd5 MacCapa : 80 updating SQN to 05
2018-11-28 22:03:33.220 (Zigate) ReadCluster - MsgData lenght is: 26 out of 24+
2018-11-28 22:03:33.220 (Zigate) DeviceExist - Found in ListOfDevices with status = 8043
2018-11-28 22:03:33.220 (Zigate) ReadCluster - 0403 NwkId: 2bd5 Ep: 01 AttrId: 0014 AttyType: 28 Attsize: 0001 Status: 00 AttrValue: ff
2018-11-28 22:03:33.220 (Zigate) decodeAttribute( 28, ff)
2018-11-28 22:03:33.220 (Zigate) Cluster0403 - decoded value: from:ff to 255
2018-11-28 22:03:33.220 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-11-28 22:03:33.222 (Zigate) receiveData - new Data coming
2018-11-28 22:03:33.222 (Zigate) receiveData - MsgType: 8102
2018-11-28 22:03:33.222 (Zigate) ZigateRead - decoded data : 018102000f41052bd5010403001000290002275a7603 lenght : 44
2018-11-28 22:03:33.222 (Zigate) ZigateRead - MsgType: 8102, MsgLength: 000f, MsgCRC: 41, Data: 052bd5010403001000290002275a; RSSI: 76
2018-11-28 22:03:33.222 (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000f41052bd5010403001000290002275a7603
2018-11-28 22:03:33.222 (Zigate) Decode8102 - Individual Attribute response : [2bd5:01] ClusterID: 0403 AttributeID: 0010 Status: 00 Type: 29 Size: 0002 ClusterData: >275a<
2018-11-28 22:03:33.222 (Zigate) DeviceExist - Found in ListOfDevices with status = 8043
2018-11-28 22:03:33.222 (Zigate) Decode8102 : Attribute Report from 2bd5 SQN = 05 ClusterID = 0403 AttrID = 0010 Attribute Data = 275a
2018-11-28 22:03:33.222 (Zigate) Device : 2bd5 MacCapa : 80 updating SQN to 05
2018-11-28 22:03:33.223 (Zigate) ReadCluster - MsgData lenght is: 28 out of 24+
2018-11-28 22:03:33.223 (Zigate) DeviceExist - Found in ListOfDevices with status = 8043
2018-11-28 22:03:33.223 (Zigate) ReadCluster - 0403 NwkId: 2bd5 Ep: 01 AttrId: 0010 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 275a
2018-11-28 22:03:33.223 (Zigate) decodeAttribute( 29, 275a)
2018-11-28 22:03:33.223 (Zigate) Cluster0403 - decoded value: from:275a to 10074
2018-11-28 22:03:33.223 (Zigate) MajDomoDevice - Device ID : 00158d0002273acf - Device EP : 01 - Type : 0403 - Value : 1007.4 - Hue :
2018-11-28 22:03:33.223 (Zigate) MajDomoDevice - Type = Baro
2018-11-28 22:03:33.223 (Zigate) ReadCluster - ClusterId=0403 - 0010 reception atm: 1007.4
2018-11-28 22:03:33.223 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-11-28 22:03:35.030 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-11-28 22:03:35.031 (Zigate) processNotinDBDevices - ready for creation: {'Version': '3', 'Type': '', 'RSSI': 118, 'MacCapa': '80', 'NbEp': '1', 'RIA': '0', 'Battery': {}, 'Model': 'lumi.weather', 'Ep': {'01': {'0000': {}, 'Type': 'Temp/Humi/Baro', '0405': 50.1, 'ClusterType': {}, '0001': {}, '0403': 1007.4, '0003': {}, 'ffff': {}, '0402': 21.5}}, 'ReadAttributes': {'Ep': {'01': {'0001': {'0000': '86', '0007': '86', '0004': '86', '0020': '00', '0021': '86', '0006': '86', '0001': '86', '0005': '86', '0003': '86', '0002': '86'}}}}, 'ProfileID': '0104', 'Stamp': {'MsgType': '8102', 'Time': '2018-11-28 22:03:33'}, 'SQN': '05', 'App Version': '3', 'Heartbeat': '4', 'Status': '8043', 'IEEE': '00158d0002273acf', 'ZDeviceID': '5f01'}
2018-11-28 22:03:35.031 (Zigate) CreatDomoDevice - Process EP : 01
2018-11-28 22:03:35.031 (Zigate) CreateDomoDevice - Type via ListOfDevice: ['Temp', 'Humi', 'Baro'] Ep : 01
2018-11-28 22:03:35.031 (Zigate) CreateDomoDevice - Creating devices based on Type: ['Temp', 'Humi', 'Baro']
2018-11-28 22:03:35.031 (Zigate) FreeUnit - device 1 available
2018-11-28 22:03:35.030 Status: (Zigate) [0] NEW OBJECT: 2bd5 Ox8043 received Infos
2018-11-28 22:03:35.030 Status: (Zigate) [0] NEW OBJECT: 2bd5 Trying to create Domoticz device(s)
2018-11-28 22:03:35.031 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-28 22:03:35.031 Error: (Zigate) ----> Line 290 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:03:35.031 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:03:35.031 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 22:03:35.031 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
2018-11-28 22:03:35.031 Error: (Zigate) ----> Line 96 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py', function CreateDomoDevice
2018-11-28 22:03:39.757 (Zigate) receiveData - new Data coming
2018-11-28 22:03:39.757 (Zigate) receiveData - MsgType: 8102
2018-11-28 22:03:39.757 (Zigate) ZigateRead - decoded data : 018102000f44092bd5010402000000290002086d7603 lenght : 44
2018-11-28 22:03:39.757 (Zigate) ZigateRead - MsgType: 8102, MsgLength: 000f, MsgCRC: 44, Data: 092bd5010402000000290002086d; RSSI: 76
2018-11-28 22:03:39.757 (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000f44092bd5010402000000290002086d7603
2018-11-28 22:03:39.757 (Zigate) Decode8102 - Individual Attribute response : [2bd5:01] ClusterID: 0402 AttributeID: 0000 Status: 00 Type: 29 Size: 0002 ClusterData: >086d<
2018-11-28 22:03:39.757 (Zigate) DeviceExist - Found in ListOfDevices with status = 8043
2018-11-28 22:03:39.757 (Zigate) Decode8102 : Attribute Report from 2bd5 SQN = 09 ClusterID = 0402 AttrID = 0000 Attribute Data = 086d
2018-11-28 22:03:39.757 (Zigate) Device : 2bd5 MacCapa : 80 updating SQN to 09
2018-11-28 22:03:39.757 (Zigate) updSQN - Device : 2bd5 updating SQN to 09
2018-11-28 22:03:39.757 (Zigate) ReadCluster - MsgData lenght is: 28 out of 24+
2018-11-28 22:03:39.757 (Zigate) DeviceExist - Found in ListOfDevices with status = 8043
2018-11-28 22:03:39.757 (Zigate) ReadCluster - 0402 NwkId: 2bd5 Ep: 01 AttrId: 0000 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 086d
[... 92 lines cut ...]
2018-11-28 22:03:45.029 (Zigate) CreateDomoDevice - Creating devices based on Type: ['Temp', 'Humi', 'Baro']
2018-11-28 22:03:45.029 (Zigate) FreeUnit - device 1 available
2018-11-28 22:03:45.028 Status: (Zigate) [0] NEW OBJECT: 2bd5 Ox8043 received Infos
2018-11-28 22:03:45.028 Status: (Zigate) [0] NEW OBJECT: 2bd5 Trying to create Domoticz device(s)
2018-11-28 22:03:45.029 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-28 22:03:45.029 Error: (Zigate) ----> Line 290 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:03:45.029 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:03:45.029 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 22:03:45.029 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
2018-11-28 22:03:45.029 Error: (Zigate) ----> Line 96 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py', function CreateDomoDevice
2018-11-28 22:03:50.040 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-11-28 22:03:50.040 (Zigate) processNotinDBDevices - ready for creation: {'Version': '3', 'Type': '', 'RSSI': 115, 'MacCapa': '80', 'NbEp': '1', 'RIA': '0', 'Battery': {}, 'Model': 'lumi.weather', 'Ep': {'01': {'0000': {}, 'Type': 'Temp/Humi/Baro', '0405': 50.2, 'ClusterType': {}, '0001': {}, '0403': 1007.4, '0003': {}, 'ffff': {}, '0402': 21.6}}, 'ReadAttributes': {'Ep': {'01': {'0001': {'0000': '86', '0007': '86', '0004': '86', '0020': '00', '0021': '86', '0006': '86', '0001': '86', '0005': '86', '0003': '86', '00 [... cut ...]
2018-11-28 22:03:50.041 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:03:50.041 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 22:03:50.041 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
2018-11-28 22:03:50.041 Error: (Zigate) ----> Line 96 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py', function CreateDomoDevice
```

yvin, 2018-11-28:
New log

```
 2018-11-28 22:39:03.109 (Zigate) Decode8043 - Reception Simple descriptor response : EP : 01, Profile : 0104, Device Id : 5f01, Bit Field : 01
2018-11-28 22:39:03.109 (Zigate) Decode8043 - Reception Simple descriptor response : In Cluster Count : 06
2018-11-28 22:39:03.109 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0000
2018-11-28 22:39:03.110 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0003
2018-11-28 22:39:03.110 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: ffff
2018-11-28 22:39:03.111 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0402
2018-11-28 22:39:03.111 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0403
2018-11-28 22:39:03.111 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0405
2018-11-28 22:39:03.111 (Zigate) Decode8043 - Reception Simple descriptor response : Out Cluster Count : 03
2018-11-28 22:39:03.111 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0000
2018-11-28 22:39:03.111 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0003
2018-11-28 22:39:03.111 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: ffff
2018-11-28 22:39:03.723 (Zigate) Out of sequence for Device: 9101 SQN move from 00 to 05 gap of : 5
2018-11-28 22:39:03.723 (Zigate) ReadCluster - 0402 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 0881
2018-11-28 22:39:03.724 (Zigate) ReadCluster - 0405 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 21 Attsize: 0002 Status: 00 AttrValue: 18f4
2018-11-28 22:39:03.726 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 03ef
2018-11-28 22:39:03.728 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0014 AttyType: 28 Attsize: 0001 Status: 00 AttrValue: ff
2018-11-28 22:39:03.729 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0010 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 2758
2018-11-28 22:39:07.401 (Zigate) processNotinDBDevices - ready for creation: {'SQN': '07', 'RIA': '0', 'Ep': {'01': {'0003': {}, '0001': {}, 'ffff': {}, '0000': {}, 'Type': 'Temp/Humi/Baro', '0403': 1007.2, '0402': 21.8, '0405': 63.9}}, 'Version': '3', 'IEEE': '00158d0002273acf', 'App Version': '3', 'ZDeviceID': '5f01', 'MacCapa': '80', 'Model': 'lumi.weather', 'Heartbeat': '1', 'Type': '', 'Stamp': {'MsgType': '8102', 'Time': '2018-11-28 22:39:03'}, 'Battery': {}, 'NbEp': '1', 'RSSI': 115, 'ProfileID': '0104', 'ReadAttributes': {'Ep': {'01': {'0001': {'0003': '86', '0001': '86', '0002': '86', '0007': '86', '0004': '86', '0000': '86', '0021': '86', '0006': '86', '0005': '86', '0020': '00'}}}}, 'Status': '8043'}
2018-11-28 22:39:07.453 (Zigate) ReadCluster - 0402 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 0881
2018-11-28 22:39:07.456 (Zigate) ReadCluster - 0405 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 21 Attsize: 0002 Status: 00 AttrValue: 165b
2018-11-28 22:39:07.457 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 03ef
2018-11-28 22:39:07.458 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0014 AttyType: 28 Attsize: 0001 Status: 00 AttrValue: ff
2018-11-28 22:39:07.459 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0010 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 2757
2018-11-28 22:39:07.400 Status: (Zigate) [0] NEW OBJECT: 9101 Ox8043 received Infos
2018-11-28 22:39:07.400 Status: (Zigate) [0] NEW OBJECT: 9101 Trying to create Domoticz device(s)
2018-11-28 22:39:07.401 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
2018-11-28 22:39:07.401 Error: (Zigate) ----> Line 290 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:39:07.401 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:39:07.401 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 22:39:07.401 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
2018-11-28 22:39:07.401 Error: (Zigate) ----> Line 99 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_domoticz.py', function CreateDomoDevice
2018-11-28 22:39:11.226 (Zigate) ReadCluster - 0402 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 087c
2018-11-28 22:39:11.228 (Zigate) ReadCluster - 0405 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 21 Attsize: 0002 Status: 00 AttrValue: 1534
2018-11-28 22:39:11.231 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0000 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 03ef
2018-11-28 22:39:11.235 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0014 AttyType: 28 Attsize: 0001 Status: 00 AttrValue: ff
2018-11-28 22:39:11.238 (Zigate) ReadCluster - 0403 NwkId: 9101 Ep: 01 AttrId: 0010 AttyType: 29 Attsize: 0002 Status: 00 AttrValue: 2757
2018-11-28 22:39:12.391 (Zigate) processNotinDBDevices - ready for creation: {'SQN': '02', 'RIA': '0', 'Ep': {'01': {'0003': {}, '0001': {}, 'ffff': {}, 'ClusterType': {}, '0000': {}, 'Type': 'Temp/Humi/Baro', '0403': 1007.1, '0402': 21.7, '0405': 54.3}}, 'Version': '3', 'IEEE': '00158d0002273acf', 'App Version': '3', 'ZDeviceID': '5f01', 'MacCapa': '80', 'M [... cut ...]
2018-11-28 22:39:37.401 Error: (Zigate) ----> Line 290 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:39:37.401 Error: (Zigate) ----> Line 239 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function onHeartbeat
2018-11-28 22:39:37.401 Error: (Zigate) ----> Line 244 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processListOfDevices
2018-11-28 22:39:37.401 Error: (Zigate) ----> Line 189 in '/home/pi/domoticz/plugins/Domoticz-Zigate/z_heartbeat.py', function processNotinDBDevices
```

pipiche38, 2018-11-29:
Hello,

Could you kindly do the same but with Debug = Verbose 

Patrick

pipiche38, 2018-11-29:
bonsoir @yvin 

I just push an update on beta-3.3 which should help in your case when the Creation of Widget on domoticz failed.

pipiche38, 2018-12-01:
@yvin any news ?

yvin, 2018-12-02:
Hi

No problem to add a device.
But after I have added a device if I reboot Domoticz (Stop & Start)
Domoticz stay Offline
I need to remove the DeviceList-16.txt to restart normally Domoticz

My DeviceList-16.txt

`1bf2 : {'RIA': '0', 'App Version': '3', 'Heartbeat': '8', 'Status': 'inDB', 'SQN': '0a', 'ProfileID': '0104', 'DeviceType': 'RFD', 'Stamp': {'MsgType': '8102', 'Time': '2018-12-01 19:38:57'}, 'NbEp': '1', 'ReadAttributes': {'Ep': {'01': {'0001': {'0006': '86', '0001': '86', '0007': '86', '0004': '86', '0020': '00', '0003': '86', '0005': '86', '0000': '86', '0002': '86', '0021': '86'}}}}, 'PowerSource': 'Battery', 'Version': '3', 'Model': 'lumi.weather', 'Ep': {'01': {'ClusterType': {'596': 'Temp', '595': 'Temp+Hum', '597': 'Humi', '594': 'Temp+Hum+Baro', '598': 'Baro'}, '0001': {}, '0405': 67.6, '0403': 998.4, '0003': {}, '0402': 22.1, '0000': {}, 'ffff': {}, 'Type': 'Temp/Humi/Baro'}}, 'LogicalType': 'End Device', 'MacCapa': '80', 'Battery': {}, 'RSSI': 91, 'ZDeviceID': '5f01', 'Manufacturer': '1037', 'IEEE': '00158d0002273acf', 'ReceiveOnIdle': 'Off', 'Type': ''}`

pipiche38, 2018-12-02:
If you remove the file DeviceList-16.txt you are removing the Zigate plugin database, and the plugin is not going to work. 
Do you have several USB stick on your system ?

pipiche38, 2018-12-02:
It would also be interesting to have a look to the Domoticz log file, why it is not starting ?

pipiche38, 2018-12-08:
Could you give us some update and answer the following questions:
(1) What is the python version you are using 
(2) Could you cross-check that Domoticz is using the right python version in case you have python2 and python3 installed 
(3) In case of problem, please provide the Logs

pipiche38, 2018-12-10:
I'm closing that issue. I suggest that you report the issue on the Forum if the problem is still there.
Please cross-check Python version

closed 2018-12-10 by pipiche38

yvin, 2018-12-10:
Ok to close it but I still have the problem

```
pi@domoticzpi ~ $ python --version
Python 2.7.9
pi@domoticzpi ~ $ python3 --version
Python 3.4.2
```

How to have the information of the log of Domoticz
If I have a device on the device list, Domoticz doesn't start

pipiche38, 2018-12-11:
@yvin We strongly recommend Python 3.5 for the plugin.
You can check here: https://github.com/sasu-drooz/Domoticz-Zigate/wiki/Plugin-Installation

And I would also recommend to check on the Domoticz Forum, where there is a Python section
https://www.domoticz.com/forum/viewforum.php?f=65