zigbeefordomoticz/Domoticz-Zigbee #264: Ikea on/off switch/remote E1743

deennoo, 2018-11-29:
there cluster for this device : 

​
Type: 0x8043 (Simple Descriptor Response)
  SQN: 0x83
  Status: 0x00
  Short Address: 0x80A5
  Length: 36
  EndPoint: 0x01
  Profile ID: 0xC05E (ZigBee LL)
  Device ID: 0x0820 (Unknown)
  Input Cluster Count: 7
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0x0001 (General: Power Config)
    Cluster 2:  Cluster ID: 0x0003 (General: Identify)
    Cluster 3:  Cluster ID: 0x0009 (General: Alarms)
    Cluster 4:  Cluster ID: 0x0102 (Unknown)
    Cluster 5:  Cluster ID: 0x1000 (ZLL: Commissioning)
    Cluster 6:  Cluster ID: 0xFC7C (Unknown)
  Output Cluster Count: 7
    Cluster 0:  Cluster ID: 0x0003 (General: Identify)
    Cluster 1:  Cluster ID: 0x0004 (General: Groups)
    Cluster 2:  Cluster ID: 0x0006 (General: On/Off)
    Cluster 3:  Cluster ID: 0x0008 (General: Level Control)
    Cluster 4:  Cluster ID: 0x0019 (General: OTA)
    Cluster 5:  Cluster ID: 0x0102 (Unknown)
    Cluster 6:  Cluster ID: 0x1000 (ZLL: Commissioning)

pipiche38, 2018-12-29:
@deenno would you mind posting a picture of this swicth ?
Do I understand that the state of this remote is visible in Domoticz as a switch ?

pipiche38, 2019-02-07:
@deennoo , can you give me a picture of this remote ?
What are the possible commands ?
How should we map to a domoticz Widget ? Simple switch (working as a single button) ?

pipiche38, 2019-02-07:
9b04def..e3c590b

deennoo, 2019-02-07:
Pic of the product : https://www.numerama.com/content/uploads/2018/09/capture-decran-2018-09-10-a-11-38-39.png

Running Last Beta 4.1 + pre release 3.0f fw

That just an ON/Off switch nothing else is allow, so a simple on/off switch is the good widget
Plugin create a level switch, who doesn't react to the switch

pipiche38, 2019-02-13:
@deennoo could you update us on the status on that one, based on the latest pre-4.1

deennoo, 2019-02-16:
Nothing better with pre 4.1

Still a level switch is created (instead off a simple On/Off)

Device discovery can be found here : 

https://drive.google.com/open?id=1hAMvEm4w8-7c7Icb_58OT98eUnPh9eoX

And the switch once on the Zigbee network still drive a left relay on a xiaomi dual neutral fire switch

pipiche38, 2019-02-16:
{
    "004d": "38fe000d6ffffe9a746980",
    "8043": {
        "01": "5d0038fe2401010408200007000000010003000900201000fc7c070003000400060008001901021000"
    },
    "8045": "5c0038fe0101",
    "Attribute Discovery": {
        "Ep": {
            "01": {
                "0001": {
                    "fffd": "21"
                }
            }
        }
    },
    "Ep": {
        "01": {
            "0000": {},
            "0001": {},
            "0003": {},
            "0004": {},
            "0006": {},
            "0008": {},
            "0009": {},
            "0019": {},
            "0020": {},
            "0102": {},
            "1000": {},
            "ClusterType": {
                "10166": "LvlControl"
            },
            "Type": "Switch",
            "fc7c": {}
        }
    },
    "IEEE": "000d6ffffe9a7469",
    "MacCapa": "80",
    "Manufacturer": "IKEA of Sweden",
    "Model": "TRADFRI on/off switch",
    "NbEP": "1",
    "PowerSource": "3",
    "ProfileID": {
        "0104": "01"
    },
    "ZCL_Version": "3",
    "ZDeviceID": {
        "0820": "01"
    }
}

pipiche38, 2019-02-16:
> And the switch once on the Zigbee network still drive a left relay on a xiaomi dual neutral fire switch

I need the log when you are making action on the switch, in order to see what is triggered on the plugin side.

pipiche38, 2019-02-16:
Alternative if you can capture what are the message sent by the switch, via ZGUI ... This would be also usefull

deennoo, 2019-02-24:
This switch looks to be a group driver only.

After a reset of a xiaomi 230 fire+line switch, and including IKEA ON/OFF switch to the network, a group call : 0000 appear

On this group 0000 where enrolling : 
 - IKEA on/off Switch
 - Xiaomi 230v Plug
 - Xiaomi Fire + line double switch

result : action on IKEA Switch where reported on Xiaomi device.

I had to earse Group 0000 on Domoticz, this action erase group 0000 on xiaomi devices

Xiaomi Fire + line switch need to erease Group from ZWGUI

After This i can create a group test with an IKEA ON/OFF switch and a Ikea Bulb using ZIgate as coordinateur and all works fine

I try to includ Zigate Ieee as this done for ikea round remote, but widget created for the switch isn't update for now.

pipiche38, 2019-02-24:
@deennoo , most -likely the plugin receive the update from the Zigate/Switch but are not managed.
Just push 715f6b4c49d9292f7c901c8a99b3643555a557c3 with some debuging .

Please try again and watch the Log

deennoo, 2019-04-17:
Just test it on 4.2 beta : 

OTA Upgrade is ok

But Switch still works on group 0000

I can bind it on the zigate (5 back button press in less than 5 second)
Create  groupe with Zigate and a bulb
Group react to : 
 - On 
 - Off
 - On long press : dimmer Up
 - Off long press : dimmer Down

pipiche38, 2019-04-20:
Looks like the swicth can be either:
- paired to a device
- parired to Zigate

But not to the both at the same time.

In case it is paired with Zigate, we have the creation of a Switch and the On and Off event are reported via message type 0x8095.

Merge in beta: a9eac89

pipiche38, 2019-04-20:
For On long press we have : 
Cluster : 0008 Cmd 05
when Release after long press we have
Cluster: 0008 Cmd 07

For Off long press : 
Cluster 0008 Cmd 01
when Release after long press we have 07

pipiche38, 2019-04-22:
On/Off implemented for 4.2

closed 2019-04-22 by pipiche38

ganaux, 2019-08-21:
Hi,
I'd like to add a request for this switch.
on and off work well for this switch (domoticz 4.11.215 and zigate plugin 4.5.3).

The switch can send command for long press too but no command are send to domoticz.

in domoticz log with plugin debug i can see : 

for up button long press and release : 
2019-08-21 18:05:35.584 (zigate) Decode8085 - SQN: 15, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 05, Unknown: 02
2019-08-21 18:05:35.935 (zigate) Decode8085 - SQN: 16, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 07, Unknown: 02

for down button long press and release : 
2019-08-21 18:07:43.418 (zigate) Decode8085 - SQN: 1b, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 01, Unknown: 02
2019-08-21 18:07:44.876 (zigate) Decode8085 - SQN: 1c, Addr: 6fcc, Ep: 01, Cluster: 0008, Cmd: 07, Unknown: 02

(the release event is the same for up and down button).

So the plugin receive the long press from zigate but don't send any event to domoticz object.

Is there any way to send this events (long press) to domoticz object ?

pipiche38, 2020-12-23:
@ganaux this should work on branche stable-fix of today.
Sorry didn't see your comment which is more than a 1.5 year old !