zigbeefordomoticz/Domoticz-Zigbee #269: factory resetting the Philips Hue bulb without the Hue Bridge

dpocock, 2018-12-01:
**Is your feature request related to a problem? Please describe.**
Other sources of documentation suggest that Philips Hue bulbs can only be factory reset with a Hue bridge.

**Describe the solution you'd like**
Use the Zigate to send a factory reset command to any Philips Hue bulb, even if it has already been used with another Hue bridge that is no longer available.

**Describe alternatives you've considered**
Using other brands of bulb, e.g. OSRAM.

**Additional context**
Many used Philips Hue bulbs are available on ebay but they can't be used without buying a Hue bridge.

Smanar, 2018-12-02:
But the zigate can send command only on device who are on the same network.
You can use the remote too.

dpocock, 2018-12-02:
Thanks for that feedback.  I don't have a Philips Hue remote and I wasn't aware of that.

Is this the method you are suggesting?  If so, buying a Philips Hue remote is a viable solution for me.
https://www.youtube.com/watch?v=qvlEAELiJKs

Whatever the remote is doing in this case, could the Zigate be programmed to emulate it?  Or could an SDR such as LimeSDR take the place of the remote?

pipiche38, 2018-12-02:
@dpocock don't know your french, but here are some tips for reseting http://faire-ca-soi-meme.fr/zigate/2017/11/27/maitrisez-vos-ampoules-zigbee/#Factory_Reset

Smanar, 2018-12-02:
Some devices have a "touchlink" feature,  to communicate with near by located devices but I m looking the ZGWUI application and I don't see something about that on zigate.
So yet, in your case I think the remote is the better solution.

pipiche38, 2018-12-03:
On the other side, broadcasting is a possibility, but 
(1) what is the message to send for a Reset 
(2) You can broadcast only to Router and End Devices with Receive on Iddle on
(3) You might Reset non Philipps Hue devices !

Smanar, 2018-12-03:
I have search LimeSDR on google but I only found realy expensive cards. You have a link to see what is it ?

dpocock, 2018-12-03:
[Here it is](https://www.crowdsupply.com/lime-micro/limesdr) - $299 is expensive for Zigbee, but it is very cheap if you want to implement multiple protocols.  Now there is [LimeSDR Mini for $159](https://www.crowdsupply.com/lime-micro/limesdr-mini) too.

We had a brief [discussion](https://lists.gnu.org/archive/html/discuss-gnuradio/2018-11/msg00020.html) in the GNU Radio mailing list.

pipiche38, 2018-12-07:
@dpocock are you ok to close this issue ? As I don't think much we can provide

closed 2018-12-08 by pipiche38

dpocock, 2018-12-16:
I ordered a Hue remote, it arrived and it resets the bulbs, thanks for that feedback.