zigbeefordomoticz/Domoticz-Zigbee #272: Update Device Only if needed

deennoo, 2018-12-06:
When recieving a reporting from a bulb, even if the device is already on the same state, the plugin update the status.

Steps to reproduce the behavior:
Add an ikea Bulb to the plugin
Turn it Off
and after 1 hour looks at the Logs on Domoticz Device

**Expected behavior**
Update device status only when it's needed

**Screenshots**

2018-12-06 20:44:22 | Off
2018-12-06 20:39:22 | Off
2018-12-06 20:34:22 | Off
2018-12-06 20:29:22 | Off

**Plugin  (please complete the following information):**
 - OS: Debian Stretch
 - Plugin version : 4.0.0 stable 
 - Ikea Bulb : Tradfri 980 lm
 - Domoticz : Version: 4.10256

pipiche38, 2018-12-07:
In fact this is a bit more tricky and apply for all objects which have configure reporting enabled on clusters 0x0006 and 0x0008 and which are defined in Domoticz in 1 single widget .

The main issue in linked to the fact that Domoticz store On/Off and Level in one single datafield sValue.

So I suspect that you get Off but also Level update and that is where the mix is happening.

pipiche38, 2018-12-08:
Problem occured only for Color bulb. Fix to be done in UpdateDevice

pipiche38, 2018-12-10:
Implemented in dev-grpmgt branch

closed 2018-12-10 by pipiche38