zigbeefordomoticz/Domoticz-Zigbee #273: Hue bulbs discovered, but inactive.

santricity, 2018-12-07:
Hi,
I bought Zigate USB stick, successfully installed Zigate Plugin on my Raspberry-Domoticz, added Osram Smart Plug, Ikea Tradfri bulbs and Philips Hue bulbs. 

I can control Osram and Ikea devices, but I can't do anything with Hue bulbs. 

[image]

As you can see the RF Signal Level is also missing for Hue bulbs. Bulbs are brand new, never paired with anything else. I assume I don't need to reset them before pairing them with ZiGate/Domoticz? 

Is there a special procedure for pairing Hue bulbs with Zigate/Domoticz?

santricity, 2018-12-07:
OK, I updated plugin to ver. 4.0.0, removed devices and added them again. Hue bulbs detected, I can turn them on and off and also dim them, but I cant change the color.

[image]

[image]

I can only set the brightness and light temperature.

pipiche38, 2018-12-08:
From the screenshot you are sharing it looks like the plugin detect the Bulb as a WW bul and not a Full RGB one. Is that correct ?

Could you kindly attached the file DeviceList-xx.txt ?

santricity, 2018-12-08:
Yes, that is correct, it was discovered as WW only. 

I fixed it by removing all Zigate devices, the Zigate Plugin hardware, the plugin itself from domoticz/plugin directory and installing it again. One Bulb was discovered immediately and properly as RGBWW. 

I'm still trying to discover the second Bulb, but without success. Is there something I can do? Before plugin reinstallation it was discovered but as WW Bulb as well. Now it can't show up in the devices list anymore. 

My latest DeviceList file (one Osram plug and one Hue Bulb discovered):
[DeviceList-7.txt_2018-12-08-21_22_22.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2660028/DeviceList-7.txt_2018-12-08-21_22_22.txt)

[image]

pipiche38, 2018-12-08:
When you say , without success, you mean wrong detection ? WW instead of RGBWW ?

pipiche38, 2018-12-08:
OkI, so that is an other issue. What do you have in the Log when you try to Pair them ?
If you want to chat I'm reachable on keybase.io

pipiche38, 2018-12-09:
@santricity could you kindly let me known what are the following devices:
62a2 (000b57fffe20ba82) which has a Domoticz Color Widget DeviceID: 65 ?
9a52 (000b57fffe117dcf) which has a Domoticz Witch, Color widget DeviceID: 66 ?

Why am I asking that is because there is a strange Model for those 2.

santricity, 2018-12-09:
These are my two Ikea Tradfri bulbs.

pipiche38, 2018-12-09:
I'm closing the issue.

closed 2018-12-09 by pipiche38