zigbeefordomoticz/Domoticz-Zigbee #275: Allow Domoticz Discover new devices fonction

deennoo, 2018-12-08:
When strating the plugin the zigate enter in join mode but domoticz doesn't

When adding a new object, it goes to the plugin & zigate database but not on domoticz one.

I have to stop plugin, erease device on DeviceList84.txt, allow new device on domoticz and restart the all procedure who is a bit tricky for device who aren't easy to reset.

pipiche38, 2018-12-08:
I don't understand what you mean by Domoticz does'nt. There is no such mode for Domoticz. 
The Join mode in the plugin is related to the Zigate HW component.

If now your issue, is that Domoticz doesn't allow to Create new devices, because you have disabled it, I'm afraid that we are not going to fix this issue as Domoticz doesn't give such information to the user.

I would be interested to have your logs, as if the creation of widgets in Domoticz doesn't succeed, you have an error message displaying this issue

deennoo, 2018-12-08:
No specifique message on the log.

Allowing Discovering new device is quiete easy : 

http://ip:port/json.htm?type=command&param=allownewhardware&timeout=5

pipiche38, 2018-12-08:
I'm not in favour to have the plugin overwriting (changing) a Domoticz settings. From a security standpoint this doesn't looks good to me.

Now, the reverse might be the right way to do , so for instance having Domoticz 'allow new hardware' to set the Permit to Join

pipiche38, 2019-02-18:
I'll start the implementation on the following principle:

At statup of the plugin and then every 5' , we will be checking Domoticz 'Accept New Hardware' flag

If Flag is set to '1':
    - Switch ON the Zigate in Permit to Join, for 255 seconds
    - When Zigate in Permit to Join , issue a Notification to the Notification Widget

If Flag is set to '0':
    - Switch Off the Zigate in Permit to Join 
    - When Zigate in not in Permit to Join; issue a Notification to the Notification Widget

pipiche38, 2019-02-18:
b0e4adc..f89c8cf

closed 2019-02-18 by pipiche38