zigbeefordomoticz/Domoticz-Zigbee #276: Allow the possibility to store report on different/customized folder

pipiche38, 2018-12-08:
It would be convenient if the user could configure where the LQI report can be stored

zaraki673, 2018-12-15:
reports should now be created in [Startup folder]/www/zigate/reports
folder is created by z_WebGui.py when GUI files are copied/updated

pipiche38, 2018-12-20:
https://github.com/sasu-drooz/Domoticz-Zigate/wiki/PluginConf

pipiche38, 2018-12-20:
Here after are the default directories based on pluginHome which is given by the Plugin Framework

```
Plugin Directories
-pluginHome: /var/lib/domoticz/plugins/Domoticz-Zigate/
-pluginData: /var/lib/domoticz/plugins/Domoticz-Zigate/./
-pluginZData: /var/lib/domoticz/plugins/Domoticz-Zigate/./Zdatas/
-pluginReports: /var/lib/domoticz/plugins/Domoticz-Zigate/../../www/templates/zigate/reports/
-pluginConfig: /var/lib/domoticz/plugins/Domoticz-Zigate/./
-pluginWWW: /var/lib/domoticz/plugins/Domoticz-Zigate/../../www/templates/
```

pipiche38, 2019-01-06:
available on the beta branch.
Closing

closed 2019-01-06 by pipiche38