zigbeefordomoticz/Domoticz-Zigbee #278: (Zigate) 'onStart' failed 'OSError'.

gte477k, 2018-12-08:
Hi, I am using plugin 4.0.1, Domoticz V4.10232, Python 3.5 on Windows 10. When I am starting Domoticz, I get the following error:
(Zigate) 'onStart' failed 'OSError'. Please see the log below.
[zigate1.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2660043/zigate1.txt)

zaraki673, 2018-12-08:
Hi,

yes plugin doesn't work under windows,
only linux for the moment

sorry

pipiche38, 2018-12-08:
@gte477k , can you try to install this file on the plugin directory Domoticz-Zigate.

You'll have to unzip it. 

[z_database.py.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2660063/z_database.py.zip)

gte477k, 2018-12-09:
Good morning zaraki673, it seems to work, I could add a device and the switch states are correctly entered into the database. There are still some error messages, if you like to have a look:

[zigate2.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2660327/zigate2.txt)

pipiche38, 2018-12-09:
@gte477k , glad that the patch makes it working 👍 

As the error message in the log, could you confirm that the Device and the Widget in Domoticz were created correctly ? If I well understood from the log it is Motion Xiaomi device

gte477k, 2018-12-09:
@pipiche38, many thanks for your help. 

Yes, I would say so. I added a Motion Xiaomi device and an Ikea bulp. Both work as expected and the Widgets are correctly created incl. dimming for the bulp. Please see attached screenshot.

[zigate3.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2660385/zigate3.zip)

pipiche38, 2018-12-09:
Patch integrated.

closed 2018-12-09 by pipiche38

gte477k, 2018-12-10:
Hi, I checked out the latest version, removed all devices and tried to pair them again. However, this fails with this version (althouh it has worked in version 4.0.1). Please see the log:

[zigate4.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2662200/zigate4.txt)

pipiche38, 2018-12-10:
Most-likely it failed, because you didn't authorized new hardware in Domoticz !
Basically Domoticz didn't allow the creation of the widget

gte477k, 2018-12-10:
Sorry, my fault. I forgot. Everything is fine and working.

gte477k, 2018-12-10:
Many thanks for your support!