zigbeefordomoticz/Domoticz-Zigbee #288: Aqara bouton reconnu en cube magik

Hydci, 2018-12-12:
**Describe the bug**
Bouton aqara reconnu comme un cube magik 

Log:

2018-12-12 19:32:34.009 Status: (Zigate) Permit Join is On
2018-12-12 19:33:27.121 Status: (Zigate) [0] NEW OBJECT: f1a5 Device Annouce
2018-12-12 19:33:27.323 Status: (Zigate) [0] NEW OBJECT: f1a5 Device Annouce
2018-12-12 19:33:31.151 Status: (Zigate) [0] NEW OBJECT: f1a5 (re)Starting process
2018-12-12 19:33:36.325 (Zigate) Decode8043 - Overwrite ProfileID {} with 0104 from Ep: 01
2018-12-12 19:33:36.325 (Zigate) Decode8043 - Overwrite ZDeviceID {} with 0104 from Ep: 01
2018-12-12 19:33:36.165 Status: (Zigate) [0] NEW OBJECT: f1a5 Ox8045 received Infos
2018-12-12 19:33:36.165 Status: (Zigate) [0] NEW OBJECT: f1a5 Request Simple Descriptor for Ep: 01
2018-12-12 19:33:36.166 Status: (Zigate) [0] NEW OBJECT: f1a5 Request Attributes for Cluster 0x0001
2018-12-12 19:33:36.324 Status: (Zigate) [-] NEW OBJECT: f1a5 Simple Descriptor EP 01
2018-12-12 19:33:36.325 Status: (Zigate) [-] NEW OBJECT: f1a5 ProfileID 0104
2018-12-12 19:33:36.325 Status: (Zigate) [-] NEW OBJECT: f1a5 ZDeviceID 0103
2018-12-12 19:33:36.325 Status: (Zigate) [-] NEW OBJECT: f1a5 Cluster In 1: 0000
2018-12-12 19:33:36.325 Status: (Zigate) [-] NEW OBJECT: f1a5 Cluster In 2: 0012
2018-12-12 19:33:36.325 Status: (Zigate) [-] NEW OBJECT: f1a5 Cluster In 3: 0003
2018-12-12 19:33:36.325 Status: (Zigate) [-] NEW OBJECT: f1a5 Cluster Out 1: 0000
2018-12-12 19:33:41.185 (Zigate) processNotinDBDevices - ready for creation: {'Battery': {}, 'SQN': '01', 'Stamp': {'Time': '2018-12-12 19:33:27', 'MsgType': '8102'}, 'MacCapa': '80', 'RSSI': 180, 'App Version': '2', 'Version': '3', 'Heartbeat': '1', 'Model': 'lumi.remote.b1acn01', 'RIA': '0', 'ZDeviceID': '0103', 'Type': {}, 'NbEp': '1', 'Status': '8043', 'ProfileID': '0104', 'Ep': {'01': {'0003': {}, '0012': {}, '0000': {}}}, 'ReadAttributes': {'Ep': {'01': {'0001': {}}}}, 'IEEE': '00158d0001ef862a'}
2018-12-12 19:33:41.195 (Zigate) Devices size has changed , let's write ListOfDevices on disk
2018-12-12 19:33:41.184 Status: (Zigate) [0] NEW OBJECT: f1a5 Ox8043 received Infos
2018-12-12 19:33:41.184 Status: (Zigate) [0] NEW OBJECT: f1a5 Trying to create Domoticz device(s)

**Plugin (please complete the following information):**
 - OS: raspbian
 - Plugin Dernière version
 - Plugin branch Stable

pipiche38, 2018-12-12:
Seems to be a non certified yet device.
Model: lumi.remote.b1acn01'

Please provide the reference of this Device and expected behaviour

Hydci, 2018-12-12:
Voici le bouton concerné https://zigate.fr/xiaomi-bouton-switch-aqara-clusters/
Comportement attendu click double click

pipiche38, 2018-12-12:
Probably a new version as the Model name is not the same.
Could you please Edit the file DeviceConf.txt and line 10 add the following line

`'lumi.remote.b1acn01':{'Ep':{'01':{'0006'}},'Type':'SwitchAQ2'},`

Then restart the plugin, remove the device (from Domoticz ) which is currently created.
Then redo a commissioning.

Let me know

Hydci, 2018-12-12:
Yes en rajoutant la ligne cela fonctionne il et reconnu en temps que sélecteur par contre impossible d’interagir avec 

Ps: je vous et envoyer un message aussi sur le forum domoticz français sous le même pseudo

pipiche38, 2018-12-12:
Could you please put the plugin in Debug mode "Verbose" and then do few actions with the Switch.
Please then attach the log into this issue 
This will allow us to see if we could implement this new switch (as from a firmware standpoint it doesn't use the mechanism of the previous one)

Hydci, 2018-12-13:
Quand je clique avec le bouton voici ce que cela me fait avec l'option Verbose demander 

2018-12-13 06:48:27.173 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:27.175 (Zigate) HB count = 36
2018-12-13 06:48:31.435 (Zigate) receiveData - new Data coming
2018-12-13 06:48:31.435 (Zigate) receiveData - MsgType: 8102
2018-12-13 06:48:31.435 (Zigate) ZigateRead - decoded data : 018102000f0f0ede950100120055002100020001a203 lenght : 44
2018-12-13 06:48:31.436 (Zigate) ZigateRead - MsgType: 8102, MsgLength: 000f, MsgCRC: 0f, Data: 0ede950100120055002100020001; RSSI: a2
2018-12-13 06:48:31.436 (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000f0f0ede950100120055002100020001a203
2018-12-13 06:48:31.436 (Zigate) Decode8102 - Individual Attribute response : [de95:01] ClusterID: 0012 AttributeID: 0055 Status: 00 Type: 21 Size: 0002 ClusterData: >0001<
2018-12-13 06:48:31.436 (Zigate) DeviceExist - Found in ListOfDevices with status = inDB
2018-12-13 06:48:31.436 (Zigate) Decode8102 : Attribute Report from de95 SQN = 0e ClusterID = 0012 AttrID = 0055 Attribute Data = 0001
2018-12-13 06:48:31.437 (Zigate) Device : de95 MacCapa : 80 updating SQN to 0e
2018-12-13 06:48:31.437 (Zigate) updSQN - Device : de95 updating SQN to 0e
2018-12-13 06:48:31.437 (Zigate) ReadCluster - MsgData lenght is: 28 out of 24+
2018-12-13 06:48:31.437 (Zigate) DeviceExist - Found in ListOfDevices with status = inDB
2018-12-13 06:48:31.437 (Zigate) Cluster0012 - unknown Model: lumi.remote.b1acn01 for this Attribute: 0055
2018-12-13 06:48:31.437 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:32.189 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:32.190 (Zigate) HB count = 37
2018-12-13 06:48:37.199 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:37.200 (Zigate) HB count = 38
2018-12-13 06:48:42.210 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:42.211 (Zigate) HB count = 39
2018-12-13 06:48:47.221 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:47.222 (Zigate) HB count = 40
2018-12-13 06:48:52.182 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:52.183 (Zigate) HB count = 41
2018-12-13 06:48:57.193 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:48:57.194 (Zigate) HB count = 42
2018-12-13 06:49:02.204 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:02.206 (Zigate) HB count = 43
2018-12-13 06:49:07.214 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:07.215 (Zigate) HB count = 44
2018-12-13 06:49:12.224 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:12.225 (Zigate) HB count = 45
2018-12-13 06:49:17.184 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:17.185 (Zigate) HB count = 46
2018-12-13 06:49:22.194 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:22.196 (Zigate) HB count = 47
2018-12-13 06:49:27.205 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:27.207 (Zigate) HB count = 48
2018-12-13 06:49:32.216 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:32.218 (Zigate) HB count = 49
2018-12-13 06:49:37.228 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:37.229 (Zigate) HB count = 50
2018-12-13 06:49:42.188 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:42.190 (Zigate) HB count = 51
2018-12-13 06:49:47.199 (Zigate) checkTOwaitFor - Cmd: 0000 waitQ: 0 dataQ: 0 normalQ: 0
2018-12-13 06:49:47.200 (Zigate) HB count = 52

Hydci, 2018-12-13:
J'ai supprimer le bouton pour tout recommencer et quand je mais votre patch et redémarre tout je synchronise mon bouton a la zigate et voila ensuite ça bloque et ne fait plus rien 

2018-12-13 09:08:09.875 Status: (zigate) Permit Join is On
2018-12-13 09:08:34.752 (zigate) ZigateRead - MsgType 8048 - Reception Leave indication : 018048000a1800158d0001ef862a000003
2018-12-13 09:08:34.752 (zigate) Calling leaveMgt to request a rejoin
2018-12-13 09:08:34.752 (zigate) Switch to Permit to Join for 5s
2018-12-13 09:08:34.753 (zigate) leaveMgt - sAddr: a0c7 , ieee: 00158d0001ef862a
2018-12-13 09:08:34.752 Status: (zigate) Decode8048 - Leave indication, IEEE : 00158d0001ef862a Status : Success
2018-12-13 09:08:36.761 (zigate) DeviceExist - Update Status from 'Left' to 'inDB' for NetworkID : 4893
2018-12-13 09:08:36.760 Status: (zigate) [0] NEW OBJECT: 4893 Device Annouce
2018-12-13 09:08:36.761 Status: (zigate) NetworkID : 4893 is replacing a0c7 and is attached to IEEE : 00158d0001ef862a
2018-12-13 09:08:37.164 Status: (zigate) [0] NEW OBJECT: 4893 Device Annouce

Hydci, 2018-12-13:
Bonsoir,

D'accord merci pour l'ajout du correctif j'ai répondu sur le forum Français pour d'autre truc ;)

pipiche38, 2018-12-14:
[z_domoticz.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2679835/z_domoticz.zip)
[z_output.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2679836/z_output.zip)

closed 2018-12-15 by Hydci