zigbeefordomoticz/Domoticz-Zigbee #292: Domoticz crashes after invalid transport type error

hkhk12, 2018-12-15:
A few times per week domoticz crashes/stops.
I've been able to catch the log once and the last message in domoticz.log was:

_[1;31mError: (zigate) Invalid transport type for connecting specified: '61', valid types are TCP/IP and Serial.[0;0m_

other zigate errors in the log (about one day) are:

_(zigate) Out of sequence for Device: bf15 SQN move from 47 to 4e gap of : 7
(zigate) Out of sequence for Device: 6079 SQN move from c9 to cf gap of : 6
(zigate) Out of sequence for Device: 7f92 SQN move from 35 to 3b gap of : 6_

I running domoticz on a Synology NAS and use version 4.9700

Using stable v 4.0.1 branch

Have the following zigbee devices:
5 aqara motion sensors
1 aqara single switch (battery powered)
13 ikea gu10 leds
2 ikea e27 leds
2 ikea wall plugs
10 osram wall plugs

pipiche38, 2018-12-15:
As regards to the 1st topic, can you provide the domoticz log around this error message ?

On the 2nd topic SQN this just tell that  there is some in the Sequence number. No issue at that time. SQN needs to be reworked

hkhk12, 2018-12-15:
Here it is.
Last 90 minutes before the crash.
[domoticz.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2682655/domoticz.txt)

pipiche38, 2018-12-15:
@hkhk12 
could you kindly update the plugin to the latest Stable version 4.0.4
and then install the attached file in Domoticz-Zigate (it is not fixing anything, but it will print additional information when getting the invalid transport issue). 

[z_Transport.py.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2682657/z_Transport.py.zip)

hkhk12, 2018-12-15:
Did that but shows version 4.0.3 instead of 4.0.4 when the plugin is started.

hkhk12, 2018-12-15:
Please ignore.

_Initialized version 4.0.3, author 'zaraki673 pipiche38'
Status: (zigate) onStart called - Zigate plugin V 4.0.4_

pipiche38, 2018-12-15:
No issue, in order to for domoticz to take in consideration update a restart of domoticz is required. If you simply disable/enable the plugin you don't reload the full plugin and you get such discrapency

hkhk12, 2018-12-20:
Had domoticz crash again today after it had been running fine for 5 days.
Did set join to 255 earlier this week (was set top 0 before) . Could this be related?

attached the log.

[domoticz.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2700500/domoticz.txt)

pipiche38, 2018-12-21:
It looks like indeed the Connection was totally crap, something like some parameters have been overwritten !!!!

I For me there are 2 issues:
	- Root cause is a loss of the communication. This  needs to be investigated . If that is a USB Zigate. Is the Zigate connected directly to the UBS port, or do you go via USB hub or something like that ? If that is Wifi, It would be interesting to know why such lost

	- The consequence of the connection loss is the restart which is not happening. It might be related to a plugin problem. I’m attaching a revised version of the Transport module. Which should tackle that one. As it looks like Domoticz has totally reset the connection.

Kind regards
Patrick

hkhk12, 2018-12-21:
Hi Patrick,

The zigate is WiFi connected.
It's not the join setting since it happened again this afternoon while join was set to 0.
Did notice yesterday that 1 Ikea would not let itself being controlled. I reset the led by switching on/off 6 times but since it was on the same switch as another one, both where reset. After that they had switched place in Domoticz and the other one wold not let itself being controlled.
After the Domoticz crash the behavior is ok again though.

Did you miss the attachment? I didn't receive it.

pipiche38, 2018-12-22:
Indeed it looks like I have missed the attachment (in fact I suspect the mail gateway had filtered out)

[z_Transport.py.zip](https://github.com/sasu-drooz/Domoticz-Zigate/files/2705357/z_Transport.py.zip)

pipiche38, 2018-12-22:
498ceac

pipiche38, 2019-01-14:
As per the various log file, it looks as a Connection Stop comming from Domoticz

pipiche38, 2019-01-14:
Suggest to close this issue, until we find the root cause why Domoticz is shutting down the connection

closed 2019-01-14 by pipiche38