zigbeefordomoticz/Domoticz-Zigbee #293: "Erase Persistent Data" setting is persistent

dpocock, 2018-12-15:
**Describe the bug**
Should the "Erase Persistent Data" setting be a one-shot operation?  At the moment, if you set the option to True, shutdown, start up again the next day, it is still True.

**To Reproduce**
Steps to reproduce the behavior:
1. Set "Erase Persistent Data" to True
2. Click "Add"
3. Add some devices
4. stop domoticz, maybe shut down the host, come back later, etc
5. start the host again, start domoticz again
6. Look at the hardware config, notice that "Erase Persistent Data" is still True

**Expected behavior**
After erasing the data, the setting should go back to "False", or maybe there should be an "Erase" button instead of a True/False setting like this.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Raspbian stretch
 - Browser Firefox
 - Version [e.g. 22]

**Plugin (please complete the following information):**
 - OS: [e.g. Linux Fedora 28]
 - Plugin version
 - Plugin branch

**Additional context**
Add any other context about the problem here.

pipiche38, 2018-12-15:
The main issue is that the setting is doing outside of the plugin : In Domoticz
and we don't have API to change it

zaraki673, 2018-12-15:
Should be possible with a device for the ZIGATE, where a button is for
enter in join mode, another erase persistent data, and other if need

pipiche38, 2018-12-15:
Indee, might be a better solution to remove this option from the Plugin Option and get a special widget created at Plugin start in Domoticz, where you could have : 
- Erase PDM
- Reset
- Network Scan
- LQI report

pipiche38, 2019-02-14:
002a8ea..c22f3c3

Fixed on pre-4.1 branche and required Domoticz > 4.10335

closed 2019-02-14 by pipiche38