zigbeefordomoticz/Domoticz-Zigbee #294: support for the 3A "Neu" smart plug/socket

dpocock, 2018-12-15:
**Is your feature request related to a problem? Please describe.**
Tried to join the 3A smart socket, it doesn't appear in the list of devices.  Is some code change required to support this device?

**Describe the solution you'd like**
The socket should appear in the list of devices.

**Describe alternatives you've considered**
Other brands of smart socket

**Additional context**
https://www.amazon.co.uk/gp/product/B07GSNJWPN
https://www.youtube.com/watch?v=SUGsmZ4BjQg

pipiche38, 2018-12-15:
Please:
Edit the file PluginConf.txt and set  'enablegroupmanagement':'1' (from 0 to 1)

Then try to pair this device, 

Please provide the Logfile and check if there is any file produce on the Zdatas folder

dpocock, 2018-12-16:
Yesterday, I tried adding the setting enablegroupmanagement and then tried pairing again and it still didn't work.  I tried it a few times.

Today, I deleted the Zigate from the hardware page and told it to clear persistent data and then I started again with the logging enabled.  However, this time it paired successfully on the first attempt and I was able to turn the socket and off using the web interface.

After a Domoticz restart, however, I can still see it in the web interface but if I click to turn it off, the little status window appears to say it is turning it off but it remains on.  Could this be related to the other problem I am having, where my temperature sensors don't update any more? (issue #295)

dpocock, 2018-12-16:
Given that it was able to join today, is it still useful to try that setting?

Now that it has joined with Zigate, the problem is that it doesn't allow me to control the socket after restarting the domoticz process.

dpocock, 2018-12-16:
looks like this was the same as issue #295 so closing as a duplicate

closed 2018-12-16 by dpocock