zigbeefordomoticz/Domoticz-Zigbee #295: sensors joined but data never changes

dpocock, 2018-12-15:
**Describe the bug**
I joined 2 temperature sensors (Xiaomi) and 1 motion sensor (Osram Lightify).  In the devices tab, they appear in the list with the initial readings and the "Last Seen" column shows the time they joined.  The values and "Last Seen" times are not changing.

**To Reproduce**
Steps to reproduce the behavior:
1. Hardware tab: put the Zigate in permit join mode
2. put the device in joining mode, for example, Xiaomi temperature sensor: press button a few times until blue light flashes on sensor
3. Go to Devices tab
4. See if the device has appeared, if not, go back to step 2
5. look at the devices tab again after 30 minutes, has the "Last Seen" time changed or is it still the initial time and value when the device joined?

**Expected behavior**
Values updating periodically

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Raspbian stretch
 - Browser firefox
 - Version [e.g. 22]

**Plugin (please complete the following information):**
 - OS: [e.g. Linux Fedora 28]
 - Plugin version
 - Plugin branch

**Additional context**
Add any other context about the problem here.

pipiche38, 2018-12-15:
Please provide the Log file as we cannot really see what is going on. Reporting via the forum would a good start

zaraki673, 2018-12-15:
and can you explain how you put your device in discovery mode ? (for the xiaomi temp sensor for example)

dpocock, 2018-12-16:
for the Xiaomi temp sensor, I hold down the button until the blue light flashes three times and then I press the button again a few times until it flashes again.

dpocock, 2018-12-16:
I erased persistent data and deleted the Zigate from the hardware tab so that I would start again fresh and capture the logs from the beginning.

After joining the Xiaomi temp sensor, it was working for about 15 minutes.  Then I restarted the domoticz process and the values didn't update any more.

Is there anything in particular to search for in the logs or do I need to post the whole log file?

Here are the steps I followed:

1. erase persistent data and "Update" button
2. click the zigate and "Delete" button
3. shutdown domoticz
4. start domoticz from command line with logging enabled
5. add the Zigate again, with join time = 255
6. press and hold the button on my 3A smart socket (issue #294 ) and it joined
7. tested turning the 3A smart socket and and off a couple of times from the web browser
8. stopped domoticz
9. started domoticz again
10. tried to use the browser to control the smart socket, it says it is turning off but it doesn't work
11. joined the Xiaomi temperature sensor
12. left it running for 15 minutes, noticed it was updating the temperature regularly
13. stopped domoticz
14. started domoticz
15. waited another 10 minutes, no temperature updates from the temp sensor and I still can't control the socket either

pipiche38, 2018-12-16:
Please provide logs, we cannot analyse anything without it.
Then when you do such Erase, did you also remove the DeviceList-xx which is the Plugin database ?

Again, this is not the best place , please use the forum where the community will benefits from that and where you can also get support.

dpocock, 2018-12-16:
I posted it in the forum:
https://www.domoticz.com/forum/viewtopic.php?f=68&t=26121

I attached the logs but can't see them in the post

dpocock, 2018-12-16:
It appears difficult putting attachments in the forum, so here are the logs:

[domoticz.log](https://github.com/sasu-drooz/Domoticz-Zigate/files/2683633/domoticz.log)
[domoticz-stdout-stderr.log](https://github.com/sasu-drooz/Domoticz-Zigate/files/2683634/domoticz-stdout-stderr.log)

dpocock, 2018-12-16:
> Then when you do such Erase, did you also remove the DeviceList-xx which is the Plugin database ?

It turns out this was quite a useful clue, it occurred to me some time after seeing this question that maybe the plugin is trying to write a file somewhere that it can't.

I'm currently testing the [Debian packages of Domoticz](https://bugs.debian.org/899058) and preparing to make an official package of the Zigate plugin to go with that.  So my environment is like this:

- all files under ```/usr/lib/domoticz/plugins/Domoticz-Zigate```
- files there are owned by ```root``` and not modifiable by other users
- domoticz process runs as the ```domoticz``` user on Linux, so it can't write to that directory

To test, I tried making the directory writeable by the ```domoticz``` user, repeated the whole setup process (erase, join, etc) and then I saw that ```DeviceList-2.txt``` was created.  I tried a few restarts of the process and everything still works after each restart.

Can you please tell me:

- how many files need to be created or modified by the plugin at runtime?
- how can we ensure those files are written under ```/var/lib/domoticz/something``` ?  Is there a config option or will it require a patch?
- can you think of any other issues that might be encountered by somebody using a package like this?

I'll also post more details in the [Debian ITP](https://bugs.debian.org/899058) about how I cross-compile the Domoticz package and Zigate package on a Debian machine to be installable with ```apt``` on Raspbian stretch.

pipiche38, 2018-12-16:
@dpocock Daniel, I don't know if you had a look to the Wiki, but there are some information available there.

I understand a bit better what you try to do , and I appreciate your effort to make a package of it.

The plugin has to manage its own database as Domoticz doesn't allow to do so via the Python Plugin Framework.
- DeviceList-xx.txt is the plugin DB , where xx stands for the HardwareID of the plugin in Domoticz.
( at each startup, the plugin is creating a local copy of the file DeviceList-xx.txt )
- PluginConf.txt ( PluginConf-xx.txt ) are config files which can be updated by the end user

- In case you 'allowStoreDiscoveryFrames':'0' enable with '1', then a file will be created under Zdatas (which is under Domoticz-Zigate)

Some reports could be created and today 'Stable' 4.0.x they will be created on the Domoticz-Zigate directory:

LQI_report-<date> if you enable 

We are in the process of working on 4.1.x (currently under dev) and we will be implementing various report and also going to write some files on:

<Domoticz HomDire>/www/templates

So for me this is quiet a project to kick-off in order to make the plugin packageable. Especially to take in consideration of /var/lib/domoticz/plugin/Domoticz-Zigate

dpocock, 2018-12-16:
This is actually not as hard as it may appear, I've worked on many packages with similar issues.  Package maintainers can do small workarounds with symlinks.

If the whole directory needs to be writeable then it is difficult but if all the writeable files are in a subdirectory then it is very easy.  Can you just change the code to look for ```Zdatas/DeviceList-xx.txt``` instead of ```DeviceList-xx.txt``` ?

For files that the user needs to modify, my package will install them in ```/etc/domoticz/plugins/Domoticz-Zigate``` and I can symlink to ```/usr/lib/domoticz/plugins/Domoticz-Zigate/PluginConf-xx.txt``` or you could also have a directory ```plugins/Domoticz-Zigate/etc``` and I symlink the whole directory, that is easier to maintain in future.

If you can make every directory name configurable in a config file that is ideal but if that is too much work right now, simply having dedicated subdirectories for each type of file means that it is easy for a package maintainer to create the symlinks.

pipiche38, 2018-12-16:
`if the whole directory needs to be writeable then it is difficult but if all the writeable files are in a subdirectory then it is very easy. Can you just change the code to look for Zdatas/DeviceList-xx.txt instead of DeviceList-xx.txt ?`
I need to think , as current installation must continue to work.

`For files that the user needs to modify, my package will install them in /etc/domoticz/plugins/Domoticz-Zigate and I can symlink to /usr/lib/domoticz/plugins/Domoticz-Zigate/PluginConf-xx.txt or you could also have a directory plugins/Domoticz-Zigate/etc and I symlink the whole directory, that is easier to maintain in future.`
Sound Good

`If you can make every directory name configurable in a config file that is ideal but if that is too much work right now, simply having dedicated subdirectories for each type of file means that it is easy for a package maintainer to create the symlinks.`
That I think what I'm going to address later, but that is the proper way forward

pipiche38, 2018-12-20:
Implementation part of #299 
https://github.com/sasu-drooz/Domoticz-Zigate/wiki/PluginConf

pipiche38, 2018-12-29:
Link to #299. If not please feel free to re-open

closed 2018-12-29 by pipiche38