zigbeefordomoticz/Domoticz-Zigbee #296: plugin doesn't display error in UI when permissions problem occurs accessing ttyUSB0

dpocock, 2018-12-15:
**Describe the bug**
Domoticz running as Linux user 'domoticz'
The USB device /dev/ttyUSB0 can only be accessed by users in group 'dialout'
The domoticz user is not in the dialout group.
The permission denied error only appears in the log file and only if it is in debug mode.
It would be desirable to make errors like this more visible in the UI to help first time users start more quickly.

pipiche38, 2018-12-15:
Please report it to Domoticz, this is not a plugin issue

dpocock, 2018-12-16:
could you be more specific?  For example, do the Domoticz developers provide an API method for plugins to display errors or is that a missing feature in their API?

pipiche38, 2018-12-18:
@dpocock I'm going to close this case as it is a pure Domoticz one.

The Zigate Plugin use the Python Plugin Framework to open outside connection as Serial or TCP/IP.

closed 2018-12-18 by pipiche38