zigbeefordomoticz/Domoticz-Zigbee #299: Make the plugin packageable and get the proper Directory Tree

pipiche38, 2018-12-16:
In order to get the plugin in a Debian, Fedora or whaterver distribution packaged, it is important to review the directory structure.

#295

pipiche38, 2018-12-16:
Here after is what could the Directory tree looks like

Domoticz-Zigate
     - config/PluginConf.txt
     - config/DeviceConf.txt
     - data/DeviceList-xx.txt
     - data/DiscoveryDevice-xxxx.txt
     - reports/LQI-reports-xx.txt
     - reports/Network_scan-xx.txt

Using of configparser module. To validate if this comply with Domoticz python Framework and Windows environment.

[DEFAULT]
pluginHome = './'
pluginData = './data/'
pluginReports = './reports'
pluginConfig = './config'

[GROUPSMANAGEMENT]
enablegroupmanagement = 1
discoverZigateGroups = 1
enableConfigGroups = 1

[NETWORKTOPOLOGY]
enableLQI = 2600
enableSCAN = 3600

[DEBUGING]
debugReadCluster = 1

pipiche38, 2018-12-20:
https://github.com/sasu-drooz/Domoticz-Zigate/wiki/PluginConf

pipiche38, 2018-12-21:
@dpocock if you want to try this is on the dev-grpmgt branche

pipiche38, 2018-12-29:
Feel free to check with the latest dev-grpmgt branch

pipiche38, 2019-01-06:
Available now on the beta-branch.
Closing this issue

dpocock, 2019-01-17:
I can probably test this in the next few days, should I just checkout from beta?  Is there anything else I need to be aware of when testing?

I also made a note in the forum about another problem, not receiving values from an Osram Lightify/Smart+ Motion sensor, could you please give me feedback there?  I would like to test that at the same time.
https://www.domoticz.com/forum/viewtopic.php?f=68&t=26559

pipiche38, 2019-01-17:
Daniel,

Beta is the best place .

Here after is the default setup: 
        self.homedirectory is the Plugin Home directory 

        pluginData = self.pluginHome + './'                                         RW
        pluginZData = self.pluginHome + './Zdatas/'                            RW
        pluginConfig = self.pluginHome + './'                                      RO
        pluginWWW = self.pluginHome + '../../www/templates/'          RW
        pluginReports = self.pluginWWW + 'zigate/reports/'               RW

I suggest to leave pluginConfig to default . Or you will have to move manualy the file DeviceConf.txt

The file PluginConf.txt is expected to be in the Plugin Home directory. So ether you do a Sym Link to the Config directory to access it in order to have the Home readonly.

In any case let me know how it works.

pipiche38, 2019-02-07:
For information:
https://github.com/sasu-drooz/Domoticz-Zigate/wiki/Plugin-directory-structure

pipiche38, 2019-02-19:
Done in pre-4.1

closed 2019-02-19 by pipiche38

dpocock, 2019-07-14:
@pipiche38 thanks for your efforts on this.

I made a package and pushed it here on the debian/sid branch:
https://gitlab.com/dpocock/domoticz-zigate/tree/debian/sid

You can test it:
git clone https://gitlab.com/dpocock/domoticz-zigate
cd domoticz-zigate
git checkout debian/sid
dpkg-buildpackage -rfakeroot -i.git -b

I also made an updated version of the domoticz.deb package, it includes patches from the Debian ITP and uses a more recent version of the Domoticz code:
https://gitlab.com/dpocock/domoticz-debian-raspbian/tree/debian/backports/stretch

and it can be built with the same procedure:

git clone https://gitlab.com/dpocock/domoticz-debian-raspbian
cd domoticz-debian-raspbian
git checkout debian/backports/stretch
dpkg-buildpackage -rfakeroot -i.git -b

I was able to install both packages onto a Debian amd64 system using the following command:

```
dpkg -i domoticz_4.10717-1_amd64.deb domoticz-zigate-plugin_4.3.4-1_amd64.deb
```

and they just work.  I'll make them available in an APT repository for other people to test, along with the packages of the firmware update utilities.

I haven't done a Raspbian build yet.  I was able to build older versions of the package on Raspbian.  I'll provide more feedback after testing that.

pipiche38, 2019-08-01:
@dpocock Daniel, for your information release 4.5 has been out .